zephyrproject-rtos/zephyr #60769: posix: pthread_self() returns a non-existing thread ID

cspiel1, 2023-07-25:
**Description**
`pthread_self()` returns a wrong thread index number. All functions that depend on `pthread_self()` do not work, e.g. `pthread_getspecific()`, `pthread_setspecific()`.

Target platform: `mimxrt1170_evk_cm7`

I tried some switches that are used at relevant sections in `kernel.h`:
- `CONFIG_THREAD_LOCAL_STORAGE=y/n`
- `CONFIG_USERSPACE=y/n`

**Analysis**
Here is a faulty line:
```c
t = (struct posix_thread *)CONTAINER_OF(k_current_get(), struct posix_thread, thread);
```
The pointer returned by `k_current_get()` does not lie in the thread pool `posix_thread_pool`.

As an aftereffect in function `to_posix_thread()` the index `bit` is to high.
```
if (bit >= CONFIG_MAX_PTHREAD_COUNT) {
		return NULL;
}
```

Then `pthread_setspecific()` crashes with a NULL pointer access.

**To Reproduce**
1. Apply this patch:
```patch
index 182debd5f9..0d79b92b86 100644
--- a/samples/posix/gettimeofday/src/main.c
+++ b/samples/posix/gettimeofday/src/main.c
@@ -11,9 +11,26 @@
 #include <time.h>
 #include <unistd.h>
 
+#include <pthread.h>
+
 int main(void)
 {
        struct timeval tv;
+       pthread_key_t key;
+       int err;
+
+       pthread_t tid = pthread_self();
+
+       err = pthread_key_create(&key, NULL);
+       if (err)
+               return err;
+
+       printf("%s:%d tid: %u\n", __func__, __LINE__, (unsigned) tid);
+       err = pthread_setspecific(key, "test");
+       if (err) {
+               printf("%s:%d err=%s\n", __func__, __LINE__, strerror(err));
+               return err;
+       }
 
        while (1) {
```

2. build with west and flash
```bash
cd samples/posix/gettimeofday
west build -b mimxrt1170_evk_cm7 . && west flash 
```

**Expected behavior**
- `pthread_self()` returns a thread that is tracked by `posix_thread_pool`
- `bit=get_posix_thread_idx(pthread_self())` is an index in the array `posix_thread_pool`
- `pthread_setspecific()` and `pthread_getspecific()` work like expected

**Impact**
showstopper

**Logs and console output**
```
main:28 tid: 3729840033
[00:00:30.572,000] <err> os: ***** USAGE FAULT *****
[00:00:30.572,000] <err> os:   Unaligned memory access
[00:00:30.572,000] <err> os: r0/a1:  0x800034bc  r1/a2:  0x00000000  r2/a3:  0x800054
[00:00:30.572,000] <err> os: r3/a4:  0x0000000c r12/ip:  0x0000001b r14/lr:  0x30008f
[00:00:30.572,000] <err> os:  xpsr:  0x210e0000
[00:00:30.572,000] <err> os: Faulting instruction address (r15/pc): 0x30020776
[00:00:30.572,000] <err> os: >>> ZEPHYR FATAL ERROR 31: Unknown error on CPU 0
[00:00:30.572,000] <err> os: Current thread: 0x80003208 (unknown)
[00:00:30.590,000] <err> os: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 22.04
 - Toolchain: Zeyphyr  Wed Jul 19 2023
 - 13a6c454529e64cdb9f69cab0f0a9fb8c539b192

cfriedt, 2023-07-25:
@cspiel1 - what you're describing is actually expected behaviour on Zephyr.

That might sound confusing, so here it is in a nutshell:

* Zephyr's `main` thread is a separate `struct k_thread`
* allocated externally to the POSIX API
* not even partially a `pthread_t` or `struct posix_thread`
* we filter-out invalid pointers, so ..
* attempting to use Zephyr's `main` `k_thread` as a `pthread_t` will fail (in most cases)
* the exception to that is `pthread_equal()`, because..
* we still need to be able to compare arbitrary `pthread_t` values
* even the one that would otherwise represent Zephyr's `main` thread

I do sympathize though and we should absolutely include a caveat in the documentation to that effect.

For the foreseeable future, one does need to call `pthread_create()` in order to get an actual `pthread_t` handle to perform actions on POSIX threads.

[`tests/posix/common/src/key.c`](https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/posix/common/src/key.c)

In the mean time, I'll adjust this issue into a `Documentation` bug. If there is anything that needs clarification, feel free to jump on the Zephyr [Discord](https://discord.com/invite/Ck7jw53nU2) to discuss.

cspiel1, 2023-07-26:
Okay, this makes sense. Thanks for the quick clarification!

closed 2023-07-26 by cspiel1

cspiel1, 2023-07-26:
Beside putting all the code in a pthread, I think that also these settings are necessary. But please correct me, if I am wrong!
```
CONFIG_DYNAMIC_THREAD=y
CONFIG_DYNAMIC_THREAD_PREFER_POOL=y
CONFIG_DYNAMIC_THREAD_POOL_SIZE=1
```
And optionally a larger stack:
```
CONFIG_DYNAMIC_THREAD_STACK_SIZE=8192
```