zephyrproject-rtos/zephyr #60800: west flash for nucleo_l011k4

JE-Tv1, 2023-07-25:
**Describe the bug**
I followed the [getting started](https://docs.zephyrproject.org/latest/boards/arm/nucleo_l011k4/doc/index.html#l). Nothing occured badly until _west flash_ was done. The output is shown in the logs.

Trying to debug this via comparison, I checked the nucleo_l031k6 and tried to build and flash it. Similar output in the log is generated but my board now has the script and runs as well. I also tried with hello_world and same works. I noticed the deconfig file between them is the only difference so I adapted the nucleo_l031k6_deconfig to nucleo_l011k4. Ultimately changing the "Kernel Option" CONFIG_MAIN_STACK_SIZE, CONFIG_IDLE_STACK_SIZE, CONFIG_ISR_STACK_SIZE.

**To Reproduce**
Follow along with the tutorial

**Logs and console output**
```
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /Users/johnedwardtan/ZephyrOS/zephyrproject/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 300 kHz
Info : STLINK V2J31M21 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.262549
Info : [stm32l0.cpu] Cortex-M0+ r0p1 processor detected
Info : [stm32l0.cpu] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for stm32l0.cpu on 3333
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32l0.cpu        hla_target little stm32l0.cpu        running

Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
[stm32l0.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x08000e14 msp: 0x20000530
STM32L0: Enabling HSI16
Info : Unable to match requested speed 2500 kHz, using 1800 kHz
Info : Unable to match requested speed 2500 kHz, using 1800 kHz
Info : Device: STM32L0xx (Cat.1)
Info : STM32L flash size is 16kb, base address is 0x8000000
auto erase enabled
wrote 16384 bytes from file /Users/johnedwardtan/ZephyrOS/zephyrproject/zephyr/build/zephyr/zephyr.hex in 1.869062s (8.560 KiB/s)

Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
shutdown command invoked
```

**Environment (please complete the following information):**
 - OS: MacOs
 - Zephyr SDK 0.16.1

**Additional context**
Used the nucleo_l011k4 out of the box and never used stm32cubeprogrammer nor its IDE

nordicjm, 2023-07-26:
> wrote 16384 bytes from file /Users/johnedwardtan/ZephyrOS/zephyrproject/zephyr/build/zephyr/zephyr.hex in 1.869062s (8.560 KiB/s)

So it wrote the file? I do not see any issues here

erwango, 2023-07-26:
> So it wrote the file? I do not see any issues here

Agree with @nordicjm, this looks like a normal flash output. I agree the last info message could sound confusing, but that's normal situation which is recovered, not an error message.

Converting to discussion for now.