zephyrproject-rtos/zephyr #60815: kernel/sched: broken thread selection when ABORTING + PENDING

vdsh-sc, 2023-07-26:
**Describe the bug**
In the test_smp_switch_torture test, the following situation sometimes occurred: there are cpu 0 and cpu 1, and, accordingly, thread 0 is the abortion triggering thread and thread 1 is the thread being aborted. In this test, at some point, the k_poll function is executed on thread 1, and thread 0 starts executing k_thread_abort. If it so happens that by the time thread 0 captured sched_spinlock in z_thread_abort, thread is still running (base.thread_state = 0), then we have the following sequence of actions:
thread0:
- lock sched_spinlock
- sets the ABORTING flag for thread 1
- sends out ipi, after which thread 1 is still running on cpu 1 - and in k_poll it reaches z_pend_curr, in which it tries to capture the still busy sched_spinlock
- becomes in waitq: thread_1->join_queue
- goes to z_swap -> z_swap_next_thread -> next_up, switches to idle_thread
- releases sched_spinlock
Now sched_spinlock grabs thread 1 in z_pend_curr and:
- goes to z_swap -> z_swap_next_thread -> next_up. At this point, before choosing an idle thread, it is worth calling end_thread(thread 1) in order to free thread 0, so a check was added. Without it, idle_thread on cpu 1 is selected - and thread 0 remains blocked

I believe it is possible to reproduce this error on any smp platform, I reproduced on qemu_riscv64_smp

This is a regression (commit d537267)

**To reproduce**
This bug rarely occurs when running tests/kernel/smp (maybe 1 out of 10 times), so I cut it into a separate test test_smp_switch_torture and looped it for 25 iterations. Usually, at 5 - 12 iterations, the test hangs for the reasons described above.

**Logs and console output**
Example of console output:
```
*** Booting Zephyr OS build 356c8cbe63ae ***
Running TESTSUITE smp
=================================================================
START - test_coop_resched_threads
  PASS - test_coop_resched_threads in 0.052 seconds
=================================================================
START - test_coop_switch_in_abort
  PASS - test_coop_switch_in_abort in 0.201 seconds
=================================================================
START - test_cpu_id_threads
  PASS - test_cpu_id_threads in 1.006 seconds
=================================================================
START - test_fatal_on_smp
E: a0: 0000000000000003 t0: 0000000000000000
E: a1: 0000000000000000 t1: 0000000000000008
E: a2: 0000000000000000 t2: 000000008000d670
E: a3: 0000000000000000 t3: 000000008000db40
E: a4: 0000000000000000 t4: 0000000080006564
E: a5: 0000000080000aba t5: 000000008000d670
E: a6: 0000000000000000 t6: 0000000000000000
E:a7:0000000000000000
E:ra:0000000080001856
E:mepc:0000000080000ac6
e:mstatus:0000000a00021880
E:
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 1
E: Current thread: 0x8000d670 (unknown)
E: a0: 0000000000000003 t0: 0000000000000000
E: a1: 0000000000000000 t1: 0000000020003f84
E: a2: 0000000000000000 t2: 000000008000d830
E: a3: 0000000000000326 t3: 0000000000000000
E: a4: 000000008000ab48 t4: 0000000080006564
E: a5: 000000008000a2f0 t5: 000000008000d830
E: a6: 0000000000000008 t6: 0000000000000000
E:a7:1ffffffffffffff
E:ra:0000000080001222
E:mepc:0000000080000ac6
e:mstatus:0000000a00021880
E:
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Current thread: 0x8000d830 (test_fatal_on_smp)
  PASS - test_fatal_on_smp in 0.105 seconds
=================================================================
START - test_get_cpu
  PASS - test_get_cpu in 0.051 seconds
=================================================================
START - test_inc_concurrency
type 0: cnt 60000 spend 42 ms
type 1: cnt 60000 spend 2595 ms
type 2: cnt 60000, spend 2582 ms
  PASS - test_inc_concurrency in 5.220 seconds
=================================================================
START - test_preempt_resched_threads
  PASS - test_preempt_resched_threads in 0.523 seconds
=================================================================
START - test_sleep_threads
  PASS - test_sleep_threads in 1.009 seconds
=================================================================
START - test_smp_coop_threads
  PASS - test_smp_coop_threads in 0.563 seconds
=================================================================
START - test_smp_ipi
cpu num=2 PASS - test_smp_ipi in 0.326 seconds
=================================================================
START - test_smp_release_global_lock
  PASS - test_smp_release_global_lock in 0.021 seconds
=================================================================
START - test_smp_switch_torture                                                                                            <<< Hangs there
```

**Environment:**
  - OS: Linux
  - Toolchain: Zephyr SDK
  - Commit SHA: 356c8cbe63ae (Zephyr 3.4 Release)

I also attach an archive with a cut test for playback
[cutted_smp_test.tar.gz](https://github.com/zephyrproject-rtos/zephyr/files/12169810/cutted_smp_test.tar.gz)

vdsh-sc, 2023-07-26:
Pull request for this issue: https://github.com/zephyrproject-rtos/zephyr/pull/60740

nordicjm, 2023-07-26:
> Pull request for this issue: #60740

When making PRs, can you put `Fixed #<issue_id>` into the github message? That way it automatically links the issues

cfriedt, 2023-07-26:
@vdsh-sc - can you update the description with other important information such as
* The `west` or `twister` command used for reproducing the issue using an upstream, `qemu`-based board

The reason we suggest `qemu`-based boards for repro-recipes is due to the ever-shrinking likelihood that the reviewers have the same physical board.

andyross, 2023-07-26:
OK, I think I see it.  It's a good catch.  The case that fails is where (see if this matches your understanding):

1. We asynchronously abort a thread from another CPU, triggering an IPI
2. The thread being abort enters a context switch of its own **before** its CPU handles the IPI (see note below)
3. The IPI lands, but never finalizes the aborted thread (the `end_thread()` function) because the thread that was interrupted is now some other thread.

And you're right, we need symmetric handling for cooperative context switch because it can race with an incoming abort.  This looks correct to me, as long as I'm not misunderstanding something.

Note: This seems like it would be an **outrageously** small race on real hardware?  Basically the bug is that the IPI (which is sent with the scheduler lock held) might take so long to be delivered that the other CPU has time to grab the lock for a cooperative context switch.  But an IPI on hardware generally latches an interrupt synchronously, or within a few cycles at most.  Whereas it's **hundreds** of cycles or more of software from "arch_sched_ipi()" to "k_spin_unlock()" on the originating CPU.  The target CPU's ISR is basically guaranteed to be spinning for the lock already by the time we release it.  It works on qemu because "IPIs" there are unix signals.

PR #60740 (2023-08-01): kernel/sched: fix thread selection when ABORTING + PENDING
In commit d537267f, the check on thread abortion was moved from next_up to z_get_next_switch_handle. However, next_up is also called from z_swap_next_thread, so the check on thread abortion is now missing there. This sometimes caused the thread to be stuck in ABORTING + PENDING state during the test_smp_switch_torture in test/kernel/smp

To avoid such cases in the future, it is worth leaving the check in next_up

Fixes #60815
closed 2023-08-01 by carlescufi