zephyrproject-rtos/zephyr #60817: Waring "Macro is deprecated" when use "FLASH_AREA_ID()" in the STM32

lnhao, 2023-07-26:
Dear community,

Zephyr version: **v3.3.0**
Zephyr SDK: **zephyr-sdk-0.16.0**

I need to store some parameter in stm32 internal flash.
In board's DTS file:
```
&flash0 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;
		 boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x00000000 DT_SIZE_K(48)>;
		};
		slot0_partition: partition@C000 {
			label = "image-0";
			reg = <0x0000C000 DT_SIZE_K(436)>;
		};
		slot1_partition: partition@79000 {
			label = "image-1";
			reg = <0x00079000 DT_SIZE_K(436)>;
		};
		user_set_partition: partition@ef000 {
			label = "user-set";
			reg = <0x000ef000 DT_SIZE_K(4)>;
		};
	};
};
```
In the project code:
```
int Init_user_setting_flash(void)
{
    int rc = 0;
    rc = flash_area_open(FLASH_AREA_ID(user_set), (const struct flash_area**)&fa_user_set);
    if(rc != 0)
        LOG_ERR("error fa_user_set %d.", rc);
    return rc;
}
```
The project has waring as following when I build it.

> /home/alexliao/git/zephyrproject/stm32wb55/lib/flash_lib/user_setting.c: In function 'Init_user_setting_flash':
> /home/alexliao/git/zephyrproject/stm32wb55/lib/flash_lib/user_setting.c:263:13: **warning: Macro is deprecated**
>   263 |     rc = flash_area_open(FLASH_AREA_ID(user_set), (const struct flash_area**)&fa_user_set);

Can this waring be removed?

Thanks & Br,
Alex

nordicjm, 2023-07-26:
You are using a deprecated macro, see https://docs.zephyrproject.org/latest/releases/release-notes-3.2.html#deprecated-in-this-release

Closing, as user issue

closed 2023-07-26 by nordicjm

lnhao, 2023-07-27:
> You are using a deprecated macro, see https://docs.zephyrproject.org/latest/releases/release-notes-3.2.html#deprecated-in-this-release
> 
> Closing, as user issue

thank you very much for your support. I use macro **FIXED_PARTITION_ID** deleted this waring.