zephyrproject-rtos/zephyr #60822: net: IP Fragmentation and checksum offload

krish2718, 2023-07-26:
**Describe the bug**
When IP fragmentation is enabled and if a driver support TCP/IP checksum offload then all packets will be dropped by the driver (or device) as checksum should be calculated before fragmentation and verified after re-assembly. This only works when fragmentation/re-assembly is also offloaded to driver.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable either `NET_IPV4_FRAGMENT` or `NET_IPV6_FRAGMENT`
2. Choose a networking device with checksum offload support (I have used a downstream device)
3. `net ping <IP> -s 1473` to trigger IP fragmentation
4. See error

**Expected behavior**
Offloaded ping should work. (Applicable for UDP/TCP as well)

**Impact**
Its a showstopper.

**Logs and console output**
NA

**Environment (please complete the following information):**
 - OS: Zephyr
 - Toolchain: Zephyr SDK
 - Downstream fork

**Additional context**
Linux has couple of solutions for this problem: See https://elixir.bootlin.com/linux/latest/source/include/linux/skbuff.h#L97

1. Per-packet checksum decision
    - Each packet has a field to indicate whether driver/HW has calculated the checksum or not, so, for this case Driver can mark as `CHECKSUM_NONE` to fallback to SW checksum
2. Leverage HW for payload
    - RX case: `CHECKSUM_COMPLETE` HW does a 1's complement checksum for the payload (as submitted to networking stack) and stack can use that to derive other checksums by basic arithmetic like deleting sub-checksums from the final one and finally fold it and verify
    - TX case: AFAIK, this is still a problem in Linux, as device would need to loop through all frags and update checksum in packet.

For now checksum offload is disabled when IP fragmentation is enabled, but we need to extend existing checksum capabilities in Zephyr to cope with such cases.

nordicjm, 2023-07-26:
It's been a long time since I've read about/used this (back when it was called chimney support) but from memory there's also another case whereby the device supports offloaded checksums but the implementation is bad so it should not use it and continue to use software implementations, the prime name that comes to mind is realtek, so that is something to beware of with this also.

krish2718, 2023-07-26:
The first troubleshooting step in any networking packet drops is to disable `checksum offload` :), too many protocols/corner cases to handle.

PR #64613 (2023-11-08): net: ip: Fix checksum calculation/validation for fragmented packets
In case HW checksum offloading was enabled on an interface, the checksum calculation/verification was broken for IP fragmented packets. This PR fixes this:
  * Always calculate/verify checksum for IP fragmented packets, regardless of the HW offloading capability
  * `chksum_done` flag is introduced, to indicate checksum status on the net_pkt, which can be used by drivers to determine whether checksum calculation is needed (unless they detect IP fragments automatically, like `eth_stm32_hal.c`, which has been used to verify the  [... cut]
closed 2023-11-08 by fabiobaltieri