zephyrproject-rtos/zephyr #60845: LVGL Sample Does Not Work With STM32H747I-DISCO Board

nluckblurpd, 2023-07-26:
I am attempting to run the LVGL sample application with the STM32H747i-Disco board. I follow the steps on the Zephyr board page [https://docs.zephyrproject.org/latest/boards/arm/stm32h747i_disco/doc/index.html]:
Build:
`west build -b stm32h747i_disco_m7 samples/drivers/display -- -DSHIELD=st_b_lcd40_dsi1_mb1166`
Flash:
`west flash`
Both building and flashing report success.
During execution, the display backlight comes on, but the unit just sits there and reboots. No graphics are displayed on the screen.

I activated the console output to usuart1 which directs to the USB CDC on the dev kit. The console output indicates a boot failure:
**[00:00:00.032,000] <err> otm8009a: ID 0x0 (should 0x40)
[00:00:00.032,000] <err> otm8009a: Panel ID check failed! (-22)**
Therefore, it seems the sample cannot initialize the DSI controller on the display module.
I have tried two different dev kits, and get the same results on each, therefore I am fairly certain it is not a hardware issue.

Could the LVLG sample be made to work with the dev kit?
Am I more than willing to help investigate and fix the issue if anyone can point me in the right direction.

FRASTM, 2023-07-27:
Testing on my  stm32h747 disco kit :

```
uart:~$ help
Please press the <Tab> button to see all available commands.
You can also use the <Tab> button to prompt or auto-complete all commands or it.
You can try to call commands with <-h> or <--help> parameter for more informati.

Shell supports following meta-keys:
  Ctrl + (a key from: abcdefklnpuw)
  Alt  + (a key from: bf)
Please refer to shell documentation for more details.

Available commands:
  clear    :Clear screen.
  device   :Device commands
  devmem   :Read/write physical memory
            Usage:
            Read memory at address with optional width:
            devmem address [width]
            Write memory at address with mandatory width and value:
            devmem address <width> <value>
  help     :Prints the help message.
  history  :Command history.
  kernel   :Kernel commands
  log      :Commands for controlling logger
  resize   :Console gets terminal screen size or assumes default in case the
            readout fails. It must be executed after each terminal width change
            to ensure correct text display.
  retval   :Print return value of most recent command
  shell    :Useful, not Unix-like shell commands.
uart:~$ 
```
[image]

nluckblurpd, 2023-07-27:
Thanks for checking on it. I made the changes to proj.conf you indicated above. Still, it doesn't run on my two dev kits. On each, I get the following console output:
```
[00:00:00.032,000] <err> otm8009a: ID 0x0 (should 0x40)
[00:00:00.032,000] <err> otm8009a: Panel ID check failed! (-22)
*** Booting Zephyr OS build zephyr-v3.4.0-1676-gd7c00ee7c25b ***
```

I wonder if you have an older dev kit with different hardware? I just ordered my two from Mouser last week. I can't think of anything else that might be different. Both my display modules indicate a revision of MB1166-A09, whereas yours indicates MB1166-A03 on the sticker.

Do you have any suggestions on how I can debug further?

FRASTM, 2023-07-27:
yes, propably old LCD shield : mine is MB1166 -A03

nluckblurpd, 2023-07-27:
I searched through my spare parts bin and found an A03 display like yours. It works! So, there is some hardware difference between A03 and A09. But, the base LVGL sample runs correctly with the A03 display.

A secondary issue: I would like to turn on ChromeArt acceleration. I activated the proper (I think) define in lv_conf.h:
```
/*Use STM32's DMA2D (aka Chrom Art) GPU*/
#define LV_USE_GPU_STM32_DMA2D 1
#if LV_USE_GPU_STM32_DMA2D
    /*Must be defined to include path of CMSIS header of target processor
    e.g. "stm32f7xx.h" or "stm32f4xx.h"*/
    #define LV_GPU_DMA2D_CMSIS_INCLUDE "stm32h7xx.h"
#endif
```
It builds just fine. But the video output is incorrect:
[image]
Could you give me any pointers on how to fix that issue? I assume it's some ChromeArt setup issue, but I don't know where to look in Zephyr.

FRASTM, 2023-07-27:
I am not really aware of this dma2d of the stm32H747
Even if the application is building, I do not see where this dma2d node is defined and if it should be neither. There is a register bank  @ 0x52001000, and clock enable bit (4) on RCC_AHB3ENR, IRQ 90. That could be a first step to declare the peripheral in the dtsi.