zephyrproject-rtos/zephyr #60868: Clarify the loation for Zephyr specific LVGL code

fabiobaltieri, 2023-07-27:
Currently the bulk of the LVGL Zephyr support code lives in the lvgl mirror repository: https://github.com/zephyrproject-rtos/lvgl/tree/zephyr/zephyr, this has been moved form the main upstream repository as part of the 8.1 migration https://github.com/zephyrproject-rtos/lvgl/pull/15.

My understanding is that was a a path towards upstreaming that code on the main LVGL repo, but that hasn't happened.

There's more development work ongoing on LVGL (@faxe1008 @danieldegrasse), and I think that having the code in the module is making thing unnecessarily harder by requiring the extra PR for synchronizing the code.

Should we move the code back to the main repository in bulk for the time being and reconsider upstreaming it on a later stage?

CC: @nashif @gmarull @carlescufi @jfischer-no

jfischer-no, 2023-07-27:
> Currently the bulk of the LVGL Zephyr support code lives in the lvgl mirror repository: https://github.com/zephyrproject-rtos/lvgl/tree/zephyr/zephyr, this has been moved form the main upstream repository as part of the 8.1 migration [zephyrproject-rtos/lvgl#15](https://github.com/zephyrproject-rtos/lvgl/pull/15).
> 
> My understanding is that was a a path towards upstreaming that code on the main LVGL repo, but that hasn't happened.

Maybe some parts, but not the interface to our APIs.

> There's more development work ongoing on LVGL (@faxe1008 @danieldegrasse), and I think that having the code in the module is making thing unnecessarily harder by requiring the extra PR for [...]

I think the customizing/interfacing part to our subsystems/driver APIs can be moved to ./modules (similar to e.g. .modules/littlefs)

fabiobaltieri, 2023-07-27:
> I think the customizing/interfacing part to our subsystems/driver APIs can be moved to ./modules (similar to e.g. .modules/littlefs)

Cool, so you are suggesting that we should move this stuff: https://github.com/zephyrproject-rtos/lvgl/tree/zephyr/zephyr back to https://github.com/zephyrproject-rtos/zephyr/tree/main/modules (`lvgl` dir) and just do https://docs.zephyrproject.org/latest/develop/modules.html#module-integration-files-in-module in the module, right? I'd like that too.

jfischer-no, 2023-07-27:
> Cool, so you are suggesting that we should move this stuff: https://github.com/zephyrproject-rtos/lvgl/tree/zephyr/zephyr back to https://github.com/zephyrproject-rtos/zephyr/tree/main/modules [...]

Yes, and take a closer look at it. e.g why is there https://github.com/zephyrproject-rtos/lvgl/blob/zephyr/zephyr/lvgl_fs.c and what it does...

fabiobaltieri, 2023-07-28:
@faxe1008 would you have cycles to move the files back into the main repository under `modules/lvgl`? Then the ones in m/zephyrproject-rtos/zephyr/pull/60867 would go there as well and any other effort won't require the double PR anymore.

faxe1008, 2023-07-28:
@fabiobaltieri just to clarify, you are talking about moving the files situated here: [https://github.com/zephyrproject-rtos/lvgl/tree/zephyr/zephyr](https://github.com/zephyrproject-rtos/lvgl/tree/zephyr/zephyr) to [https://github.com/zephyrproject-rtos/zephyr/tree/main/modules](https://github.com/zephyrproject-rtos/zephyr/tree/main/modules)?

Sure, can do that after my vacation (2 weeks).

fabiobaltieri, 2023-07-28:
> @fabiobaltieri just to clarify, you are talking about moving the files situated here: https://github.com/zephyrproject-rtos/lvgl/tree/zephyr/zephyr to https://github.com/zephyrproject-rtos/zephyr/tree/main/modules?
> 
> Sure, can do that after my vacation (2 weeks).

Yeah, cool, no pressure, have a nice break!

gmarull, 2023-08-02:
I think it would be nice to upstream the code, so that one can theoretically pick any LVGL version regardless of Zephyr version (provided both are compatible). Having to upgrade complex LVGL applications just because I need to upgrade Zephyr is not great (I've been through this).

jfischer-no, 2023-08-02:
> I think it would be nice to upstream the code, so that one can theoretically pick any LVGL version regardless of Zephyr version (provided both are compatible). Having to upgrade complex LVGL applications just because I need to upgrade Zephyr is not great (I've been through this).

This is not LVGL project code, but customization of our APIs and subsystems, which means that once up-streamed it will probably always be broken with the current Zephyr tree. It will also increase upmerge and maintenance effort, like we have the situation with MCUboot.

gmarull, 2023-08-02:
> This is not LVGL project code, but customization of our APIs and subsystems, which means that once up-streamed it will probably always be broken with the current Zephyr tree. It will also increase [...]

Well, this assumes Zephyr APIs are very unstable, which in part is true because we have quite a few _toy-like_ APIs. But TBH, the LVGL glue code has not suffered major changes over time (because graphics support has not evolved either). LVGL can always claim support up to Zephyr vX.Y, and upgrade when the maintainer has enough time.

gmarull, 2023-08-02:
I see there's some Zephyr related stuff upstream already https://github.com/lvgl/lvgl/blob/master/env_support/cmake/zephyr.cmake
https://github.com/lvgl/lvgl/blob/master/CMakeLists.txt#L11

Added by the initial LVGL Zephyr maintainer: https://github.com/lvgl/lvgl/commit/ed5f91ab7261e2610e9cda6fe078d292e3ead5bb

fabiobaltieri, 2023-08-02:
Between input and https://github.com/zephyrproject-rtos/zephyr/pull/60840 there's probably going to be quite some work on LVGL Zephyr support, I'm thinking we may be better off getting that done and stabilized in the main repo and think again about upstreaming once that's stabilized.

gmarull, 2023-08-02:
> Between input and #60840 there's probably going to be quite some work on LVGL Zephyr support, I'm thinking we may be better off getting that done and stabilized in the main repo and think again about upstreaming once that's stabilized.

sounds reasonable

fabiobaltieri, 2023-08-15:
Work is ongoing to move this back to the main repo, closing this one.

closed 2023-08-15 by fabiobaltieri