zephyrproject-rtos/zephyr #60890: Littlefs: No space error

karthi012, 2023-07-27:
**Describe the bug:**
I am working on littlefs with esp32_devkitc_wrover.

Esp32 wrover kit has 8MB of flash. I have only mounted 512KB for storage partition.
I have littlefs sample, which it append data to file continuously. On the every write on the file,
i had to update some information which is present in the start of the file of 4 bytes. By doing this i have
encountered the "No space error" from littlefs, even though i have enough space for further write in the
flash partition.I can write upto around 256KB only. Here attach logs for reference.

**Logs**:
`*** Booting Zephyr OS build v3.4.0-rc3-1742-g7bfe0967879e ***                                                                                                 
Sample program to r/w files on littlefs                                                                                                                       
Area 4 at 0x250000 on flash-controller@3ff42000 for 524288 bytes                                                                                                                                                                                                                                
I: LittleFS version 2.5, disk version 2.0                                                                                                                     
I: FS at flash-controller@3ff42000:0x250000 is 128 0x1000-byte blocks with 512 cycle                                                                          
I: sizes: rd 16 ; pr 16 ; ca 64 ; la 32                                                                                                                       
E: WEST_TOPDIR/modules/fs/littlefs/lfs.c:1234: Corrupted dir pair at {0x0, 0x1}                                                                               
W: can't mount (LFS -84); formatting                                                                                                                          
I: /lfs mounted                                                                                                                                               
/lfs mount: 0                                                                                                                                                 
/lfs: bsize = 16 ; frsize = 4096 ; blocks = 128 ; bfree = 126 

file size  258048
 /lfs: bsize = 16 ; frsize = 4096 ; blocks = 128 ; bfree = 4294967295
E: WEST_TOPDIR/modules/fs/littlefs/lfs.c:589: No more free space 236
E: file seek error (-28)
E: FAIL: seek /lfs/pattern.bin: -28
`

**Expected behavior**
Able to write whole 512KB.

Laczen, 2023-07-28:
@karthi012, thank you for reporting this. The error you are seeing could be an littlefs internal problem that only appears in your specific use case (modifying something at the start of the file and then extending it). I propose you post this issue on the littlefs github issues (https://github.com/littlefs-project/littlefs/issues). To validate if it is caused by the specific use method you could try what happens when you don't modify the file at the start.

parthitce, 2023-07-31:
https://github.com/littlefs-project/littlefs/issues/816 Likely the same problem.

dkalowsk, 2023-09-21:
Took a little while to get to this, but I believe I have reproduced the issue locally on a different board with some QSPI flash.  If the issue is the same, I'm going to agree with @parthitce that this looks exactly like the issue in the LittleFS project and not much we can do to fix it.

closed 2023-12-05 by github-actions[bot]