zephyrproject-rtos/zephyr #60904: spi_nor Erase operation selection is excessively conservative

Crzyrndm, 2023-07-28:
Summary: SPI_NOR driver is selecting the smallest erase in in many situations it doesn't need to

SPI NOR FLASH devices Often have multiple erase sizes. 4k, 32k and 34k are very typical.
When `spi_nor_erase` ([link](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/spi_nor.c#L737)) is invoked part of the loop attempts to pick the best erase operation for the current address and remaining size ([link](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/spi_nor.c#L772-L783))

I was erasing parts of a 512MiB NOR device today and observed extremely slow erase speeds. I found that many of the erase operation where using only the 4kiB erase even when the address/remainder was suitable for a much larger operation.

I extracted the main parts of the op selector in order to confirm my suspicions. This can be seen [here](https://godbolt.org/z/9jdTzjh5v)

What I would expect to see is that 4k erases are used until 32kB is reached, 1 32k to get to 64k then 64k erases until the last 16 kB. What actually happens is that only 4k erases are used.

The problem is this [line](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/spi_nor.c#L778). For a larger erase op to be chosen, either the start or end address must be aligned with that op size. In my example above, the start and end are only 16kB aligned so this never occurs

The correct check would be `size >= (1 << etp->exp)` ([updated example](https://godbolt.org/z/Wrr5h6M9Y). Change is to line 30)

Laczen, 2023-07-28:
@Crzyrndm, the check should indeed be that the size is larger than the erase-block. Could you provide a PR ?

Crzyrndm, 2023-07-29:
> @Crzyrndm, the check should indeed be that the (remaining) size is larger or equal to the erase size. Could you provide a PR ?

PR has been opened. I think I got all the neccesary conventions covered but feel free to just fix it if more needs doing

commit ea2dd9fc (2023-08-04): drivers: flash: spi_nor: select largest valid erase operation

The spi_nor erase op selection was based on the alignment of the end of
the region to be erased. This prevented larger erase operations being
selected in many cases

Closes #60904
commit 5e02c0bd (2023-08-22): drivers: flash: spi_nor: select largest valid erase operation

The spi_nor erase op selection was based on the alignment of the end of
the region to be erased. This prevented larger erase operations being
selected in many cases

Closes #60904

(cherry picked from commit ea2dd9fc65588d40a31b93b7a2f0629c9ed0b354)
commit a042ba2b (2023-09-05): drivers: flash: spi_nor: select largest valid erase operation

The spi_nor erase op selection was based on the alignment of the end of
the region to be erased. This prevented larger erase operations being
selected in many cases

Closes #60904

(cherry picked from commit ea2dd9fc65588d40a31b93b7a2f0629c9ed0b354)
PR #61151 (2023-11-17): [Backport v2.7-branch] drivers: flash: spi_nor: select largest valid erase operation
Backport ea2dd9fc65588d40a31b93b7a2f0629c9ed0b354 from #60943.

Fixes: #60904
closed 2023-08-04 by carlescufi