zephyrproject-rtos/zephyr #60924: net: tcp: Potential net buf leak when receiving RST packets

ssharks, 2023-07-28:
**Describe the bug**
When a connection is being terminated by the other side, there could be a packet in the retransmission queue, that is not properly freed in all possible code paths,

**To Reproduce**
Very hard, build a connection with a host, make it terminate the connection by sending a RST packet and see that after number of iterations you are running out of net bufs.

**Expected behavior**
At the termination of a connection, all netbufs are properly cleaned up.

**Impact**
Has the potential to hang up a complete network stack

**Additional context**
See for example lines: https://github.com/zephyrproject-rtos/zephyr/blob/19f33b8ecb2667c267d527bf90aee97bc0940aa6/subsys/net/ip/tcp.c#L2117

rlubos, 2023-08-01:
>there could be a packet in the retransmission queue, that is not properly freed in all possible code paths,

Could you elaborate what packet are you referring to? The retransmission queue is cleared whenever connection is being dereferenced (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/tcp.c#L455C2-L455C22)

ssharks, 2023-08-03:
To be honost I did not spot this code path. Further investigation revealed that the device was running out of net contexts, not net bufs. The symptoms can be pretty similar however.

closed 2023-08-03 by ssharks