zephyrproject-rtos/zephyr #60929: nucleo_h745zi_q_m7 won't flash hello_world

codewitch-honey-crisis, 2023-07-28:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/60927

<div type='discussions-op-text'>

<sup>Originally posted by **codewitch-honey-crisis** July 28, 2023</sup>
I've followed the docs here on 4 brand new nucleo_h745_q_m7 boards and get the same result on each

https://docs.zephyrproject.org/latest/boards/arm/nucleo_h745zi_q/doc/index.html

```
xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2022-09-01-17:58)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
init
Info : STLINK V3J5M2 (API v3) VID:PID 0483:374E
Info : Target voltage: 3.304000
Error: init mode failed (unable to connect to the target)
```
This is either a bug in the documentation or a bug in the flashing system. I'm not sure where it belongs.
The expected result is following the documentation should lead to a successful flash.
</div>

Muddasir-Shakil, 2023-07-29:
Hi there.
I had the similar issue with it. The problem is caused by mis configuration of power settings flags. If the hardware setting mismatched the setting in firmware then you have these issues. 
You would need to recover the board in that case. The instructions can be found in the one of the guide for this baord, something related to power setting.
[image]

i don’t remember what is the default hardware config was but it can be either LDO or SMPS.
You can set this in you prj. Conf so that you everytime you flash, you still be able to reflash otherwise you need to recover the board again to be able to flash. May be you can open the PR for putting this thing in Docs

codewitch-honey-crisis, 2023-07-29:
Thank you for that! Since it did this on brand new, just unboxed boards I think a link to this information needs to be added to the documentation for these boards. Otherwise it becomes a rabbit hole.

Muddasir-Shakil, 2023-07-29:
For it happened after the first flash and next power cycle. 
If it works for you can you please creat a PR to update the docs atleast mention section on how to recover from this

codewitch-honey-crisis, 2023-07-29:
I'll do my best. I'm kind of a noob with PRs. I know how to do it by forking an entire project but I'll have to work out how to do it for just the docs. Sorry I'm still getting my sea legs with the git interface. Let me try this on another fresh board. I have more. It was a long day when I posted that and I want to make sure I get the steps right going from new. I don't know exactly what I did to end up where I was at the other day. =/ Your experience with these boards is different than what I remember though, so I'm going to try to reproduce again before I update the docs. Thanks for your help.

closed 2023-07-29 by codewitch-honey-crisis

codewitch-honey-crisis, 2023-07-30:
Hey @Muddasir-Shakil , you said "i don’t remember what is the default hardware config was but it can be either LDO or SMPS.
You can set this in you prj. Conf" 

I've been looking for that prj.conf setting, and I can't find it. I don't suppose you know where to look, do you? If not, no worries.

Muddasir-Shakil, 2023-07-30:
It is a kconfig option you can find it in the default config of the baord