zephyrproject-rtos/zephyr #60950: MS5837-30BA01-50 sensor not identified and compensated correctly after upgrading to 2.4.0

FlsZen, 2023-07-29:
**Describe the bug**
After upgrading from 2.3.0 to 2.4.0, my MS5837-30BA01-50 sensors are not identified correctly by the new code. Their factory data is 0x00 (after masking with 0x7f)—the same as a MS5837-02BA01. As a result, the code uses the `ms5837_compensate_02` instead of the `ms5837_compensate_30`.

I believe the "-01" is set at the factory to 0x00 regardless of whether it is 02BA or 30BA. The involved datasheets call out MS5837_02BA21 as being 0x15 (21 dec) and MS5837_30BA26 as being 0x1a (26 dec), so it seems they are simply storing that portion of the part number without respect to the pressure rating.

I'm using Visual Studio Code with nRF52832 modules.

I confirmed all of the prom values read in 2.3.0 match the values read in 2.4.0, so I doubt anything is wrong with communication.
My workaround is of course to modify ms5837.c to use `ms5837_compensate_30` for type 0x00.

While I haven't performed a git bisect, support for the 02BA sensors was added in d485824. Up until that point it wasn't identifying the sensor's type at all and didn't have the conversion code for the 02BA which it is now trying to call.

**To Reproduce**
Steps to reproduce the behavior:
1. Build sensor example code
2. Run sensor example code with attached MS5837-30BA01-50 sensor
3. Observe pressure reading

**Expected behavior**
Pressure reading should be near 100 kPa. (Current observation is around 5 kPa.)

**Impact**
This is an annoyance for myself, but possibly very confusing for those who are starting with the latest code and a 30BA sensor.

**Environment**
 - OS: macOS and Windows
 - Visual Studio Code, Zephyr SDK, nRF52832
 - 2.4.0

FlsZen, 2023-07-29:
In addition to my workaround, I also noticed that the pressure was 1/10 of what it should be. I suspect some other change (b4c31fd) may be involved in that, but I'm not sure that I should file a separate bug report since it would be based on my modification, not the current code.

jhedberg, 2023-08-01:
@marc-cpdesign any thoughts on this one?

fabiobaltieri, 2023-08-08:
@FlsZen hey could you send a PR with the fix? Then we can followup on that.

marc-cpdesign, 2023-08-09:
Hi

The changes I added in (d485824)  do rely on the 'factory id' (prom0 data, bits [11:5]) being unique or at least related to which compensation function should be used. At the time the only 30BA variant was the 30BA26. (I also couldn't just now find any data sheet that had any other variant for the 30BA..)

If this isn't the case (ie different variants have the same factory id) then the driver will need some other way of knowing which compensation method to use, eg a devicetree property. 

@FlsZen  The changes in (https://github.com/zephyrproject-rtos/zephyr/commit/b4c31fd4386a1a38a2dad805da88e60ee1ee11f6) were to make the driver produce values in kPa, which is the units that the zephyr sensor interface uses for 'pressure'; so yes, the readings will change by 1/10 from when you used the older version. In this instance I thought it was better to make this change so that the interface is consistent, even if it breaks existing usage.
Its been a while since I did this, but at the time the values were reading correctly for the  02BA, I didn't have any other variants to test with.

Cheers
Marc

FlsZen, 2023-08-14:
> @FlsZen hey could you send a PR with the fix? Then we can followup on that.

@fabiobaltieri Sure. I'm still very new to Zephyr, but it seems straightforward to specify the sensor submodel in a devicetree property. Feel free to nitpick.

@marc-cpdesign Thanks for the additional info Marc.

The MS5837-30BA datasheet [rev. C3 06/2023](https://www.te.com/commerce/DocumentDelivery/DDEController?Action=showdoc&DocId=Data+Sheet%7FMS5837-30BA%7FB1%7Fpdf%7FEnglish%7FENG_DS_MS5837-30BA_B1.pdf%7FCAT-BLPS0017) lists specs for 30BA01 and 30BA28, although only 30BA01 is listed as orderable at the end of the datasheet. The PROM example mentions 30BA26 which isn't mentioned elsewhere. 

The MS5837-02BA datasheet [rev. REV A8 12/2019](https://www.te.com/commerce/DocumentDelivery/DDEController?Action=showdoc&DocId=Data+Sheet%7FMS5837-02BA01%7FA8%7Fpdf%7FEnglish%7FENG_DS_MS5837-02BA01_A8.pdf%7FCAT-BLPS0059) lists an orderable part of 02BA01, and the PROM example lists 02BA01 along with 02BA21.

It seems that my 30BA01 part has the same factory id (0x00) as the 02BA01 part, so we need a different way to differentiate. I liked your idea of using a devicetree property, so that's what I went with. The new submodel enum defaults to 02BA as I presume that is the more common pressure range. Since I'm using the 30BA, I include the new property in my overlay as:
```
ms5837@76 {
    compatible = "meas,ms5837";
    reg = <0x76>;
    submodel = "30ba";
};
```

The 30BA compensation algorithm was emitting pressure in `mbar * 10` instead of `mbar * 100`, so I added a conversion just before the end of that function so it will be in `mbar * 100` to match 02BA compensation.

FlsZen, 2023-08-14:
Ok, well I made a bit of a mess on the way, but the PR has been created.

closed 2023-10-29 by github-actions[bot]