zephyrproject-rtos/zephyr #60975: mktime incorrect on 2nd Jan 1970

matt-wood-ct, 2023-07-31:
**Describe the bug**
`mktime` is returning an incorrect value for 2nd Jan 1970.
While writing unit tests for a segment of code I found that mktime appears to incorrectly convert the date above into 82800 when the correct value is 86400, all my other test case values are working correctly it is only this low value that appears to fail for me.
I am running with NEWLIB LIB C enabled

**To Reproduce**
Steps to reproduce the behavior:
1. Open Hello World sample
2. Edit main to include:
```
#include <time.h>

    struct tm dmy;
    memset(&dmy, 0, sizeof(struct tm));
    dmy.tm_year = 70;
    dmy.tm_mon = 0;
    dmy.tm_mday = 2;
    printf("time: %ul\r\n", mktime(&dmy));
```
3. Edit proj,conf to include `CONFIG_NEWLIB_LIBC=y` 
4. Build and run for `native_posix`
5. Output is `time: 82800`

**Expected behavior**
Output should be  `time: 86400`

**Impact**
Could affect behaviour of all projects utilising the standard time libraries.

**Environment (please complete the following information):**
 - WSL: Ubuntu 20.04.6 LTS
 - Zephyr SDK version 0.16.0
 - Zephyr version v3.4.0

nordicjm, 2023-07-31:
It is out by an hour, so I assume that would be because of DST and this being a local timezone.

matt-wood-ct, 2023-07-31:
> It is out by an hour, so I assume that would be because of DST and this being a local timezone.

the structure is memset to 0 so DST is set to zero. It should be UTC as far as I'm aware.

matt-wood-ct, 2023-10-23:
Why has this closed?

aescolar, 2023-12-21:
@matt-wood-ct please note that native_posix does not support NEWLIB (it only supports the host C library). 
```
warning: The choice symbol NEWLIB_LIBC (defined at lib/libc/Kconfig:94) was selected (set =y), but
EXTERNAL_LIBC (defined at lib/libc/Kconfig:111) ended up as the choice selection. See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_NEWLIB_LIBC and/or look up NEWLIB_LIBC in
the menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration
Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful too.
```
So you are building with and calling into the host C library `mktime()`, not into the newlib one.
(it is a bit out of scope to figure out here why the host mktime does not do what you expect).

native_sim can be used with embedded C libraries, but newlib does not come with source in the zephyr provided repos so that one is not an easy option.
You can instead chose to use `CONFIG_PICOLIBC` and you will get the value you expect:
`cmake -DBOARD=native_sim ../samples/hello_world/ -GNinja -DCONFIG_PICOLIBC=y && ninja && zephyr/zephyr.exe`