zephyrproject-rtos/zephyr #60990: drivers: uart: Add API to disable RX

nordic-krch, 2023-07-31:
**Is your enhancement proposal related to a problem? Please describe.**
When asynchronous API is not used in UART device then RX is always on (started during the initialization). RX being always on increases power consumption. If user wants to reduce the power consumption he would like to force RX to be disabled. Currently, the only way to do that is to call `pm_device_action_run()` and suspend the UART. However, this API is not recommended and might be deprecated in the future. With introduction of runtime device power management (PM) it's device responsibility to get to the lowest power state when not used. It would be better to provide API which disables RX and rely on runtime PM that it will get to the lowest power state.

**Describe the solution you'd like**
Introduce `uart_rx_ctrl(const struct device *uart, bool enable)`. If async API is enabled it returns `-ENOTSUP` otherwise it start (`enable` is true) or stops the UART (if `enable` is false). Since stopping UART may take some time we might consider providing a callback (if NULL then blocking). `uart_rx_ctrl` is proposed instead of `uart_rx_enable` because there is already such function in async API.

nordic-krch, 2023-07-31:
@gmarull @anangl @dcpleung

carlescufi, 2023-08-01:
Architecture WG:

- @gmarull states that `pm_device_action_run()` is not suitable for application-level calling due to it not being refcounted nor being part of the device PM framework per se
- @mmahadevan108 asks if we could add something int he power management APIs instead of having it in the UART API. @gmarull says that the "TURN _ON" and "TURN_OFF" is for power domains and not for devices
- @galak agrees that this might not be at the right level, and it solves only a small part of the problem
- @gmarull states that the problem is the UART API itself, which should be fixed to avoid having multiple variants
- @bjarki-trackunit states that the shutting down would be for both the TX and RX paths, but @nordic-krch counters that APIs like `uart_poll_in()` make assumptions about
- @bjarki-trackunit says that we could use `get()` and `put()` instead of `pm_device_action_run()` but @gmarull states that the hook should decide whether to turn off RX or not

nordic-krch, 2023-08-03:
> Need to make sure the API is clear about if called and than one calls poll_in() or what happens if any other RX API is called.

poll_in return `-1` if no new byte, `uart_irq_rx_ready` returns 0 when char is not ready. Imo, that should cover that.

dcpleung, 2023-08-03:
We have `rx_enable()` and `rx_disable()` on both interrupt driven API and async API. These can be combined with PM `get()/put()` to control power on RX line. We don't necessarily need to model TX/RX line separately this way.

For polling, I still think that we can use the line control API to enable/disable RX on demand.

nordic-krch, 2023-08-04:
> We have rx_enable() and rx_disable() on both interrupt driven API 

For interrupt driven API we have `uart_irq_rx_enable` which enables RX interrupt. It expects that receiver is on all the time. Calling `uart_irq_rx_enable()` just trampolines you to the UART interrupt context where you can check if there are any new bytes pending (`uart_irq_rx_ready()`) and read those bytes (`uart_fifo_read()`).

dcpleung, 2024-02-08:
Hm... come to think of it, could this be piggybacked into `rx_enable()` and `rx_disable()`? So if RX is disabled, turn off the RX line.

nordic-krch, 2024-02-13:
@dcpleung do you mean to extended functionality of `uart_rx_enable` and `uart_rx_disable` for polling/interrupt_driven APIs? It may be possible with a comment that `buf`,`len` and `timeout` parameters in `uart_rx_enable` are used only in async API and information that `uart_rx_disable` is a synchronous call for polling/interrupt_driven and asynchronous for async API.

dcpleung, 2024-02-13:
I am thinking that `uart_tx/rx_enable` should always be synchronous, and `uart_async_rx_disable` (new) would be asynchronous. A dual action function is going to confuse developers. And I guess this would be less controversial than calling PM functions in application. The async RX disable can be written to turn off the RX line at the first opportunity to do so.