zephyrproject-rtos/zephyr #61075: [mimxrt1060_evk][pm][device_runtim_api] after exectue this test, board can't connect with jtag

hakehuang, 2023-08-02:
**Describe the bug**
[mimxrt1060_evk][pm][device_runtim_api] after exectue this test, board can't connect with jtag

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1060_evk ..
3. make
4. See error

**Expected behavior**
test pass and the board remian active

**Impact**
PM cases

**Logs and console output**

```
ubuntu@ubuntu-OptiPlex-7050:/home/shared/disk/zephyr_project/zephyr_test/zephyr/tests/subsys/pm/device_runtime_api$ /home/ubuntu/erase_board_mimxrt1060_evk.sh
0000594 W Invalid coresight component, cidr=0x0 [rom_table]
0000614 I Loading /home/ubuntu/temp/mimxrt1060_evk_hello_world.elf at 0x60000000 [load_cmd]
0000693 C cannot write registers r0, r1, r2, r3, r4, r5, r6, r7, r8, r9, r10, r11, r12, psp, msp, lr, pc, xpsr, cfbp because core #0 is not halted [__main__]
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used:  zephyr-v3.4.0-1850-g66969fc9a7

**Additional context**

bisect found the issues is introduced by

```
6552250cb6151de4787ccab4a1545ffaed990f22 is the first bad commit
commit 6552250cb6151de4787ccab4a1545ffaed990f22
Author: Gerard Marull-Paretas <gerard@teslabs.com>
Date:   Wed Jul 19 12:29:42 2023 +0200

    dts: arm: nxp: move power-states to soc dts files

    CPU idle states are not board specific. This patch moves NXP idle states
    to the core SoC dts files. Board can always tweak some state parameters
    (if needed), but the definition belongs to core SoC dts files, same as
    e.g.  peripherals.

    Signed-off-by: Gerard Marull-Paretas <gerard@teslabs.com>

 boards/arm/mimxrt1064_evk/mimxrt1064_evk.dts    | 19 --------------
 boards/arm/mimxrt595_evk/mimxrt595_evk_cm33.dts | 35 -------------------------
 boards/arm/mimxrt685_evk/mimxrt685_evk_cm33.dts | 17 ------------
 dts/arm/nxp/nxp_rt10xx.dtsi                     | 16 +++++++++++
 dts/arm/nxp/nxp_rt5xx_common.dtsi               | 32 ++++++++++++++++++++++
 dts/arm/nxp/nxp_rt6xx_common.dtsi               | 14 ++++++++++
 6 files changed, 62 insertions(+), 71 deletions(-)
```

dleach02, 2023-08-02:
@gmarull any thoughts on what could be going on here before I get someone to dig into this?

gmarull, 2023-08-02:
Maybe some LP states are not well-supported, and it is now observed because they are present in all boards using the same SoC.

gmarull, 2023-08-02:
Btw, could you share the final zephyr.dts with working/non-working dts?

hakehuang, 2023-08-03:
[dts.zip](https://github.com/zephyrproject-rtos/zephyr/files/12246514/dts.zip)
@gmarull , I guess many of platform you enabled does not have wakeup source configured.

hakehuang, 2023-08-03:
all rt10xx have the same problem, lack of wakeup setting, which cause CI can not run.

gmarull, 2023-08-03:
```diff
--- zephyr.dts	2023-08-03 04:57:25.000000000 +0200
+++ zephyr_bad.dts	2023-08-03 05:10:25.000000000 +0200
@@ -309,7 +309,7 @@
 			gpio-controller;
 			#gpio-cells = < 0x2 >;
 			pinmux = < &iomuxc_gpio_sd_b1_00_gpio3_io00 >, < &iomuxc_gpio_sd_b1_01_gpio3_io01 >, < &iomuxc_gpio_sd_b1_02_gpio3_io02 >, < &iomuxc_gpio_sd_b1_03_gpio3_io03 >, < &iomuxc_gpio_sd_b1_04_gpio3_io04 >, < &iomuxc_gpio_sd_b1_05_gpio3_io05 >, < &iomuxc_gpio_sd_b1_06_gpio3_io06 >, < &iomuxc_gpio_sd_b1_07_gpio3_io07 >, < &iomuxc_gpio_sd_b1_08_gpio3_io08 >, < &iomuxc_gpio_sd_b1_09_gpio3_io09 >, < &iomuxc_gpio_sd_b1_10_gpio3_io10 >, < &iomuxc_gpio_sd_b1_11_gpio3_io11 >, < &iomuxc_gpio_sd_b0_00_gpio3_io12 >, < &iomuxc_gpio_sd_b0_01_gpio3_io13 >, < &iomuxc_gpio_sd_b0_02_gpio3_io14 >, < &iomuxc_gpio_sd_b0_03_gpio3_io15 >, < &iomuxc_gpio_sd_b0_04_gpio3_io16 >, < &iomuxc_gpio_sd_b0_05_gpio3_io17 >, < &iomuxc_gpio_emc_32_gpio3_io18 >, < &iomuxc_gpio_emc_33_gpio3_io19 >, < &iomuxc_gpio_emc_34_gpio3_io20 >, < &iomuxc_gpio_emc_35_gpio3_io21 >, < &iomuxc_gpio_emc_36_gpio3_io22 >, < &iomuxc_gpio_emc_37_gpio3_io23 >, < &iomuxc_gpio_emc_38_gpio3_io24 >, < &iomuxc_gpio_emc_39_gpio3_io25 >, < &iomuxc_gpio_emc_40_gpio3_io26 >, < &iomuxc_gpio_emc_41_gpio3_io27 >;
-			phandle = < 0x178 >;
+			phandle = < 0x17a >;
 		};
 		gpio4: gpio@401c4000 {
 			compatible = "nxp,imx-gpio";
@@ -326,7 +326,7 @@
 			gpio-controller;
 			#gpio-cells = < 0x2 >;
 			pinmux = < &iomuxc_snvs_wakeup_gpio5_io00 >, < &iomuxc_snvs_pmic_on_req_gpio5_io01 >, < &iomuxc_snvs_pmic_stby_req_gpio5_io02 >;
-			phandle = < 0x177 >;
+			phandle = < 0x179 >;
 		};
 		gpio6: gpio@42000000 {
 			compatible = "nxp,imx-gpio";
@@ -632,7 +632,7 @@
 					};
 				};
 				pinmux_swo: pinmux_swo {
-					phandle = < 0x175 >;
+					phandle = < 0x177 >;
 					group0 {
 						pinmux = < &iomuxc_gpio_ad_b0_10_arm_trace_swo >;
 						bias-disable;
@@ -2380,7 +2380,7 @@
 			port {
 				lcd_panel_out: endpoint {
 					remote-endpoint = < &lcd_panel_in >;
-					phandle = < 0x179 >;
+					phandle = < 0x17b >;
 				};
 			};
 		};
@@ -2609,7 +2609,7 @@
 				status = "okay";
 				pinctrl-0 = < &pinmux_flexpwm2 >;
 				pinctrl-names = "default";
-				phandle = < 0x176 >;
+				phandle = < 0x178 >;
 			};
 		};
 		flexpwm3: flexpwm@403e4000 {
@@ -2976,6 +2976,12 @@
 			reg = < 0x403c4000 0x4000 >;
 			status = "disabled";
 		};
+		dcp: dcp@402fc000 {
+			compatible = "nxp,mcux-dcp";
+			reg = < 0x402fc000 0x4000 >;
+			interrupts = < 0x32 0x0 >, < 0x33 0x0 >;
+			status = "okay";
+		};
 		enet2: ethernet@402d4000 {
 			compatible = "nxp,kinetis-ethernet";
 			reg = < 0x402d4000 0x628 >;
@@ -2998,6 +3004,7 @@
 			compatible = "arm,cortex-m7";
 			d-cache-line-size = < 0x20 >;
 			reg = < 0x0 >;
+			cpu-power-states = < &idle &suspend >;
 			#address-cells = < 0x1 >;
 			#size-cells = < 0x1 >;
 			mpu: mpu@e000ed90 {
@@ -3013,6 +3020,22 @@
 				pinctrl-names = "default";
 			};
 		};
+		power-states {
+			idle: idle {
+				compatible = "zephyr,power-state";
+				power-state-name = "runtime-idle";
+				exit-latency-us = < 0xfa0 >;
+				min-residency-us = < 0x1388 >;
+				phandle = < 0x175 >;
+			};
+			suspend: suspend {
+				compatible = "zephyr,power-state";
+				power-state-name = "suspend-to-idle";
+				exit-latency-us = < 0x1388 >;
+				min-residency-us = < 0x2710 >;
+				phandle = < 0x176 >;
+			};
+		};
 	};
 	sysclk: system-clock {
 		compatible = "fixed-clock";
```

This is the diff. As you can see, the main relevant difference is that power states are now defined, so `CONFIG_PM=y` does something (before it was in practice a no-op). So what I get from this is that PM is not well-supported on rt10xx MCUs. If certain states require configuration to work, this is what, I think, should happen:

1. States are disabled by default, or
2. SoC code makes sure the necessary configuration is applied to make all states work properly

The current system PM model automatically decides the best power state to choose based on its minimum residency time/wake-up latency. Maybe certain power states are not meant to be used automatically. This case is not really well-supported today (you can still do hacks with pm_state_force, but...)

hakehuang, 2023-08-03:
@dleach02 , if in this case, shall we disable PM for rt10xx series for now? this blocks the weekly CI testing, and any user run with this code will get his board locked

hakehuang, 2023-09-20:
fixed by https://github.com/zephyrproject-rtos/zephyr/pull/62726

closed 2023-09-20 by hakehuang