zephyrproject-rtos/zephyr #61076: Bluetooth: Host: Inconsistent CCC "link_encrypted" field propagation

alstrzebonski, 2023-08-02:
**Describe the bug**
The scenario is as follows:
1. GATT client connects without bonding and writes some CCC. The `link_encrypted` field in `struct bt_gatt_ccc_cfg` structure in `subsys/bluetooth/host/gatt.c` is set to `false` (line 2179), because the security is not established.
2. The GATT client bonds and disconnects. The CCCs are stored in Settings, because the peer is bonded.
3. The GATT client reconnects without reestablishing security.
4. The GATT client tries to overwrite the CCC.
5. Depending on `CONFIG_BT_SETTINGS_CCC_LAZY_LOADING` being enabled or not:
    - If `CONFIG_BT_SETTINGS_CCC_LAZY_LOADING` is enabled, the client can't overwrite the CCC, because it was loaded from Settings on reconnection and `link_encrypted` field was set to `true` in `ccc_load` function.
    - If `CONFIG_BT_SETTINGS_CCC_LAZY_LOADING` is disabled and the device was not rebooted, the client can overwrite the CCC as the `link_encrypted` field is set to false.
    -  If `CONFIG_BT_SETTINGS_CCC_LAZY_LOADING` is disabled and the device was rebooted after the disconnection, the client can't overwrite the CCC, because it was loaded from Settings on reboot and `link_encrypted` field was set to `true` in `ccc_load` function.

The issue results from RAM and flash memory being inconsistent. While writing CCC, the `link_encrypted` field is set to `(bt_conn_get_security(conn) >= BT_SECURITY_L2)`. However, when the CCC is loaded from Settings the `link_encrypted` field is always set to `true`, because the assumption is that only CCCs for bonded peers are written to Settings. The inconsistency occurs when the peer writes CCC before bonding.

**To Reproduce**
Steps to reproduce the behavior:
1. Build and flash zephyr/samples/bluetooth/peripheral sample with CONFIG_BT_SETTINGS_CCC_LAZY_LOADING disabled.
2. Connect (without bonding!) with the sample from your Bluetooth central (e.g. phone). You can use nRF Connect app to do that.
3. Write some CCC e.g. Battery Level in Battery Service.
4. Bond with the sample.
5. Disconnect.
6. Remove bond on the phone side.
7. Reboot sample (Optional).
8. Connect (without bonding) to sample.
9. See that: If step 7 was executed you can't read the previously written CCC. If step 7 was skipped, you can read that CCC (that shouldn't be allowed).
10. Write that CCC again.
11. Bond with the sample.
12. Read the CCC.
13. See that: If step 7 was skipped, you have successfully overwritten CCC without establishing security in step 10 (that shouldn't be allowed). If step 7 was executed your CCC write from step 10 took no effect.

**Expected behavior**
I think that the CCC overwrite shouldn't be allowed even if this CCC has been written before bonding. For sure, the behavior should be consistent and shouldn't depend on loading from Settings. I think that `link_encrypted` CCC configuration parameter should be promoted to true after the GATT client bonds.

**Impact**
It can slightly compromise security.

**Environment (please complete the following information):**
 - OS: Linux
 - Commit SHA: b37ce40878ec52295afe6350d17712d60eb6639a (recent one)

alwa-nordic, 2023-08-08:
I agree with your assessment of the expected behavior.

jori-nordic, 2023-09-25:
Made irrelevant by https://github.com/zephyrproject-rtos/zephyr/pull/62985