zephyrproject-rtos/zephyr #61088: Build error in Zephyr/smaples/net/gsm_modem for QuectelEC200U with stm32f412g_disco board

SinghKarminder, 2023-08-02:
PS D:\copy_zephyr\zephyr\samples\net\gsm_modem> west build -p always -b stm32f412g_disco
-- west build: making build dir D:\copy_zephyr\zephyr\samples\net\gsm_modem\build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: D:/copy_zephyr/zephyr/samples/net/gsm_modem
-- CMake version: 3.27.1
-- Found Python3: C:/Python310/python.exe (found suitable exact version "3.10.6") found components: Interpreter
-- Cache files will be written to: D:/copy_zephyr/zephyr/.cache
-- Zephyr version: 3.4.99 (D:/copy_zephyr/zephyr)
-- Found west (found suitable version "1.1.0", minimum required is "0.14.0")
-- Board: stm32f412g_disco
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.1 (D:/zephyr-sdk-0.16.1)
-- Found toolchain: zephyr 0.16.1 (D:/zephyr-sdk-0.16.1)
-- Could NOT find Dtc (missing: DTC) (Required is at least version "1.4.6")
-- Found BOARD.dts: D:/copy_zephyr/zephyr/boards/arm/stm32f412g_disco/stm32f412g_disco.dts
-- Found devicetree overlay: D:/copy_zephyr/zephyr/samples/net/gsm_modem/stm32f412g_disco.overlay
WARNING:devicetree.edtlib:'label' is marked as deprecated in 'properties:' in D:/copy_zephyr/zephyr/dts/bindings\modem\zephyr,gsm-ppp.yaml for node /soc/serial@40011400/gsm-modem.
-- Generated zephyr.dts: D:/copy_zephyr/zephyr/samples/net/gsm_modem/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: D:/copy_zephyr/zephyr/samples/net/gsm_modem/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: D:/copy_zephyr/zephyr/samples/net/gsm_modem/build/zephyr/dts.cmake
Parsing D:/copy_zephyr/zephyr/Kconfig
Loaded configuration 'D:/copy_zephyr/zephyr/boards/arm/stm32f412g_disco/stm32f412g_disco_defconfig'
Merged configuration 'D:/copy_zephyr/zephyr/samples/net/gsm_modem/prj.conf'
Configuration saved to 'D:/copy_zephyr/zephyr/samples/net/gsm_modem/build/zephyr/.config'
Kconfig header saved to 'D:/copy_zephyr/zephyr/samples/net/gsm_modem/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: d:/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
CMake Warning at D:/copy_zephyr/zephyr/CMakeLists.txt:892 (message):
  No SOURCES given to Zephyr library: drivers__entropy

  Excluding target from build.

-- Configuring done (9.1s)
-- Generating done (0.1s)
-- Build files have been written to: D:/copy_zephyr/zephyr/samples/net/gsm_modem/build
-- west build: building application
[1/218] Generating include/generated/version.h
-- Zephyr version: 3.4.99 (D:/copy_zephyr/zephyr), build: zephyr-v3.4.0-1766-g9dc531be481a
[213/218] Linking C executable zephyr\zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map D:/copy_zephyr/zephyr/samples/net/gsm_modem/build/zephyr/zephyr_pre0.map
cmd.exe /C "cd . && D:\zephyr-sdk-0.16.1\arm-zephyr-eabi\bin\arm-zephyr-eabi-gcc.exe  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr\zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=D:/copy_zephyr/zephyr/samples/net/gsm_modem/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/net/l2/ppp/libsubsys__net__l2__ppp.a  zephyr/subsys/net/ip/libsubsys__net__ip.a  zephyr/subsys/net/conn_mgr/libsubsys__net__conn_mgr.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/modem/libdrivers__modem.a  zephyr/drivers/net/libdrivers__net.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/reset/libdrivers__reset.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/stm32/stm32cube/lib..__modules__hal__stm32__stm32cube.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"d:/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v7e-m/nofp"  -LD:/copy_zephyr/zephyr/samples/net/gsm_modem/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie && cmd.exe /C "cd /D D:\copy_zephyr\zephyr\samples\net\gsm_modem\build\zephyr && D:\CMake\bin\cmake.exe -E true""
d:/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: zephyr/subsys/net/l2/ppp/libsubsys__net__l2__ppp.a(fsm.c.obj): in function `sys_rand32_get':
D:/copy_zephyr/zephyr/samples/net/gsm_modem/build/zephyr/include/generated/syscalls/rand32.h:32: undefined reference to `z_impl_sys_rand32_get'
d:/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: zephyr/subsys/net/ip/libsubsys__net__ip.a(net_context.c.obj): in function `sys_rand32_get':
D:/copy_zephyr/zephyr/samples/net/gsm_modem/build/zephyr/include/generated/syscalls/rand32.h:32: undefined reference to `z_impl_sys_rand32_get'
d:/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: zephyr/subsys/net/ip/libsubsys__net__ip.a(net_shell.c.obj): in function `sys_rand32_get':
D:/copy_zephyr/zephyr/samples/net/gsm_modem/build/zephyr/include/generated/syscalls/rand32.h:32: undefined reference to `z_impl_sys_rand32_get'
d:/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: D:/copy_zephyr/zephyr/samples/net/gsm_modem/build/zephyr/include/generated/syscalls/rand32.h:32: undefined reference to `z_impl_sys_rand32_get'
d:/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: zephyr/drivers/net/libdrivers__net.a(ppp.c.obj): in function `sys_rand32_get':
D:/copy_zephyr/zephyr/samples/net/gsm_modem/build/zephyr/include/generated/syscalls/rand32.h:32: undefined reference to `z_impl_sys_rand32_get'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'D:\CMake\bin\cmake.EXE' --build 'D:\copy_zephyr\zephyr\samples\net\gsm_modem\build'

SinghKarminder, 2023-08-02:
This is my prj.conf file

# UART support
CONFIG_SERIAL=y
#CONFIG_GPIO=y
#CONFIG_GSM_MUX=y
CONFIG_BOARD_STM32F412G_DISCO=y

# GSM modem support
CONFIG_MODEM=y
CONFIG_MODEM_GSM_PPP=y
CONFIG_MODEM_GSM_QUECTEL=y

# PPP networking support
CONFIG_NET_DRIVERS=y
CONFIG_NET_PPP=y
CONFIG_NET_L2_PPP=y
CONFIG_NET_NATIVE=y
CONFIG_NETWORKING=y

CONFIG_NET_L2_PPP_TIMEOUT=10000

# IPv4 enables PPP IPCP support
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n

# Network management events
CONFIG_NET_CONNECTION_MANAGER=y

# Log buffers, modem and PPP
CONFIG_LOG=y
CONFIG_NET_LOG=y
CONFIG_LOG_BUFFER_SIZE=16384
CONFIG_MODEM_LOG_LEVEL_DBG=y
CONFIG_NET_PPP_LOG_LEVEL_DBG=y
CONFIG_NET_L2_PPP_LOG_LEVEL_DBG=y
CONFIG_NET_MGMT_EVENT_LOG_LEVEL_DBG=y
CONFIG_NET_CONNECTION_MANAGER_LOG_LEVEL_DBG=y
CONFIG_MODEM_GSM_APN="airtelgprs.com"

CONFIG_NET_SHELL=y
CONFIG_MODEM_SHELL=y
CONFIG_NET_DEFAULT_IF_PPP=y
CONFIG_MODEM_GSM_FACTORY_RESET_AT_BOOT=y
CONFIG_MODEM_SIM_NUMBERS=n
#CONFIG_POSIX_API=y
#CONFIG_NET_SOCKETS_POSIX_NAMES=n
CONFIG_GSM_PPP_AUTOSTART=n
CONFIG_PPP_NET_IF_NO_AUTO_START=n
CONFIG_NET_IF_LOG_LEVEL_DBG=y

SinghKarminder, 2023-08-02:
I have also added stm32f412g_disco.overlay file in the gsm_modem

// To get started, press Ctrl+Space to bring up the completion menu and view the available nodes.

// You can also use the buttons in the sidebar to perform actions on nodes.
// Actions currently available include:

// * Enabling / disabling the node
// * Adding the bus to a bus
// * Removing the node
// * Connecting ADC channels

// For more help, browse the DeviceTree documentation at https://docs.zephyrproject.org/latest/guides/dts/index.html
// You can also visit the nRF DeviceTree extension documentation at https://nrfconnect.github.io/vscode-nrf-connect/devicetree/nrfdevicetree.html

&usart2 {
    status = "okay";
};

&usart6 {
	status = "okay";
    current-speed = <115200>;
	gsm: gsm-modem {
		status = "okay";
		compatible = "zephyr,gsm-ppp";
		label = "modem_gsm";
	};
};

nordicjm, 2023-08-03:
>  undefined reference to `z_impl_sys_rand32_get'

You haven't enabled a random number generator, look in `menuconfig` and enable one

SinghKarminder, 2023-08-03:
Thanks @nordicjm.

closed 2023-08-03 by SinghKarminder