zephyrproject-rtos/zephyr #61099: Sample STM32PM Blinky higher energy usage on nucleo_wb55rg since adding prescaler for lptim1

jbr-ia, 2023-08-03:
**Describe the bug**
When running the sample stm32 power_mgmt blinky (zephyr/samples/boards/stm32/power_mgmt/blinky/) on the nucleo_wb55rg board, the microcontroller seems to wake up from sleep every 2 ms. This causes higher energy-usage than expected (on average between 600 and 700 uA when the LED is off).
With an older version of zephyr (before commit 29ef82cfaf02a6171eb85462bda73d11e7bd84bf), the microcontroller did not wake up so frequently, and the current was a about 100 to 130 uA with the LED off. 

My assumption is that is has to do with the prescaler which is set to 32 in the overlay for the nucleo_wb55rg. This results in 1024 ticks per second, but the default CONFIG_SYS_CLOCK_TICKS_PER_SEC is at 4096.
When I set the prescaler to 8 in nucleo_wb55rg.overlay, or the CONFIG_SYS_CLOCK_TICKS_PER_SEC to 1024 in prj.conf, the microcontroller does not wake up that often, and the energy-usage is about the same as before commit 29ef82cfaf02a6171eb85462bda73d11e7bd84bf. Each of these changes (changing prescaler, or changing  CONFIG_SYS_CLOCK_TICKS_PER_SEC) can solve the problem, but I don't know the root cause of the problem, and do not know if this is the desired solution.

**To Reproduce**
build the example:
west build -b nucleo_wb55rg samples/boards/stm32/power_mgmt/blinky

flash it to the microcontroller
west flash

Analyse the energy-consumption

**Expected behavior**
I expect the microcontroller to be sleeping for longer periods of time than 2 ms

**Impact**
Higher enenergy-consumption than expected

**Logs and console output**
Current with sample:
[image]

Detail of current with peak every 2 ms:
[image]

Current with prescaler at 8:
[image]

**Environment:**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.16.1
 - Commit SHA or Version used: 0a2d538c2bc442e64b3b86f3cf05a311223466ac

**Additional context**
When the current is measured in the nucleo wb55rg, the jumpers for the STlink (JP5) are removed, the USB-cable disconnected, and voltage supplied via JP2.

FRASTM, 2023-08-03:
Thanks for your deep and pertinent analysis. 
In case of nucleo_wb55rg, the LPTIM prescaler was arbitrary set to 32 to have a large timeout period (several seconds).
The prescaler value is set in the  overlay file to be adjusted depending on the application. So that you can easily set the prescaler to 8 for your need.

jbr-ia, 2023-08-03:
Thanks a lot for the quick response. For now I will use a prescaler of 8 in my application.

fabiobaltieri, 2023-08-08:
Hey, seems like this has been resolved, I'll close this one.

FRASTM, 2023-10-04:
I do not see many diff on low power (zephyr v3.5.0.rc1) when LPTIM prescaler  = <32>;
[image]
compared to prescaler = <8>;
[image]

FRASTM, 2023-10-04:
We do see the correlation  between the lptim presecaler and the CONFIG_SYS_CLOCK_TICKS_PER_SEC

closed 2023-12-12 by fabiobaltieri