zephyrproject-rtos/zephyr #61129: acrn_ehl_crb:tests/lib/newlib/thread_safety/libraries.libc.newlib_nano.thread_safety.userspace.stress

fabiobaltieri, 2023-08-03:
**Describe the bug**
The test seems be failing fairly consistently with:

`/opt/toolchains/zephyr-sdk-0.16.1/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/12.2.0/../../../../x86_64-zephyr-elf/bin/ld.bfd: scripts/build/gen_kobject_placeholders.py did not reserve enough space for kobject rodata.`

https://github.com/zephyrproject-rtos/zephyr/runs/15459346533

Can't reproduce locally, any chance you x86 folks could look into it?

Can't quite figure if it's a regression but I've seen it failing for a while, for some reason only in the CI run.

**To Reproduce**
No idea, can't reproduce it locally, that would be

`west build -p -b acrn_ehl_crb -T tests/lib/newlib/thread_safety/libraries.libc.newlib_nano.thread_safety.userspace.stress`

**Expected behavior**
A nice green PASS.

**Impact**
Blocking PR CI run.

**Logs and console output**
https://github.com/zephyrproject-rtos/zephyr/runs/15459346533

fabiobaltieri, 2023-08-03:
https://github.com/zephyrproject-rtos/zephyr/pull/61131 for excluding it until it's fixed.

fabiobaltieri, 2023-08-03:
@enjiamai @jhedberg anyone could look into this?

najumon1980, 2023-08-07:
Currently we are reviewing ACRN support for Zephyr and will update you on this.

Meantime, will verify with the PR chttps://github.com/zephyrproject-rtos/zephyr/pull/59917 which I created recently for enhance ACPI support for x86 arch porting.

fabiobaltieri, 2023-08-07:
thanks @najumon1980, feel free to rollback the exclude and close this one if that fixes it, no urgency here anymore since it does not stop CI runs anymore

fabiobaltieri, 2023-08-08:
Update: was actually able to reproduce by upgrading to SDK 16.1, turns out I was running 16.0

fabiobaltieri, 2023-08-08:
This was reported in https://github.com/zephyrproject-rtos/zephyr/issues/61246 on other boards, guess that means it was not really related to this board.

fabiobaltieri, 2023-08-08:
@dcpleung looks like https://github.com/zephyrproject-rtos/zephyr/blob/32bd3aa781523009332f7e102fdd555cf740acdc/include/zephyr/linker/kobject-rom.ld#L45-L47 is triggering for this and few other cases, it appears that this was not happening with Zephyr SDK 16.0, but it does with 16.1, can be reproduced with

`west build -p -b acrn_ehl_crb -T tests/lib/newlib/thread_safety/libraries.libc.newlib_nano.thread_safety.userspace.stress`

Is this something you could look into?

fabiobaltieri, 2023-08-08:
Well let me dig into it: looks like the difference is in the generated `KOBJECT_RODATA_SZ`:

```
$ cat  build-16.0/zephyr/include/generated/linker-kobject-prebuilt-rodata.h
#ifndef KOBJECT_RODATA_ALIGN
#define KOBJECT_RODATA_ALIGN 32
#endif

#ifndef KOBJECT_RODATA_SZ
#define KOBJECT_RODATA_SZ 528
#endif

$ cat  build-16.1/zephyr/include/generated/linker-kobject-prebuilt-rodata.h
#ifndef KOBJECT_RODATA_ALIGN
#define KOBJECT_RODATA_ALIGN 32
#endif

#ifndef KOBJECT_RODATA_SZ
#define KOBJECT_RODATA_SZ 272
#endif
```

fabiobaltieri, 2023-08-08:
```
$ readelf -W -e ./build-16.*/zephyr/CMakeFiles/kobj_prebuilt_hash_output_lib.dir/kobject_prebuilt_hash.c.obj | grep '\.rodata'
  [ 5] .rodata           PROGBITS        0000000000000000 001f00 000200 00   A  0   0 32
  [ 5] .rodata           PROGBITS        0000000000000000 001300 000100 00   A  0   0 32
$ grep KOBJECT_RODATA_AREA_EXTRA_BYTES= build-16.*/zephyr/.config
build-16.0/zephyr/.config:CONFIG_KOBJECT_RODATA_AREA_EXTRA_BYTES=16
build-16.1/zephyr/.config:CONFIG_KOBJECT_RODATA_AREA_EXTRA_BYTES=16
```

fabiobaltieri, 2023-08-08:
```
$ readelf -W -e ./build-16.0/zephyr/CMakeFiles/kobj_prebuilt_hash_output_lib.dir/kobject_prebuilt_hash.c.obj | grep '\.rodata'
  [ 5] .rodata           PROGBITS        0000000000000000 001f00 000200 00   A  0   0 32
$ readelf -W -e ./build-16.0/zephyr/kobject_hash_renamed.o | grep '.kobject_data.rodata'
  [ 5] .kobject_data.rodata PROGBITS        0000000000000000 0012a0 000100 00   A  0   0 32

$ readelf -W -e ./build-16.1/zephyr/CMakeFiles/kobj_prebuilt_hash_output_lib.dir/kobject_prebuilt_hash.c.obj | grep '\.rodata'
  [ 5] .rodata           PROGBITS        0000000000000000 001300 000100 00   A  0   0 32
$ readelf -W -e ./build-16.1/zephyr/kobject_hash_renamed.o | grep '.kobject_data.rodata'
  [ 5] .kobject_data.rodata PROGBITS        0000000000000000 001e40 000200 00   A  0   0 32
```

This is getting hard to follow but it looks weird.

`zephyr/kobject_prebuilt_hash.c` is obviously different between the two builds, maybe it's better if I leave it up to you folks who knows how this thing actually is. :-) @dcpleung

dcpleung, 2023-08-08:
Hm... `build/zephyr/kobject_hash.c` is significantly bigger between SDK 0.16.0 and 0.16.1. It seems like 0.16.1 is placing variables differently, resulting in lots of holes in the hash table (and thus bigger size). No idea why that would be the case though...

fabiobaltieri, 2023-08-08:
@dcpleung hey now that I upgraded the compiler I realized that this works on v3.4.0, so it may be a regression after all. Hang on I'm bisecting it.

fabiobaltieri, 2023-08-08:
```
2b1106a407cc592cbce7d7bda146d9b99d482df3 is the first bad commit
commit 2b1106a407cc592cbce7d7bda146d9b99d482df3
Author: Flavio Ceolin <flavio.ceolin@intel.com>
Date:   Fri Jul 14 12:24:33 2023 -0700

    kernel: userspace: Use only list for dynamic objs
```

fabiobaltieri, 2023-08-09:
Hey, just to keep track of what's been disabled for this so we can clean properly once fixed:

https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/lib/newlib/thread_safety/testcase.yaml#L66
https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/net/socket/socketpair/testcase.yaml#L10
https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/net/socket/socketpair/testcase.yaml#L15
https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/net/socket/socketpair/testcase.yaml#L20

closed 2024-01-13 by github-actions[bot]