zephyrproject-rtos/zephyr #61155: Add a tag to the python-devicetree

xiaoxiao997, 2023-08-04:
Hi, excuse me, since I need to use the Device tree conversion tool in zyphyr. Could you please add a new tag in the zephyrproject-rtos/python-devicetree directory for the conversion of devicetree dts in the latest zephyr?

carlescufi, 2023-08-04:
> Hi, excuse me, since I need to use the Device tree conversion tool in zyphyr. Could you please add a new tag in the zephyrproject-rtos/python-devicetree directory for the conversion of devicetree dts in the latest zephyr?

Hi @xiaoxiao997, what kind of tag do you mean? a Git tag?

xiaoxiao997, 2023-08-04:
yes，i want to a git tag,for zephyr/python-devicetree

mbolivar-ampere, 2023-08-07:
What sort of tagging policy are you asking for? The directory is part of the zephyr tree so any tagged zephyr release is also a tag of that directory.

xiaoxiao997, 2023-08-09:
Hello, I just need this tag from the device tree. Could you please insert a tag only for the latest device tree?

xiaoxiao997, 2023-08-09:
[image]

xiaoxiao997, 2023-08-10:
yes，I want to  the tag for https://github.com/zephyrproject-rtos/python-devicetree

mbolivar-ampere, 2023-08-15:
> Is this being kept up to date with whatever is happening inside the zephyr tree?

It is not. It was an experiment and I think we do not need it. I think we can keep the package in the tree and deprecate this repository. Any comments from anyone else?

nashif, 2024-02-08:
Yep, lets archive that repo.

closed 2024-02-08 by nashif