zephyrproject-rtos/zephyr #61173: net: regression on native IP stack when an offload driver is present

RomainPelletant, 2023-08-04:
**Describe the bug**
In case we include ethernet driver (enc424j600) and wifi offload driver (esp-at), the ethernet interface is no more usable and go into hard fault.
We are only using one interface at a time (exchangeable).
But, surprisingly, Wi-Fi interface is working well when selected.

Please also mention any information which could help others to understand
the problem you're facing:
 - The target network interface is always set as default interface

**To Reproduce**
Steps to reproduce the behavior:
1. include enc424j600 or another native driver
2. include an offloaded driver or esp-at
3. Make something on network (UDP or TCP)
4. See error

**Expected behavior**
Connect succesfull

** Workaround **
Revert PR #58232

**Impact**
showstopper

**Additional context**
Duplicate of issue #58435

rlubos, 2023-08-07:
How do you switch between native/offloaded interfaces? I'm asking, because NET_OFFLOAD offloading is known of not being well prepared to work with multiple interfaces in the system (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/net_context.c#L258, https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/net_context.c#L330) - it will always assume the default interface should be used. Are you switching the default interface before allocating socket/net_context?

Anyway, I can see one possible issue with native interfaces when NET_OFFLOAD is enabled - after the change which installs the recv handler before connect, this line might be an issue: https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/net_context.c#L2114. In case of native interface, the `iface` pointer on the `context` may still not be set, which would lead to an assert/crash `net_if_is_ip_offloaded()`.

Given that we now assign the interface to the context during context creation for offloaded cases (see https://github.com/zephyrproject-rtos/zephyr/pull/59851) I think we could update `net_if_is_ip_offloaded()` to handle NULL pointer more gracefully (as it would indicate we deal with native interface). Could you check if changing the function as follows helps in your case:
```patch
 static inline bool net_if_is_ip_offloaded(struct net_if *iface)
 {
 #if defined(CONFIG_NET_OFFLOAD)
-       NET_ASSERT(iface);
-       NET_ASSERT(iface->if_dev);
-
-       return (iface->if_dev->offload != NULL);
+       return (iface != NULL && iface->if_dev != NULL &&
+               iface->if_dev->offload != NULL);
 #else
        ARG_UNUSED(iface);
```

RomainPelletant, 2023-08-07:
Hi @rlubos,

Thank you for your answer.
Before loading drivers, network interface is known is EEPROM and only the right driver is loaded.
It works even when two interfaces drivers are loaded, but before doing network things, I always call **_net_if_set_default();_**

Impressive : I rebased from current 3.4 and apply your patch : it works like a charm.
I looked into socket.c and issue always came around net_if_is_ip_offloaded but I didn't see the inline function in header.

I can do PR and can leave you do so as you want.
Thank you for your help

rlubos, 2023-08-07:
@RomainPelletant I've opened https://github.com/zephyrproject-rtos/zephyr/pull/61215 with the fix.

PR #61215 (2023-08-08): net: iface: Allow NULL pointer to be handled in net_if_is_ip_offloaded
In case native interface is used with NET_OFFLOAD enabled, the net_if_is_ip_offloaded() could lead to a crash, if called on unbound net_context (i.e. with no iface assigned to net_context yet).

As since commit 40d224022608a3e5d68e95cc4d6f7606f1d48fb2 the net_context allocated on offloaded interface is always assigned with the iface pointer during its creation, it can be safely assumed that in case of NULL pointer is provided we deal with a native interface. Therefore instead of asserting on the iface pointer, indicate that [... cut]
closed 2023-08-08 by fabiobaltieri