zephyrproject-rtos/zephyr #61181: Strange behavior on closing TCP connection: drops and retransmits in FIN states

AndreyDodonov-EH, 2023-08-05:
1. _If we are a TCP client and issuing a FIN, we stop processing the data,_ but keep acknowledging them,
should the other part keep sending data.
(Btw, this also was not always the case in TCP_FIN_WAIT_2 until https://github.com/zephyrproject-rtos/zephyr/pull/57509 (thanks @ssharks for the fix). This corrected a connection leak, because connection was otherwise forever stuck in TCP_FIN_WAIT_2)
_So first point - is dropping data in FIN states okay?_ That's not how I understood https://datatracker.ietf.org/doc/html/rfc9293
(Btw, it replaces many standards which are referenced in code, such as 6528, which is now considered obsolete, but content is still very similar)

2. We agressevly retransmit [FIN | ACK] both before the aforementioned fix and especially after, despite counterpart acknowledging our FIN. I don't think it's a correct behavior.

Here are excerpts from the log. 192.168.2.3 is a Zephyr system.
Before fix, FIN | ACK is retransmitted once despite being ACK-ed, then connection is leaked (which eventually leads to communication dying after all connections (`CONFIG_NET_MAX_CONN`) are leaked):
[image]

After the fix, FIN | ACK is pushed agressively, but since now we keep ACK-ing, eventually connection is closed:

[image]

Here are the complete traces:
[closing_connection_wireshark_traces.zip](https://github.com/zephyrproject-rtos/zephyr/files/12267516/closing_connection_wireshark_traces.zip)

ssharks, 2023-08-05:
It looks not very correct indeed. It looks 2 things are going wrong in your second trace. For some reason it looks like the state machine keeps hanging in state FIN1.
The ACK is not properly processed at least, although I cannot fully understand yet why.
This allows a second FIN to be transmitted with the wrong sequence number when there is new data coming in with packet 39.

I attempted to correct both issues in PR https://github.com/zephyrproject-rtos/zephyr/pull/61183. Could you have a look if this improves your situation?

ssharks, 2023-08-05:
FYI: Past week at least 4 PRs on TCP have been merged, maybe these fixes help you as well.

ssharks, 2023-08-05:
Fyi: @rlubos

ssharks, 2023-08-06:
With respect to your first comment: As the current stack does not implement half closing the connection, the RFC section 3.6.1 tells us we should send a RST packet instead of silently acknowledging it, as it does right now.

AndreyDodonov-EH, 2023-08-08:
> With respect to your first comment: As the current stack does not implement half closing the connection, the RFC section 3.6.1 tells us we should send a RST packet instead of silently acknowledging it, as it does right now.

@ssharks 
In the logs I do not see [RST] flag... Am I missing something?

AndreyDodonov-EH, 2023-08-08:
@ssharks  Or are you stating that it's another issue, that we do not send RST?

ssharks, 2023-08-09:
I started a second PR with the reset behavior https://github.com/zephyrproject-rtos/zephyr/pull/61353. The whole FIN sequence is rewritten and needs some more testcases to validate all possible corner cases.

AndreyDodonov-EH, 2023-08-10:
> I started a second PR with the reset behavior #61353. The whole FIN sequence is rewritten and needs some more testcases to validate all possible corner cases.

That's really great!
I think the main issue was eventually running into the exact scenario described here https://www.rfc-editor.org/rfc/rfc2525#page-51 (which is mentioned in https://www.rfc-editor.org/rfc/rfc9293#name-half-closed-connections),
where it is stated if the client does not support half-closed connection (i.e. accepting data upon issuing FIN), it should be issuing RST.

rlubos, 2023-08-23:
@ssharks Shall this be closed given that #61353 is now merged?

ssharks, 2023-08-23:
I think it is solved indeed, so closing this issue.

closed 2023-08-23 by ssharks