zephyrproject-rtos/zephyr #61190: DHCPv4Client Sample build throwing an error due Option Callbacks

RaysoftTechnologiesInc, 2023-08-06:
Hi,

I am currently trying to interface my nrf52840 DK with the ENC28J60 module to run the dhcpv4 client sample, however I am running into the error below when building the project, it seems there is something wrong with the Option call backs flag that we enable in the project config.. Is there something I maybe doing wrong?
[image]

thedjnK, 2023-08-06:
You haven't enabled networking or haven't setup your dts properly

closed 2023-08-07 by nordicjm

RaysoftTechnologiesInc, 2023-08-07:
Hi @thedjnK,

thank you for the response, I just took the example as it is, since i am new to the nrf, could you show me where I left out the network enabling flag and also how to correctly set up the dts.

nordicjm, 2023-08-07:
You need to add the device to device tree, either by editing the board files (generally you would do this with a custom board) or by using an overlay - see https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/shields/mikroe_eth_click/mikroe_eth_click.overlay for an example, this is one which adds it via a shield.

RaysoftTechnologiesInc, 2023-08-07:
I have added the overlay file as you mentioned above, However there are still issues in the build. 
[image]
[image]
I had to change some names in the overlay 
mikrobus_spi  to spi0
mikrobus_header  to gpio0

because it was throwing errors. Now it does not seem to recognise the full duplex command for the SPI.

nordicjm, 2023-08-07:
You are using NCS which is behind the current zephyr codebase, the version of NCS you are using does not support that because it was added recently so you'd need to remove it

RaysoftTechnologiesInc, 2023-08-07:
Hi @nordicjm,

I have downloaded the latest version of NCS and included the eth shield in the build, however I am still getting the error below.
[image]
I am really not sure as to what i am doing wrong.