zephyrproject-rtos/zephyr #61203: tests: net: socket: tcp: net.socket.tcp does to fit into nrf5340dk_nrf5340_cpuapp_ns and nrf9160dk_nrf9160_ns flash

nordic-piks, 2023-08-07:
**Describe the bug**
The tests/net/socket/tcp/net.socket.tcp.tests does not fit flash memory of nrf9160dk_nrf9160_ns and  nrf5340dk_nrf5340_cpuapp_ns

Observed for
- nrf9160dk_nrf9160_ns
- nrf5340dk_nrf5340_cpuapp_ns

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf9160dk_nrf9160_ns connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/net/socket/tcp -p nrf9160dk_nrf9160_ns --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
sanitycheck_build/nrf5340dk_nrf5340_cpuapp_ns/tests/net/socket/tcp/net.socket.tcp/zephyr && /usr/local/bin/cmake -E true
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/zephyr_pre0.elf section `rodata' will not fit in region `FLASH'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: region `FLASH' overflowed by 10878 bytes
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.16.0
 - Commit SHA or Version used: zephyr-v3.4.0-2028-g86812b155103

nordic-piks, 2023-08-07:
Seems to be regression after https://github.com/zephyrproject-rtos/zephyr/commit/5b640ec9db531169daa2babb7f2ab3b23c8ff434

rlubos, 2023-08-07:
@ssharks It seems that enabling logging by default in `tests/net/socket/tcp` (https://github.com/zephyrproject-rtos/zephyr/commit/5b640ec9db531169daa2babb7f2ab3b23c8ff434) increase the FLASH footprint of the project considerably. Are you ok with disabling the logger in the project, so that we don't have to reduce the number of platforms we can run the tests on?

ssharks, 2023-08-07:
> @ssharks It seems that enabling logging by default in `tests/net/socket/tcp` ([5b640ec](https://github.com/zephyrproject-rtos/zephyr/commit/5b640ec9db531169daa2babb7f2ab3b23c8ff434)) increase the FLASH footprint of the project considerably. Are you ok with disabling the logger in the project, so that we don't have to reduce the number of platforms we can run the tests on?

Oops, that was not intended to be pushed. Would you like me to submit a correcting PR?

rlubos, 2023-08-08:
> Oops, that was not intended to be pushed. Would you like me to submit a correcting PR?

That would be great if you could send a PR, thanks.

closed 2023-08-09 by fabiobaltieri