zephyrproject-rtos/zephyr #61227: (Re-)Enforce vendor and protocol neutrality of the IEEE 802.15.4 driver

fgrandel, 2023-08-07:
## Introduction

Transforming the IEEE 802.15.4 driver API (back) into being vendor- and protocol-neutral, see [detailed rationale including practical examples of problems and solutions](https://github.com/zephyrproject-rtos/zephyr/pull/58438#discussion_r1220059230).

### Problem description

Recent additions to the IEEE 802.15.4 driver API mostly broke encapsulation: They are either closely coupled to one specific upper layer stack (OT) or a specific vendor's choice of hardware offloading.

This created considerable (wrt to Thread protocol support: probably prohibitively high) technical debt for other vendors' community driver maintainers when different offloading choices or different upper layer stacks are to be implemented. This technical debt is currently experienced by the TSCH protocol and CC13/26xx driver developments. See e.g. #60946, https://github.com/openthread/openthread/pull/9322, #60401, #59133, #58434, #51266 for a non-exhaustive list of PRs that were required already just to fix technical debt.

NB: By my personal estimate, the linked PRs cover about 5-10% of the actual technical debt in the current API. The remainder lies in transforming IE and security support such that the underlying standard primitives become protocol-agnostic and re-usable by other standard protocols (including but not limited to TSCH) as the standard requires.

### Proposed change

The following future **IEEE 802.15.4 driver API review policies** are proposed as decision points:

* All future additions to the public IEEE 802.15.4 driver API SHALL either prove (by letting the API docs point to appropriate sections of current/future standard versions) that they are vendor and upper-layer agnostic or they SHALL be encapsulated in separate files selected via Kconfig/Devicetree if (and only if) the specific vendor/upper-layer stack is active.
* MAC or PHY features that are covered by the IEEE 802.15.4 standard (MLME, data structures, PHY/MAC PIB attributes, etc.), SHALL be divided up into primitives exactly as defined by the standard (again to be proven by pointers to the standard).
* The driver architecture SHALL continue to support all kinds of non-mandatory/non-standard hardware-offloading measures for vendors to differentiate themselves ("hard MAC", "vendor extensions", "protocol extensions").
* We promote a "soft MAC first" approach such that driver maintainers have to implement strictly only what is required by the standard no matter what the individual offloading requirements of one vendor, i.e. [driver maintainers MAY decide to not do hardware offloading and such a choice SHALL have first-class architecture support](https://github.com/zephyrproject-rtos/zephyr/pull/58438#discussion_r1267034840) even if the soft-MAC solution is not implemented right away.
* Zephyr L2 code SHALL NOT call directly into soft-MAC solutions that are exclusively provided by one vendor's driver. Such soft-MAC features SHALL be ported to vendor-agnostic (possibly but not necessarily protocol-agnostic) soft-MAC code re-usable by all drivers before being accepted into mainline. Zephyr SHALL only call into vendor-specific driver code that provably does hardware offloading for performance reasons. (Most prominent example: IE parser in the nRF5 driver that runs on the CPU and could have been contributed in a vendor-agnostic way w/o any loss of performance.)

As an **architectural measure**, it is proposed, to divide the IEEE 802.15.4 driver API up into two strictly separate sections:
 * PHY/L1
 * Hard MAC/L2

Implementing the PHY/L1 part SHALL be sufficient for basic soft MAC support of all Zephyr protocol stacks, i.e. the hard MAC/L2 layer SHALL be architecturally strictly optional even if the standard itself requires de-facto hardware offloading on *some* hardware (e.g. specialized peripherals in the absence of a fast-enough CPU). The decision whether some MAC feature needs to be offloaded SHALL strictly be left to the individual driver maintainers as the standard itself has no requirements whatsoever concerning implementation on CPUs vs. specialized peripherals or dedicated radio cores.

For historical reasons documentation and code calls L1 and L2 driver API parts jointly "IEEE 802.15.4 radio API" which is at best misleading. As future **documentation policy** it is therefore proposed to introduce the following nomenclature:
 * Only the PHY part SHALL be called "IEEE 802.15.4 PHY radio API" from now on
 * The MAC offloading part SHALL be called "IEEE 802.15.4 MAC offloading API"
 * Both parts together SHALL constitute the "IEEE 802.15.4 driver API"

Practical relevance and feasibility of these architectural patterns and policies was proven and exemplified by two recent PRs ([1](https://github.com/zephyrproject-rtos/zephyr/pull/60343), [2](https://github.com/zephyrproject-rtos/zephyr/pull/60401)) - among several others.

## Detailed RFC

[See the detailed discussion in a pull request that first introduced parts of the new architecture approach](https://github.com/zephyrproject-rtos/zephyr/pull/58438#discussion_r1220059230)

### Proposed change (Detailed)

[See the detailed discussion in a pull request that first introduced parts of the new architecture approach](https://github.com/zephyrproject-rtos/zephyr/pull/58438#discussion_r1220059230)

### Dependencies

Dependencies that can be shown rather objectively:
* The current API stands fundamentally in the way of all non-CSL IEEE 802.15.4 protocols that require IEs (which are ALL recent IEEE 802.15.4 protocols).
* The current API stands fundamentally in the way of implementing an upper-layer agnostic version of recent standard security features (i.e. secure IEs, device-specific security, etc.)

Dependencies that represent my personal opinions:
* IMO making offloading strictly optional is vital for a healthy IEEE 802.15.4 Zephyr community. De facto only a single vendor is currently able to support Thread protocol. This should urgently be fixed IMO. It cannot be in the interest of Zephyr to promote systematical barriers to entry of vendors with distinct offloading requirements or individuals with insufficient resources to implement non-standard offloading APIs. IMO slight deviations from standard timings or support for earlier versions only in some drivers are highly preferable over enforcing 100% certifiable support of the latest Thread version as long as we enable 100% certifiable solutions for those vendors that want it.
* Similar measures need to be taken in the BLE area IMO, if we want to fix the same problem there. Currently even well-resourced attempts by other vendors to support BLE on Zephyr will be difficult IMO. I believe that insufficient cross-vendor support for BLE, Thread and other wireless protocols is probably one of the most important barriers to further adoption of Zephyr as a standard solution in the embedded world. At least this is true for several users I've talked to as a consultant. Providing hooks for proprietary binary Thread or BLE stacks is a very poor replacement to a proper OSS architecture. I'm just not proposing this RFC as a general "wireless APIs RFC" as I don't know enough about BLE and the Zepyhr BLE stack. BLE maintainers feel free to chime in and reformulate this RFC accordingly in more general terms. Shouldn't be too difficult AFAICS.

### Concerns and Unresolved Questions

* Who is going to incur the effort to fix the considerable technical debt in code that has already been reviewed and upstreamed?
* Who is going to ensure mandatory reviews of the IEEE 802.15.4 driver API according to the above policies? The current IEEE 802.15.4 maintainer seems to be MIA since a long time and while the network maintainer does his best to fill in the gap, he has explicitly acknowledged on different occasions that he has a conflict of interest and lacks sufficiently deep knowledge of the IEEE 802.15.4 standard to properly fill in the role of a dedicated IEEE 802.15.4 maintainer.

## Alternatives

I personally have none - please feel free to add in case I'm missing something.

fgrandel, 2023-08-07:
@rlubos @jciupis @hubertmis We've been discussing the topic of this RFC on several occasions and I had promised to collect our proposals in one RFC. Here it is.

fgrandel, 2023-08-10:
@carlescufi @rlubos I think it makes sense that you review and comment on this RFC before we discuss it in the architecture WG. It is probably too specialized to discuss it in detail with 20 people in the call. I had asked @nashif to postpone it in the arch wg until when you have a chance to participate. I think your wholehearted approval is vital - otherwise it doesn't make sense.

rlubos, 2023-08-28:
Thanks @fgrandel for the comprehensive description of the idea. 

As a high level feedback, I'm ok with separating concepts of PHY/offloaded MAC features into separate components of the 15.4 radio driver API. I think it could even be considered to move certain APIs/configurations introduced specifically for OpenThread into recently introduced [OT extension header](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/net/ieee802154_radio_openthread.h), so that they do not stand in a way of other 15.4 based protocols (although this should be discussed with OT contributors).

There are however a few items that in my ignorance, require some clarification.

> Zephyr L2 code SHALL NOT call directly or indirectly into soft-MAC solutions that are exclusively provided by one vendor's driver. Such soft-MAC features SHALL be ported to vendor-agnostic (possibly but not necessarily protocol-agnostic) soft-MAC code re-usable by all drivers before being accepted into mainline. Zephyr SHALL only call into vendor-specific driver code that provably does hardware offloading for performance reasons. (Prominent examples: IE parser and certain security-related features in the nRF5 driver that run on the CPU and could be implemented in a vendor-agnostic way w/o loss of performance.)

Ok, but I'm not sure if nRF5 is a proper example here. The driver is intended (on newer SoCs) to run on a separate NET core, not the CPU where L2/soft-MAC implementation would be executed. Therefore, for me, I'm not convinced that the driver contents should be considered as a soft-MAC solution that could be implemented in a "vendor-agnostic way w/o loss of performance".

> The current API stands fundamentally in the way of all non-CSL IEEE 802.15.4 protocols that require IEs (which are ALL recent IEEE 802.15.4 protocols). This can be fixed such that future additions will be roughly an order of magnitude less effort for IE-based protocols.

> The current API stands fundamentally in the way of implementing an upper-layer agnostic version of recent standard security features (i.e. secure IEs, device-specific security, etc.). This can be fixed so that implementing such security features in additional drivers will be roughly an order of magnitude less effort.

By "stands fundamentally in the way" I understand there are some major blockers preventing other L2 implementations. I think it'd make sense to list the blockers, and decide one by one on how to proceed (either fix, in a way we don't break OT support, or make a particular config/api introduced for the sake of OT, OT specific).

> The current API makes support of [Open]Thread by other drivers unnecessarily hard. This can be fixed such that providing basic Thread protocol support by additional drivers is roughly an order of magnitude less effort.

Is "basic Thread protocol support" really that hard to achieve? I think we should settle on what "basic" means, but all of the advanced MAC features (CSL, IE support for Link Metrics , TX security) are either optional (need to be explicitly enabled) or have a soft-MAC variant in OT itself (like TX security).

fgrandel, 2023-08-28:
@rlubos Thanks for your review! :-)

> Ok, but I'm not sure if nRF5 is a proper example here. The driver is intended (on newer SoCs) to run on a separate NET core, not the CPU where L2/soft-MAC implementation would be executed.

Code that is always running on a dedicated (non-Zephyr) NET core is by definition offloaded code. Such code is not covered by this clause and most of the time cannot be ported anyway. CC13/26xx's radio core for example only accepts proprietary "blobs".

If, however, code may optionally run as part of Zepyhr-scheduled threads or ISRs, then it can be pulled into a shared "Soft MAC" w/o any loss of performance: Calls into the proprietary driver can be replaced by equivalent "inversion of control" calls into a shared L2 - no matter on which core it runs. Whether this is "fast enough" on a single core is not an L2 architecture choice but a decision to be taken by individual driver maintainers.

AFAICS the nRF5 driver contains code that (at least optionally) runs in Zephyr-scheduled threads or ISRs. The driver has a dedicated "mac_features" folder mostly implementing standard L2 primitives plus implements further standard L1/L2 primitives sprinkled throughout the rest of the code. Such code would be covered by this RFC.

This doesn't mean that I'm proposing to migrate existing code (see "Concerns and Unresolved Questions"). I'm just proposing that future PRs SHALL NOT introduce further such code and that existing code MAY be ported into shared L2 by whoever finds it worth the time.

> By "stands fundamentally in the way" I understand there are some major blockers preventing other L2 implementations. I think it'd make sense to list the blockers

No, nothing blocks alternative L2 implementations and you're right that the way I wrote it was ambivalent (hopefully fixed). Alternative implementations are just unnecessarily hard as the Zephyr-runnable code I mention above has not been pulled into granular re-usable primitives as prescribed by the standard.

By "fixing this situation" I mean that alternative re-usable implementations of these standard primitives should be provided rather than trying to migrate existing code, see "Concerns and Unresolved Questions". I would only "repair" existing code when it's actually less effort. Once such an alternative exists, existing drivers may or may not choose to migrate to the community-maintained re-usable version.

> I think it'd make sense to list the blockers

The individual features that cannot be re-used have been listed in the linked discussion thread. 

> make a particular config/api introduced for the sake of OT, OT specific

Which OT-specific features are you speaking about?

IMO the latest OT stack is exclusively based on plain IEEE 802.15.4 with some additional refinements inside the limits of the standard. The only non-standard OT feature I'm aware of is multi-CCA which we already pulled into its own namespace. All features required to support Thread 1.3.1 (not only OT) fall into the categories covered by this RFC. By the rules of this RFC it would therefore be wrong to place anything of this in an OT-specific API. 

> Is "basic Thread protocol support" really that hard to achieve? I think we should settle on what "basic" means, but all of the advanced MAC features (CSL, IE support for Link Metrics , TX security) [...]

By "basic" I mean: "The most recent version of the standard with restrictions only in precise timing". I've updated the RFC text accordingly to be more precise.

Prior versions of the standard may be even easier to support, as you say. Could you or the OT team easily name the parts of the API to be implemented for the different Thread versions?

rlubos, 2023-08-29:
> Which OT-specific features are you speaking about?

I was referring to the configs that were introduced to address OT radio API requirements:
* `IEEE802154_CONFIG_MAC_KEYS`
* `IEEE802154_CONFIG_FRAME_COUNTER`
* `IEEE802154_CONFIG_FRAME_COUNTER_IF_LARGER`
* `IEEE802154_CONFIG_RX_SLOT` 
* `IEEE802154_CONFIG_CSL_PERIOD` 
* `IEEE802154_CONFIG_CSL_RX_TIME`
* `IEEE802154_CONFIG_ENH_ACK_HEADER_IE` 

I understand they cover standard 802.15.4 features, although from the RFC I also understood that in their current form they're difficult to reuse in other upper-layer implementations. Therefore, as part of the "alternative" solutions approach you suggested:

> The easiest way to do this is to live with "alternative" approaches for some time (example: leave the monolithic CSL implementation as it is and build up an alternative IE and standards primitive API in parallel for future protocol extensions).

 they could be moved to OT namespace and possibly deprecated in the future if possible.

>  Could you or the OT team easily name the parts of the API to be implemented for the different Thread versions?

I can think of improving documentation in this area.

fgrandel, 2023-08-29:
> I was referring to the configs that were introduced to address OT radio API requirements:

@rlubos Yeah, we have the same features in mind, except for RX slots. RX slots are already re-usable and do not need any further refactoring beyond what we've already done.

We agree that these are all standard features. The fact that we've been thinking of them as "Thread requirements" is the problem I'm trying to address. IMO we should stop relating anything in L1/L2 specifically to [Open]Thread or a vendor's offloading choices unless it is really specific to that protocol stack or SoC.

This has nothing to do with offloading. It just means that both, soft MAC and offloading APIs that implement standard features, SHALL NOT be tied in any way to upper layers or to a specific vendor's SoC.

> they could be moved to OT namespace and possibly deprecated in the future if possible.

I'm not in favor of moving standard-based features into an underspecified OT architecture block where they conceptually do not belong. This would just cement and "legitimize" the current status quo and provide an unwanted "safe haven" for it. IMO our limited resources should go into eliminating the problem rather than moving it.

>> Could you or the OT team easily name the parts of the API to be implemented for the different Thread versions?

> I can think of improving documentation in this area.

Wow, that would be great. :-)

closed 2023-08-29 by fgrandel