zephyrproject-rtos/zephyr #61230: I2C endianess settings

frax84, 2023-08-07:
**Is your feature request related to a problem? Please describe.**
I'm developing a board that comunicates with two i2c sensors. I developed some i2c routines to use with these sensors. Everything works fine (comunication establishes and i can exchange data), but i noticed that the sensors need different endianess in transmission: one needs MSBfirst, the other LSBfirst. The i2c library does not have an internal management for endianess in data exchange. Considering a large project, this requires to produce some boilerplate code and increase the chance of mistakes from the user.

**Describe the solution you'd like**
It would be nice having the i2c library to manage the endianess internally using `const struct i2c_dt_spec *spec` parameter. In particular, i imagine that having a flag in the configuration such as `I2C_BIT_ORDER_MSB` and `I2C_BIT_ORDER_LSB` could be an efficient way to act. The piece of cake would be having a dedicated devicetree param too, so that in a specific dts overlay it can be declared the endianess. This would make the code more readable and maintanable.

**Describe alternatives you've considered**
In general, it is possible and easy to implement an additional step to swap bytes for the devices that need it, but in a context with many devices this would require is some boilerplate code and it should be considered bad practice if some options is available instead.

henrikbrixandersen, 2023-08-07:
I2C address and data bits, as per the specification, are transmitted MSB first. 

Any I2C device using LSB first would be violating the specification?

frax84, 2023-08-08:
hi @henrikbrixandersen and thank you very much for your comment. I took the time to verify my code. I discovered that there was the overlap of two different matters that caused the issue, but in general i failed to notice that one of the two sensor has the memory mapping that is the opposite of the other one. More explicitly:
- Sensor 1:
Reg [N] = LSBdata
Reg[N+1] = MSBdata
- Sensor 2
Reg[N] = MSBdata
Reg[N+1] = LSBdata

My bad! I'm sorry. I'm going to close the request. 

Thank you again

closed 2023-08-08 by frax84