zephyrproject-rtos/zephyr #61246: vmu_rt1170:tests/net/socket/socketpair/net.socket.socketpair.newlib

bperseghetti, 2023-08-08:
**Describe the bug**
The test has failed repeatedly with:

`
/opt/toolchains/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: scripts/build/gen_kobject_placeholders.py did not reserve enough space for kobject rodata.
`

https://github.com/zephyrproject-rtos/zephyr/actions/runs/5790244946/job/15692992818
https://github.com/zephyrproject-rtos/zephyr/actions/runs/5790244946/job/15697429521

Can't reproduce locally.

Wondering if it's because the test plan only has [min_ram: 21](https://github.com/zephyrproject-rtos/zephyr/blob/1e58610b2a9666db910165697d29d8b070acacf9/tests/net/socket/socketpair/testcase.yaml#L7)

**To Reproduce**
No idea, can't reproduce it locally, these pass just fine on local system:

`west twister -v -p vmu_rt1170 -s tests/net/socket/socketpair/net.socket.socketpair`
`west twister -v -p vmu_rt1170 -s tests/net/socket/socketpair/net.socket.socketpair.newlib`

**Expected behavior**
A green pass in CI.

**Impact**
Blocking PR CI.

**Logs and console output**
https://github.com/zephyrproject-rtos/zephyr/actions/runs/5790244946/job/15692992818
https://github.com/zephyrproject-rtos/zephyr/actions/runs/5790244946/job/15697429521

bperseghetti, 2023-08-08:
@manuargue here is the issue, I'm going to exclude the platform from this test for now.

fabiobaltieri, 2023-08-08:
@bperseghetti what version of Zephyr SDK are you running? I was able to repro with 16.1

fabiobaltieri, 2023-08-08:
I added some more info on the other bug, let's close this one and keep https://github.com/zephyrproject-rtos/zephyr/issues/61129, it's likely the same issue.

closed 2023-08-08 by fabiobaltieri