zephyrproject-rtos/zephyr #61248: mgmt: mcumgr: improve custom payload types

kunoh, 2023-08-08:
**Is your enhancement proposal related to a problem? Please describe.**

The SMP Protocol allow users to add their own custom groups (user groups) after group 64, using any type of data as payload. Currently the implementation is tightly coupled with cbor and adds a start and end byte in the payload which happens outside of the users scope. The start byte indicates the cbor format used in the payload, and is not changeable from userspace.

As a user, I would like to add custom handlers that uses Protobuf as the payload format, but I would now have to ignore the 2 byte overhead start- and end byte as they are pointless to custom payload format.

**Describe the solution you'd like**

Extend the Major Types to allow users to add their own payload types, currently described in `c:struct:zcbor_major_type_t` and give the user access to set the start byte using the relevant payload type.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

nordicjm, 2023-08-08:
There was a previous PR about this before the transition to zcbor, it is worth looking at again. Though errors would still be reported in zcbor format.

nordicjm, 2023-08-08:
https://github.com/zephyrproject-rtos/zephyr/pull/42611

kunoh, 2023-08-08:
Thank you, will have a look :-)

PR #61318 (2023-12-11): mgmt: mcumgr: allow for user defined payload
Use a flag in `c:struct:mgmt_handler` to skip the cbor start and end byte, and instead use pure custom user defined payload.

Fixes #61248
closed 2023-12-11 by carlescufi