zephyrproject-rtos/zephyr #61249: mgmt: mcumgr: add support for indications/notifications

kunoh, 2023-08-08:
**Is your feature request related to a problem? Please describe.**

The current implementation of the SMP only allows for handling requests coming from a client. The server side may want to inform the client about a time critical issue but would have to wait until a client poll for that specific request.

A device using the SMP server may go to error state and would therefore like to inform the client, by sending either a notification or indication, that it is in error state.

**Describe the solution you'd like**
Open up for an api a user can use to send notifications/indication through the SMP server

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

nordicjm, 2023-08-08:
Asynchronous responses are not supported and will not be supported due to issues noticed when testing it before whereby flash is stalled during a flash erase/program for an extended period of time and commands sent during that time (i.e. over UART) are corrupted.

nordicjm, 2023-08-08:
Though as an alternative system, there is now MCUmgr client support in-tree with implementations for os_mgmt and img_mgmt so you could use that in an application, it can act as a server and respond to incoming queries and act as a client to asynchronously send commands to the other device (which would be in server mode)

kunoh, 2023-08-08:
What about in a synchronous way where the SMP server will handle each (request/response, indication, notification) in a blocking way?

nordicjm, 2023-08-08:
There would be nothing to do as that is already supported, MCUmgr client writes to characteristic with data, MCUmgr server processes command, then writes response back to client via characteristic write. Indication/notification isn't going to improve anything

kunoh, 2023-08-08:
Okay, can the client and server co-exist?

nordicjm, 2023-08-08:
Yes

kunoh, 2023-08-08:
Cool, I will try to work out a solution using the smp client also, thank you :)

closed 2023-08-08 by kunoh