zephyrproject-rtos/zephyr #61253: Zephyr HTTP Client to HTTP Server

hemanth4228, 2023-08-08:
Facing Issue on Zephyr HTTP client connection with HTTP Server. 

**Describe the bug**
- I'm trying to connect Zephyr HTTP client to HTTP server in Ubuntu. 
- Target platform: **native_posix**
- I'm using Zephyr official provided information as the reference. 
[https://docs.zephyrproject.org/latest/samples/net/sockets/http_client/README.html](url)
- I sat up the virtual network between Linux host and Zephyr application. 
[https://docs.zephyrproject.org/latest/connectivity/networking/native_posix_setup.html#networking-with-native-posix](url)

**To Reproduce**
Steps to reproduce the behavior:
1. In terminal 1, 
   cd net-tools/ 
   ./net-setup.sh
2. In terminal 2,
   west build -b native_posix samples/net/sockets/echo_server
   west build -t run
3. In another terminal, I'm executing the below commands, once I entered into Zephyr source environment.
   source ~/zephyrproject/.venv/bin/activate
   cd zephyrproject/zephyr/
   west build -b native_posix samples/net/sockets/http_client -- -DCONF_FILE=prj.conf
   west build -b native_posix samples/net/sockets/http_client -- -DCONF_FILE="prj.conf overlay-tls.conf" (For enabling TLS suport)
   west build -t run     
   

**Logs and console output**
Referred to above steps,
* In step 2 I'm getting the below output. 
-- west build: running target run
[0/1] cd /home/hemanth/zephyrproject/z...project/zephyr/build/zephyr/zephyr.exe
WARNING: Using a test - not safe - entropy source
uart connected to pseudotty: /dev/pts/2

* And in step 3, I'm getting the below output on the console. 
-- west build: running target run
[0/1] cd /home/hemanth/zephyrproject/z...project/zephyr/build/zephyr/zephyr.exe
WARNING: Using a test - not safe - entropy source
uart connected to pseudotty: /dev/pts/5
Segmentation fault (core dumped)
FAILED: zephyr/CMakeFiles/run_native /home/hemanth/zephyrproject/zephyr/build/zephyr/CMakeFiles/run_native 
cd /home/hemanth/zephyrproject/zephyr/build && /home/hemanth/zephyrproject/zephyr/build/zephyr/zephyr.exe
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /bin/cmake --build /home/hemanth/zephyrproject/zephyr/build --target run

**Environment:**
 - OS: Ubuntu 20.04 LTS (OS installed in virtual machine on windows 11)
 - Toolchain: zephyr-sdk-0.16.1

**Additional context**
Please look into my issue and help me out to overcome this. 
Thanks in advance.

rlubos, 2023-08-09:
A few issues:
* `echo_server` sample is **not** an HTTP server - there's no point running it against HTTP client, as it **won't** send HTTP response, just echo the request back
* If you want to bridge mutiple `native_posix` instances over a network, you should follow https://docs.zephyrproject.org/latest/connectivity/networking/networking_with_multiple_instances.html# and not https://docs.zephyrproject.org/latest/connectivity/networking/native_posix_setup.html - the latter is for single instance only.
* For `http_client` sample, you need to setup a local HTTP server, as described in the documentation: https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/net/sockets/http_client#running-http-server-in-linux-host
For now, it seems that the sample simply exits due to connection error.

carlescufi, 2023-08-16:
@hemanth4228 please follow @rlubos advice to proceed.

closed 2023-08-16 by carlescufi