zephyrproject-rtos/zephyr #61284: Docleaf tool is using a non-OSI approved license

kartben, 2023-08-08:
**Describe the bug**

#59570 introduced Docleaf as a replacement for Breathe to make Doxygen documentation accessible from Sphinx.
As opposed to Breathe which was BSD licensed, Docleaf is published under the [Parity Public License](https://paritylicense.com/versions/7.0.0). This license is not OSI approved and potentially problematic for downstream Zephyr adopters that may have closed source bits, as they would depend on acquiring a commercial license for being able to keep using the Zephyr documentation system. The [project's README](https://github.com/docleaf-labs/docleaf#license) makes this even more clear: "If you have a closed source project that you would like to document with Docleaf then you must purchase a commercial license"

This was an oversight from the reviewers of the original PR (myself included) to not look more closely at the license of Docleaf, as I guess (for me at least) the assumption was that it was a continuation/replacement for Breathe and hence would have the same license.

**Impact**

See above. The Parity Public License is implying that anyone "develop[ing], operat[ing], or analyz[ing] with this software [ie. Docleaf]" needs to make their software available under an open source license such as APL, MIT, ... ("a license that allows everything [the Parity Public License] does").
This is putting unwanted restrictions on how downstream adopters may be able to freely extend Zephyr for commercial / closed source purposes should they be interested in surfacing documentation for closed-source stuff.

I see a few paths going forward (assuming I understood the terms of the license properly, and that Docleaf sticks to it for the foreseeable future):

- switch back to Breathe. It's slower than Docleaf, and potentially unmaintained/less maintained going forward (?) but it's been doing the job until now.
- should Docleaf maintain some level of compatibility with Breathe, we might be able to provide both options? But this would need to be tested in CI and I am not sure it'd be desirable to systematically spend 2x the already long-ish time to build our docs in CI for each PR...

**Environment (please complete the following information):**

3.4.99 / 84e4ffcd56348f96ea532ece89f5eda9c62e8c56

**Additional context**

@michaeljones, feel free to add/comment here, especially if I misunderstood the terms of the license.

cc @nashif @gmarull

michaeljones, 2023-08-08:
@kartben - I think you have correctly understood the license. I made an effort to flag the license as a potential issue in the PR and with the people I first approached. I realise it might be an issue.

The choice of license is deliberate and aimed to create a more sustainable project in the long run even at the cost of adoption. I realise that it means Docleaf might not meet the needs of the Zephyr project and, if so, should be removed.

kartben, 2023-08-08:
Thanks @michaeljones for the crystal clear and prompt reply. I certainly get where you're coming from regarding trying to find a sustainable model for your project, and you did indeed flag the license when submitting the PR, so no worries at all. FTYI we'll probably have this as a topic for tomorrow's Zephyr Technical Steering Committee meeting.

kartben, 2023-08-08:
> I'd personally recover the idea of #39702 :-)

Interesting, thanks for digging this up. I like the idea. 

> * Doxygen is the place to browse API documentation

I'd tend to agree, yes. If we'd look at making Doxygen more of a first-class citizen, I think issues such as #56855 should be addressed too. 
Today, the entire contents of include/ (and then some) is included in the generated docs, even when it has no Doxygen comments. We'd want to rise the bar for Doxygen documentation so that to remove some of the existing clutter that we have today (ex. Doxgen search may yield results for stuff no one cares or should know about) if more people are going to use it as  a more primary source of truth for navigating API documentation.

> * Sphinx is kept for contextual information, guides, interactive stuff, etc. Some details now in Sphinx could also be moved to Doxygen to improve the user experience (also when just browsing header files!), see e.g. https://docs.zephyrproject.org/latest/doxygen/html/group__linear__range.html#details

Yep! Slightly related, I would love to see header files point to relevant code samples (#60647), to make it easier to take people back to the Sphinx website when needing the more contextualized experience.

> Sphinx + Doxygen will always be much slower than Doxygen alone (which is incredibly fast). Another side effect is that PDF is unusable, in part because there are tons of pages with API docs.

Do you know how good/bad the Doxygen PDF output currently looks like for us?

kartben, 2023-08-08:
> > Now PDF seems to be missing https://docs.zephyrproject.org/latest/zephyr.pdf
> 
> Oops :/ It's actually an interesting data point that no one noticed or at least reported it until now...

Interestingly it seems like it's the introduction of Docleaf that broke the PDF build, as it started failing then https://github.com/zephyrproject-rtos/zephyr/actions/runs/5444198760
Seems like an exception is being thrown somewhere in docutils. FWIW it seems that both breathe and docleaf are pulling docutils-0.18.1.

nashif, 2023-08-09:
> I'd personally recover the idea of #39702 :-)

yes, maybe it is time to revisit all of this and focus on making doxygen output usuable and complete, someting @kartben  already doing

kartben, 2023-08-09:
> Sorry I meant we (and by we I mean the safety people, but I might be wrong as maybe "source" documentation is an OK deliverable) would probably want a Latex/PDF of the Doxygen documentation, and my question was then more along the lines of "does it work today, and if yes how good does it look styling wise etc"

FWIW enabling `GENERATE_LATEX` in zephyr.doxyfile leads to tons of warnings and eventually an error when trying to get a PDF.

nashif, 2023-08-09:
> FWIW enabling `GENERATE_LATEX` in zephyr.doxyfile leads to tons of warnings and eventually an error when trying to get a PDF.

I do not see a need for a PDF version of the doxygen API docs, that makes sense for some usecases for the rst docs. API docs should be fine to be packed and published as html IMO.

nashif, 2023-08-09:
The TSC has voted to revert the change and move back to breathe

kartben, 2023-08-09:
Working on reverting docleaf introduction as part of (currently draft) PR #61339

kartben, 2023-08-09:
> Another options brought by @jfischer-no  was using kernel-doc from linux, to replace doxygen completely: https://docs.kernel.org/doc-guide/kernel-doc.html. Not my favorite, but can be looked at in case doxygen is seen as the problem.

**If** we still want a solution where API documentation get embedded in main documentation then I am not sure the Linux kernel approach would get my vote. 

Doxygen comment styles have the merit of being an industry standard and the Doxygen itself is actively maintained. 

The custom Perl parser that Linux has for turning C comments (again, non "standardized" ones) into ReStructuredText is super scary IMO -- see https://github.com/torvalds/linux/blob/master/scripts/kernel-doc -- and while it seems to get the job done for adding API docs to their Sphinx website, it is also pretty limited (ex. the doc is not "navigable" and you can't e.g. click on a function parameter to get to its type definition).

nashif, 2023-08-09:
> **If** we still want a solution where API documentation get embedded in main documentation then I am not sure the Linux kernel approach would get my vote.

neither it will get mine. I just mentioned that for completeness here.

PR #61339 (2023-08-24): doc: Revert back to using breathe instead of docleaf
This reverts the subset of commits from #59570 that introduced docleaf and fixes #61284.

- Revert "doc: Switch to using docleaf for doxygen entities"
- Revert "doc: Update requirements-doc.txt to use docleaf"
- Revert "doc: Replace instances of 'Breathe' with 'Docleaf'"
- Revert "doc: Adjust docleaf configuration as needed"
- Revert "doc: Remove :members: on doxygengroup directives"

Also added a commit marking some apparently unsupported and recently added constructs in `wifi.h` (namely, arrays with designated initializers) as INTERNAL_HID [... cut]
closed 2023-08-24 by fabiobaltieri