zephyrproject-rtos/zephyr #61287: Bluetooth: CIS accepted by Host gets rejected by LL

MariuszSkamra, 2023-08-08:
**Describe the bug**
The CIS that has been accepted by Host, gets rejected by Controller with `Unsupported Link Layer Parameter Value`. The controller does not generate `LE CIS Established` event to the host, thus Host keeps the reference to the connection that has not been established.

**Expected behavior**
The parameters should be validated before `LE Accept Connected Isochronous Stream Request` is sent to host.
On the other hand, in case the Host accepts the CIS, but the CIS was not created for some reason, the `LE CIS Established` event shall be generated to inform the host, so that it can release it's resources.

e.g.
[image]

**Logs and console output**
The Host accepts the CIS Request: 
```
< ACL Data TX: Handle 0 flags 0x00 dlen 12                                                                    #467 102.477800
      ATT: Handle Value Notification (0x1b) len 7
        Handle: 0x0014
          Data: 0301010000
        ASCS: Audio Stream Endpoint Control Point (Handle: 0x0014)
        Operation response: Enable
        Number of ASEs: 1
         ASE ID: 1
          Response code: Success
< ACL Data TX: Handle 0 flags 0x00 dlen 16                                                                    #468 102.478300
      ATT: Handle Value Notification (0x1b) len 11
        Handle: 0x0017
          Data: 010301010403020600
        ASCS: Audio Stream Sink Endpoint (Handle: 0x0017)
        ASE ID: 1
        Enabling
          CIG: 1
          CIS: 1
          Metadata: 0x03020600
> HCI Event: LE Meta Event (0x3e) plen 7                                                                      #471 102.805300
      LE Connected Isochronous Stream Request (0x1a)
        ACL Handle: 0
        CIS Handle: 6
        CIG ID: 0x01
        CIS ID: 0x01
< HCI Command: LE Accept Connected Isochronous Stream Request (0x08|0x0066) plen 2                            #472 102.805500
        CIS Handle: 6
> HCI Event: Command Status (0x0f) plen 4                                                                     #473 102.805700
      LE Accept Connected Isochronous Stream Request (0x08|0x0066) ncmd 1
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4                                                                #474 104.443600
        Status: Success (0x00)
        Handle: 0
        Reason: Remote User Terminated Connection (0x13)
```

The controller rejects the LL_CIS_REQ
[image]

The parameters look as follows
[image]

[LL_CIS_REQ_rejected.btt.zip](https://github.com/zephyrproject-rtos/zephyr/files/12290967/LL_CIS_REQ_rejected.btt.zip)
[BAP_USR_STR_BV_367_C_2023_08_08_13_48_46.zip](https://github.com/zephyrproject-rtos/zephyr/files/12291520/BAP_USR_STR_BV_367_C_2023_08_08_13_48_46.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Execute PTS test case BAP_USR_STR_BV_367_C

zephyr-v3.4.0-2075-g32bd3aa781

Thalley, 2023-08-15:
Duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/59173?

Thalley, 2023-10-20:
@MariuszSkamra Is this a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/59173? Can it be closed? If not, please respond and remove the `Stale` label.

closed 2023-10-20 by MariuszSkamra