zephyrproject-rtos/zephyr #61293: Networking: net_tcp_connect() sets incorrect source IP address

Frauschi, 2023-08-08:
**Describe the bug**
I have a setup with multiple IP addresses on one network interface. The addresses are within the same IP address range (e.g., `192.168.0.10` and `192.168.0.11`). I want to establish a new TCP connection to a remote peer on the same network (e.g., `192.168.0.1`) using the Sockets API. I have to make sure that the TCP connection is definitely using the IP address `192.168.0.10` as the outgoing source IP. To achieve this, I call `bind()` before `connect()` to bind the socket to the desired IP address. However, the network stack always selects the other IP address as the source. 

I traced the bug down to the call to `net_tcp_connect()` in this line: https://github.com/zephyrproject-rtos/zephyr/blob/d821655def365f85c805b4b1c375484bab64ed70/subsys/net/ip/tcp.c#L3068C3-L3071C33
Due to the call to `bind()`, the `local_addr` parameter is already correctly set to the desired IP address. However, the source IP parameter of the TCP connection object (`conn->src.sin.sind_addr`) is always set to the result of the call to `net_if_ipv4_select_src_addr()`, with internally selects the other IP address of the interface as the “best match”. This results in the incorrect IP address within the TCP SYN packet.
 
**To Reproduce**
The following code demonstrates the incorrect behavior (the first IP address of the device is set using `CONFIG_NET_CONFIG_MY_IPV4_ADDR` in Kconfig to, e.g., `192.168.0.10`; Error handling is left out for clarity):
```
struct net_if* iface = net_if_get_default();
char const* local_ip_1 = CONFIG_NET_CONFIG_MY_IPV4_ADDR;
char const* local_ip_2 = "192.168.0.11";
char const* remote_ip = "192.168.0.1";
int port = 8080;

struct sockaddr_in bind_addr = {
	.sin_family = AF_INET,
	.sin_port = 0
};
struct sockaddr_in remote_addr = {
	.sin_family = AF_INET,
	.sin_port = htons(port)
};
struct sockaddr_in tmp_addr = {
	.sin_family = AF_INET,
	.sin_port = 0
};

net_addr_pton(bind_addr.sin_family, local_ip_1, &bind_addr.sin_addr);
net_addr_pton(remote_addr.sin_family, remote_ip, &remote_addr.sin_addr);
net_addr_pton(tmp_addr.sin_family, local_ip_2, &tmp_addr.sin_addr);

net_if_ipv4_addr_add(iface, &tmp_addr.sin_addr, NET_ADDR_MANUAL, 0);

int sock = socket(remote_addr.sin_family, SOCK_STREAM, IPPROTO_TCP);

bind(sock, (struct sockaddr*) &bind_addr, sizeof(bind_addr));

connect(sock, (struct sockaddr*) &remote_addr, sizeof(remote_addr));
```
The board sends a TCP SYN packet to `192.168.0.1` with the source IP address set to `192.168.0.11`

**Expected behavior**
The board sends a TCP SYN packet to `192.168.0.1` with the source IP address set to `192.168.0.10`

**Potential fix**
With a small code change, the bug is fixed, and the proper IP address is used when set using `bind()`. In the function  `net_tcp_connect()`, we check if the `local_addr` parameter is set to a valid address or not before setting `conn->src.sin.sind_addr`. In case `bind()` is not called before `connect()`, the socket is bound to `INADDR_ANY` from the call to `bind_default()` within `net_context_connect()`. Hence, `local_addr` contains the unspecified address with all zeros. Otherwise, `local_addr` contains the desired valid IP address. 

The following code snippet fixes the bug:
```
if (net_ipv4_is_addr_unspecified(&net_sin(local_addr)->sin_addr)) {
	ip4 = net_if_ipv4_select_src_addr(
	net_context_get_iface(context),
	&net_sin(remote_addr)->sin_addr);
	conn->src.sin.sin_addr = *ip4;
} else {
	net_ipaddr_copy(&conn->src.sin.sin_addr,
		&net_sin(local_addr)->sin_addr);
}
net_ipaddr_copy(&conn->dst.sin.sin_addr,
	&net_sin(remote_addr)->sin_addr);
```
This shouldn't produce any problems, as the `local_addr` parameter is properly set by the calling function.

rlubos, 2023-08-10:
Thanks @Frauschi for the detailed analysis. The solution you propose makes perfect sense to me (we should apply similar change for IPv6 address as well). Are you willing to send a PR with the fix?

Frauschi, 2023-08-10:
Hi @rlubos. Thanks for the feedback. I just submitted a PR with the fix for both IPv4 and IPv6.

PR #61390 (2023-08-15): net: tcp: Set correct source IP address when connecting
In case a TCP connection should be established with a specific source IP address (e.g., by calling bind() prior to connect() on the socket), we have to actually place the address in the outgoing packet.

Fixes #61293
closed 2023-08-15 by fabiobaltieri