zephyrproject-rtos/zephyr #61302: LVGL sample does not work with MXChip / AZ3166 (on Mac M1)

danuw, 2023-08-08:
**Describe the bug**
I am trying the sample with the MXChip (aka AZ3166 IoTDevKit). I have tested some other basic samples successfully such as the temperature one which showed in the serial monitor in VS Code.
That board's screen seems to be a ssd1306_128x64.
I have tried the LVGL sample to use the display, but setting it for that screen type errors. 
I was however able to compile with that shield. Am I missing a configuration (as it seems to say "configuration incomplete")?

**To Reproduce**

First for the mxchip to work I had to install `bossa` and `open-ocd` (using brew, e.g. `brew install bossa`)

Steps to reproduce the behavior:
1. Browse to lvgl sampel folder`cd ~/zephyrproject/zephyr/samples/subsys/display/lvgl`
2. Build the solution for that board and that screen shield `west build -b az3166_iotdevkit . -- -DSHIELD=ssd1306_128x64`
3. Build errors (see below for output)

I have also tried without the DSHIELD param and it compiles successfully and flashing shows a screen that is blue and orange (with missing pixels, but that does not show "Hello World".
I have also deleted the build folder as well as used the `--pristine` parameter while building to ensure this is a fresh build (as simply adding the DSHIELD after a successful build without does not error but has the same failing outcome as described above).

**Expected behavior**
I expected that after step 2 where it builds, the build would be successful, I would flash the board and `hello world` would show on the screen.

**Impact**
This is the only board I have available with a screen at the moment to run my experiments and start prototyping some scenarios.

**Logs and console output**

```sh 
lvgl % west build -b az3166_iotdevkit . -- -DSHIELD=ssd1306_128x64           
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /Users/dan.benitah/zephyrproject/zephyr/samples/subsys/display/lvgl
-- CMake version: 3.26.4
-- Found Python3: /Users/dan.benitah/.pyenv/versions/3.10.8/envs/zephyrmxchip/bin/python3.10 (found suitable exact version "3.10.8") found components: Interpreter 
-- Cache files will be written to: /Users/dan.benitah/Library/Caches/zephyr
-- Zephyr version: 3.3.99 (/Users/dan.benitah/zephyrproject/zephyr)
-- Found west (found suitable version "1.0.0", minimum required is "0.7.1")
-- Board: az3166_iotdevkit
-- Shield(s): ssd1306_128x64
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.1 (/Users/dan.benitah/.local/opt/zephyr-sdk-0.15.1)
-- Found toolchain: zephyr 0.15.1 (/Users/dan.benitah/.local/opt/zephyr-sdk-0.15.1)
-- Found Dtc: /opt/homebrew/bin/dtc (found suitable version "1.7.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /Users/dan.benitah/zephyrproject/zephyr/boards/arm/az3166_iotdevkit/az3166_iotdevkit.dts
-- Found devicetree overlay: /Users/dan.benitah/zephyrproject/zephyr/boards/shields/ssd1306/ssd1306_128x64.overlay
devicetree error: /Users/dan.benitah/zephyrproject/zephyr/boards/shields/ssd1306/ssd1306_128x64.overlay:13 (column 1): parse error: undefined node label 'arduino_i2c'
-- In: /Users/dan.benitah/zephyrproject/zephyr/samples/subsys/display/lvgl/build/zephyr, command: /Users/dan.benitah/.pyenv/versions/3.10.8/envs/zephyrmxchip/bin/python3.10;/Users/dan.benitah/zephyrproject/zephyr/scripts/dts/gen_defines.py;--dts;/Users/dan.benitah/zephyrproject/zephyr/samples/subsys/display/lvgl/build/zephyr/zephyr.dts.pre;--dtc-flags;'';--bindings-dirs;/Users/dan.benitah/zephyrproject/zephyr/dts/bindings;--header-out;/Users/dan.benitah/zephyrproject/zephyr/samples/subsys/display/lvgl/build/zephyr/include/generated/devicetree_generated.h.new;--dts-out;/Users/dan.benitah/zephyrproject/zephyr/samples/subsys/display/lvgl/build/zephyr/zephyr.dts.new;--edt-pickle-out;/Users/dan.benitah/zephyrproject/zephyr/samples/subsys/display/lvgl/build/zephyr/edt.pickle;--vendor-prefixes;/Users/dan.benitah/zephyrproject/zephyr/dts/bindings/vendor-prefixes.txt
CMake Error at /Users/dan.benitah/zephyrproject/zephyr/cmake/modules/dts.cmake:261 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  /Users/dan.benitah/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:115 (include)
  /Users/dan.benitah/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /Users/dan.benitah/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /opt/homebrew/bin/cmake -DWEST_PYTHON=/Users/dan.benitah/.pyenv/versions/3.10.8/envs/zephyrmxchip/bin/python3.10 -B/Users/dan.benitah/zephyrproject/zephyr/samples/subsys/display/lvgl/build -GNinja -DBOARD=az3166_iotdevkit -DSHIELD=ssd1306_128x64 -S/Users/dan.benitah/zephyrproject/zephyr/samples/subsys/display/lvgl
```

**Environment (please complete the following information):**
 - OS: MacOS M1/arm64
 - Toolchain : I believe I am using `West version: v1.0.0` 
 - I seem to be using `commit 7b3db7c15d1d9ef0c41bbd80357751ade691cfa4`

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

kartben, 2023-08-08:
You would have to set LV_COLOR_DEPTH to 1 but IIRC this is probably not enough, and monochrome displays can be painful with LVGL.
Can you confirm that you're able to get the cfb sample to run though? https://docs.zephyrproject.org/latest/samples/subsys/display/cfb/README.html

kartben, 2023-08-08:
You would have to set LV_COLOR_DEPTH to 1 but IIRC this is probably not enough, and monochrome displays can be painful with LVGL.
Can you confirm that you're able to get the cfb sample to run though? https://docs.zephyrproject.org/latest/samples/subsys/display/cfb/README.html -- see https://youtu.be/GiDsMlRipJY?t=439

kartben, 2023-08-08:
Also you don't need to mess with shields as the Devicetree for the AZ3166 already defines a `zephyr_display` pointing to a properly configured SSD1306, so e.g. a sample such as the cfb one should work out of the box.