zephyrproject-rtos/zephyr #61312: bug: CANFD: mimxrt1160_evk_cm7 :  build failure after switch to the nxp,flexcan-fd

hakehuang, 2023-08-09:
**Describe the bug**
mimxrt1160_evk_cm7 build failure.

bisect find this is caused by
```
commit 41e0a2e9df5c3c6de33ce16b84e5401d4b7a2986
Author: Benjamin Perseghetti <bperseghetti@rudislabs.com>
Date:   Mon Aug 7 11:10:25 2023 -0400

    soc: nxp_rt11xx: add CANFD compatible.

    Enable CANFD for rt11xx by including nxp,flexcan-fd
    compatibility for all CANFD capable CAN with associated
    properties.

    Signed-off-by: Benjamin Perseghetti <bperseghetti@rudislabs.com>
```

**To Reproduce**
Steps to reproduce the behavior:
1. scripts/twister -p mimxrt1160_evk_cm7 -T tests/kernel/ --build-only
4. See error

**Expected behavior**
build pass

**Impact**
mimxrt1160_evk_cm7 fails for all builds

**Logs and console output**

```
Loading Zephyr module(s) (Zephyr base): zephyr_default:kconfig
-- Application: /home/shared/disk/zephyr_project/zephyr_dev/zephyr/tests/kernel/workq/critical
-- CMake version: 3.24.1
-- Found Python3: /usr/bin/python3.10 (found suitable exact version "3.10.6") found components: Interpreter
-- Cache files will be written to: /home/ubuntu/.cache/zephyr
-- Zephyr version: 3.4.99 (/home/shared/disk/zephyr_project/zephyr_dev/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.14.0")
-- Board: mimxrt1160_evk_cm7
-- Found host-tools: zephyr 0.16.2 (/home/ubuntu/zephyr-sdk/zephyr-sdk-0.16.2-rc1)
-- Found toolchain: zephyr 0.16.2 (/home/ubuntu/zephyr-sdk/zephyr-sdk-0.16.2-rc1)
-- Found Dtc: /home/ubuntu/zephyr-sdk/zephyr-sdk-0.16.2-rc1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/shared/disk/zephyr_project/zephyr_dev/zephyr/boards/arm/mimxrt1160_evk/mimxrt1160_evk_cm7.dts
devicetree error: 'bus-speed-data' is marked as required in 'properties:' in /home/shared/disk/zephyr_project/zephyr_dev/zephyr/dts/bindings/can/nxp,flexcan-fd.yaml, but does not appear in <Node /soc/can@40c3c000 in '/home/shared/disk/zephyr_project/zephyr_dev/zephyr/misc/empty_file.c'>
-- In: /home/shared/disk/zephyr_project/zephyr_dev/zephyr/twister-out/mimxrt1160_evk_cm7/tests/kernel/workq/critical/kernel.workqueue.critical/zephyr, command: /usr/bin/python3.10;/home/shared/disk/zephyr_project/zephyr_dev/zephyr/scripts/dts/gen_defines.py;--dts;/home/shared/disk/zephyr_project/zephyr_dev/zephyr/twister-out/mimxrt1160_evk_cm7/tests/kernel/workq/critical/kernel.workqueue.critical/zephyr/zephyr.dts.pre;--dtc-flags;'';--bindings-dirs;/home/shared/disk/zephyr_project/zephyr_dev/zephyr/dts/bindings;--header-out;/home/shared/disk/zephyr_project/zephyr_dev/zephyr/twister-out/mimxrt1160_evk_cm7/tests/kernel/workq/critical/kernel.workqueue.critical/zephyr/include/generated/devicetree_generated.h.new;--dts-out;/home/shared/disk/zephyr_project/zephyr_dev/zephyr/twister-out/mimxrt1160_evk_cm7/tests/kernel/workq/critical/kernel.workqueue.critical/zephyr/zephyr.dts.new;--edt-pickle-out;/home/shared/disk/zephyr_project/zephyr_dev/zephyr/twister-out/mimxrt1160_evk_cm7/tests/kernel/workq/critical/kernel.workqueue.critical/zephyr/edt.pickle;--edtlib-Werror;--vendor-prefixes;/home/shared/disk/zephyr_project/zephyr_dev/zephyr/dts/bindings/vendor-prefixes.txt
CMake Error at cmake/modules/dts.cmake:276 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  cmake/modules/zephyr_default.cmake:115 (include)
  share/zephyr-package/cmake/ZephyrConfig.cmake:75 (include)
  share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  cmake/package_helper.cmake:115 (find_package)

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux,)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v3.4.0-2112-g9c3f9dc1ec

dleach02, 2023-08-09:
@bperseghetti, can you take a look at this since your PR was identified in the bisect

dleach02, 2023-08-09:
fixed by #61307

closed 2023-08-09 by dleach02