zephyrproject-rtos/zephyr #61319: Zephyr RTOS: Getting bind error with respect to sockets

sanjatpanigrahi, 2023-08-09:
Hi,
I am trying to run a simple socket program, the below are my environment:

i. Intel Agilex 7 FPGA board
ii. Zephyr RTOS

I am able to successfully create a socket but get the below error is generated during the binding process:
(I am printing the error number and the error converted through stderrr())

ErrNo= -1, Error = Not owner 

Can anyone please tell me why the binding is throwing an error "Not owner"?

Below are my socket configuration parameters in the pro.conf file:

# Sockets
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_SOCKETS_POLL_MAX=4
CONFIG_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y

Networking configuration parameters:
CONFIG_NETWORKING=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n
CONFIG_NET_TCP=y
CONFIG_NET_UDP=y
CONFIG_NET_ARP=y
CONFIG_SLIP_STATISTICS=n
### CONFIG_NET_SHELL=y

Best Regards,
Sanjat

rlubos, 2023-08-10:
@sanjatpanigrahi I'm not sure if you're checking the error code correctly, `-1` suggests that you're only providing the return value from `bind()` (`-1` is returned in case of **any** error). For the actual error code, you need to check the `errno` variable after `bind()` function returns `-1`.

Also, it'd be good to see some sample code on how are you actually using `bind()`.

sanjatpanigrahi, 2023-08-10:
@rlubos I do check the actual error returned from bind(). Below is the code snippet:

int sockfd, connfd;                                                                                                                                    
unsigned int len;                                                                                                                                      
struct sockaddr_in servaddr, cli;                                                                                                                      
// socket create and verification                                                                                                                      
sockfd = socket(AF_INET, SOCK_STREAM, IPPROTO_TCP)                                                         

if (sockfd == -1)                                                                                                                                      
die("socket creation failed...");                                                                                                                  
else                                                                                                                                                   
printk("Socket successfully created..\n");

// assign IP, PORT 
servaddr.sin_family = AF_INET; // IPv4
servaddr.sin_addr.s_addr = htonl(INADDR_ANY); /* host-to-network endian */
servaddr.sin_port = htons(PORT); /* for listening */

bind_status = bind(sockfd, (struct sockaddr *)&servaddr, sizeof(servaddr));
if(bind_status < 0))
printk("Error number = %d, Error status = %s \n", bind_status, strerror(-bind_status)); 

Few more information:

I am running my application program compiled with Zephyr RTOS below is my command:

cmake -GNinja -DBOARD=intel_socfpga_agilex_socdk ../../samples/server
ninja

I flash the Zephyr RTOS on an Agilex 7 Intel FPGA having an ARM Cortex A53 processor on which first the u-boot runs. Then using ARMDS IDE I flash the "zephyr.elf" which got generated with the above command.

If instead of using Zephyr RTOS, I use Linux on the FPGA and run my program it works. This means, the FPGA board is not broke and the networking works.

I am facing bind error only when I use the Zephyr RTOS to run my application.

sanjatpanigrahi, 2023-08-10:
phooo... how did I miss it!!!!
Thanks a lot for sharing the webpage.

Does the below:
bind_result = -1, bind_error_number = 125 Error = Address not available

does the above means that the Zephyr RTOS doesn't enables the networking stuffs?
If so, then shouldn't the socket() should also return error?

rlubos, 2023-08-11:
> does the above means that the Zephyr RTOS doesn't enables the networking stuffs?

Since you don't request any particular address to bind to (`INADDR_ANY`), this means there isn't **any** netowrk interface in the system. I. e. the board you use does not define any network device. 

> If so, then shouldn't the socket() should also return error?

During socket creation, it is not verified whether there are any network interfaces in the system (because at that point socket is not bound to any). But any operation that would require to bind it with particular interface (bind(), send() etc.) would fail.