zephyrproject-rtos/zephyr #61320: Update hal_st/sensor/vl53l1x from version 2.3.3 to the current 2.4.5

boaks, 2023-08-09:
That update requires mainly to copy the files and adapt the code to one API change:

[VL53L1_Error VL53L1_GetTickCount(uint32_t *ptime_ms)](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/sensor/vl53l1x/vl53l1_platform.h#L123-L129)

to

`VL53L1_Error VL53L1_GetTickCount(VL53L1_Dev_t *pdev, uint32_t *ptime_ms);`

boaks, 2023-08-09:
Not sure, if I choose the right category for issues.
The [zephyrproject-rtos/hal_st](https://github.com/zephyrproject-rtos/hal_st) doesn't offer means to request that update.

FRASTM, 2023-08-09:
Preparing a PR  https://github.com/zephyrproject-rtos/hal_st/pull/17  to update  [hal_st/sensor/vl53l1x](https://github.com/zephyrproject-rtos/hal_st/tree/master/sensor/vl53l1x)  with the v2.4.5

I realize the API is   ` VL53L1_GetTickCount(Dev, &current_time_ms);  ` in the new v2.4.5   of _../modules/hal/st/sensor/vl53l1x/api/core/src/vl53l1_wait.c_,   but also found that it is still `VL53L1_Error VL53L1_GetTickCount(uint32_t *ptime_ms);` in the _https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/sensor/vl53l1x/vl53l1_platform.h_

Which is right ?

FRASTM, 2023-08-09:
Changed in https://github.com/zephyrproject-rtos/hal_st/pull/17
also requires an update in **west.yml**
```
diff --git a/west.yml b/west.yml
index 6fb0fb7052..0c0e88137f 100644
--- a/west.yml
+++ b/west.yml
@@ -225,7 +225,7 @@ manifest:
         - hal
     - name: hal_st
       revision: 5948f7b3304f1628a45ee928cd607619a7f53bbb
-      path: modules/hal/st
+      path: pull/17/head
       groups:
         - hal
     - name: hal_stm32
```

erwango, 2023-09-04:
https://github.com/zephyrproject-rtos/hal_st/pull/17 has been merged. You can update west.yml

closed 2023-09-05 by MaureenHelm