zephyrproject-rtos/zephyr #61345: zassert_equal fails for float comparisons on native_posix

martinjaeger, 2023-08-09:
**Describe the bug**

Below code snippet fails if run on `native_posix`:

```C
float f = 4.56F;
zassert_equal(f, 4.56F); /* evaluates to false */
```

**To Reproduce**

Steps to reproduce the behavior:
1. Copy above snippet into a test, e.g. `tests/ztest/base/src/main.c`
2. Run `west build -b native_posix tests/ztest/base -t run`
3. See that the test fails

**Expected behavior**

The variable and the floating constant should be equal.

**Impact**

Hard to reproduce and unexpected bugs (in unit tests or even somewhere else).

**Logs and console output**

GCC options:

```
$ ninja -v
[1/1] : && ccache /usr/bin/gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr.elf  zephyr/CMakeFiles/offsets.dir/./arch/posix/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/martin/XXX/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/arch/posix/core/libarch__posix__core.a  zephyr/soc/posix/inf_clock/libsoc__posix__inf_clock.a  zephyr/boards/posix/native_posix/libboards__posix__native_posix.a  zephyr/subsys/testsuite/ztest/libsubsys__testsuite__ztest.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L/home/martin/XXX/zephyr/build/zephyr  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -Wl,-no-pie  -m32  -ldl  -pthread  -lm && cd /home/martin/XXX/zephyr/build/zephyr && /usr/bin/cmake -E true && cd /home/martin/XXX/zephyr/build/zephyr && /usr/bin/cmake -E copy zephyr_pre0.map zephyr.map && /usr/bin/readelf -e zephyr.elf > zephyr.stat && /usr/bin/cmake -E copy zephyr.elf zephyr.exe
```

**Environment:**

 - OS: Manjaro Linux on x86_64 (Intel Core i5 Gen 8)
 - Toolchain: gcc-Version 12.2.1 20230201 (GCC)
 - Commit SHA: b40c052289ee46406e73a08974d903d9f10b0693

**Additional context**

After investigation by @keith-packard the emulated old 8087 FPU seems to be the root cause of the issue. Compiler flags `-msse` or `-std=C11` fix the issue.

We need to decide which compiler flags we should use for `native_posix` in Zephyr.

Related Discord discussion: 
https://discord.com/channels/720317445772017664/883404732423606303/1138773080198627408

Related GCC bug / discussion:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=92875

CC @aescolar @stephanosio @nashif

keith-packard, 2023-08-09:
Given the GCC discussion linked above, it seems clear that this wouldn't be considered a bug and so is unlikely to get changed.

If possible, switching to `-msse -mfpmath=sse` would both fix this issue as well as offer true 32/64 bit float computations so that results would match the usual floating point formats used in every other machine.

aescolar, 2023-08-10:
@keith-packard @martinjaeger Thanks for the very detailed analysis :)
I just posted a PR with one the fixes proposed here (I lean much more towards the switch for enabling the sse float, than towards switching the language mode)

keith-packard, 2023-08-10:
> @keith-packard @martinjaeger Thanks for the very detailed analysis :) I just posted a PR with one of the fixes proposed here (I lean much more towards the switch for enabling the sse float, than towards switching the language mode)

I agree -- it makes the native_posix port work more like other environments *and* improves performance. Probably need the TCS to read on that?

aescolar, 2023-08-10:
> Probably need the TCS to read on that?

You think any of this will be controversial?

keith-packard, 2023-08-10:
> > Probably need the TCS to read on that?
> 
> @keith-packard You think any of this will be controversial?

It changes the base platform requirements for the `native_posix` target so that it only works on targets with sse2 (see comment indicating that we need `-msse2` instead of `-msse`). That's only been around since 2004 :-)

aescolar, 2023-08-12:
> That's only been around since 2004 :-)

Given that the native boards are development/test targets, I think that is ok. If some developer finds it a problem, they can file an issue and is so we add a kconfig choice to disable this switch.

PR #61363 (2023-08-14): POSIX arch: Fix literal floating comparison in 32bit targets
When building the 32bit native board targets variants for x86(-64) hosts, gcc will promote float literals to double (See https://gcc.gnu.org/bugzilla/show_bug.cgi?id=92875 )

This can result in unexpected comparison differences.

This is due to the compiler using the 8087 float mode by default.
Instead let's tell the compiler to use the SSE float path, which is the default for 64 bit x86-64 builds.

The assumption that any x86 host used for development will have SSE support should be safe enough.

For more background see
https://gith [... cut]