zephyrproject-rtos/zephyr #61393: ztest crashes with "aborting essential thread" if zassert fails in IRQ context

amr-sc, 2023-08-10:
**Describe the bug**
ztest ends ungracefully if zassert_* test condition is not met in IRQ handler.

With ztest it is possible to run a test function in Timer ISR context (ZTRESS_TIMER).
If zassert_ happens to fail then the following routine is called:

```
static void test_finalize(void)
{
    if (IS_ENABLED(CONFIG_MULTITHREADING)) {
        k_thread_abort(&ztest_thread);
        k_thread_abort(k_current_get());
    }
}
```

The "current" thread is not applicable to ISR context and could be any thread which 
has been running in the system before the ISR, including the "essential" threads 
like "idle" thread which should never be aborted.

**To Reproduce**
The issue could be reproduced by triggering a zassert_* failure from ISR context.
E.g. you may change stress_read function in tests/lib/spsc_pbuf/:

```
bool stress_read(void *user_data, uint32_t cnt, bool last, int prio)
{
    zassert_true(false, "ASSERT");        
    return true;
}
```

Now you can reproduce the issue with qemu:
```
scripts/twister --no-clean -v  --platform native_posix -T tests/lib/spsc_pbuf/
```

**Expected behavior**
ztest should finish gracefully and should be irrelevant to the context in which the zassert_* test condition
fails

**Impact**
The issue is triggered when there IS an actual problem in the tests, i.e. when zassert_* fails.
So it's more of an annoyance kind, it could disturb the test results and confuse the developers.

But the way of how the ztest got teared down looks very fragile: even if we don't abort
the essential threads in test_finalize, we still may leave some zombie threads from previous
test cases unburied, so this problem could in fact be deeper than it seems.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.4.0-83-gdba488e76098 ***
Running TESTSUITE test_spsc_pbuf
===================================================================
START - test_stress
Reading from an interrupt, writing from a thread
    Assertion failed at WEST_TOPDIR/zephyr/tests/lib/spsc_pbuf/src/main.c:424: stress_read: (false is false)
ASSERT
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/kernel/sched.c:1741
	aborting essential thread 0x40000ded0
E:      a0: 0000000000000004    t0: 0000000000000000
E:      a1: 00000000000006cd    t1: 0000000000000009
E:      a2: 000000040000cdb6    t2: 0000000000000001
E:      a3: 0000000000000001    t3: 0000000000000019
E:      a4: 0000000000000001    t4: 0000000000000020
E:      a5: 000000040000d820    t5: 0000000000000023
E:      a6: 0000000000000004    t6: 0000000000000000
E:      a7: 000000000000002f
E:      ra: 0000000400008852
E:    mepc: 0000000400002354
E: mstatus: 0000000a00003800
E: 
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x40000ded0 (idle 00)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Zephyr v3.4

amr-sc, 2023-10-25:
Hi @carlescufi , @cfriedt ,

Guys, as I already mentioned in
https://github.com/zephyrproject-rtos/zephyr/pull/64157#issuecomment-1776679993

I'm doubting that we really fixed this issue

carlescufi, 2023-10-25:
Reopening as per @amr-sc

peter-mitsis, 2023-10-30:
> Guys, as I already mentioned in
https://github.com/zephyrproject-rtos/zephyr/pull/64157#issuecomment-1776679993
>
> I'm doubting that we really fixed this issue

The _ztest_ infrastructure does have a mechanism to allow an _after_ routine to be executed after each test in a specified test suite. It is that mechanism that should be leveraged for any thread aborts (along with any other state resets) to ensure that the next test starts in a clear and consistent environment.

Given this, I think ...

1. Any cases where the clean up is required should be tracked separately from this issue.
2. The current issue should be closed.

closed 2023-10-30 by peter-mitsis

amr-sc, 2023-10-31:
> > Guys, as I already mentioned in
> > [#64157 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/64157#issuecomment-1776679993)
> > I'm doubting that we really fixed this issue
> 
> The _ztest_ infrastructure does have a mechanism to allow an _after_ routine to be executed after each test in a specified test suite. It is that mechanism that should be leveraged for any thread aborts (along with any other state resets) to ensure that the next test starts in a clear and consistent environment.
> 
> Given this, I think ...
> 
>     1. Any cases where the clean up is required should be tracked separately from this issue.
> 
>     2. The current issue should be closed.

Hi @peter-mitsis , oh I see your point, thanks! 

Such a "shut-down" functionality would be a nice extension for the ztest. We even can reduce the scope of the tests to be modified to just the tests which show unstable behavior (like tests/lib/spsc_pbuf) while the other tests may remain with default common shut-down procedure. Do you already have a ticket for this task?

PR #64285 (2023-10-31): [Backport v3.5-branch] Add CODE_UNREACHABLE after k_thread_abort(k_current_get())
Backport c166685fcfeb9a11157c8d5aba8b2806df2aaf7c~2..c166685fcfeb9a11157c8d5aba8b2806df2aaf7c from #64157.

Fixes #61393