zephyrproject-rtos/zephyr #61407: bluetooth:eatt:Creating a credit based channel with insufficient authentication returns wrong response.

MSEEHenrik, 2023-08-11:
Using **L2CAP_CREDIT_BASED_CONNECTION_REQ** to create and configure new L2CAP channels.
The device returns a  **L2CAP_CREDIT_BASED_CONNECTION_RSP** package with a result code.
If the link is not encrypted, a status reply with a result code of 0x000E is expected.

0x000E - All connections pending - authentication pending.

Zephyr generates a status reply with a result code of 0x0005.

0x0005 - Connection refused - insufficient authentication.

The behavior is described in Core Specification 5.4 | Vol 3, Part A, section 4.25 and 4.26.

Tests: This behavior is tested with qualification test **L2CAP/ECFC/BV-43-C**.

Target platform: nrf52840dk_nrf52840.

Kconfig flags:
CONFIG_BT_L2CAP_ECRED=y
CONFIG_BT_EATT=y
CONFIG_BT_EATT_MAX=5
CONFIG_BT_EATT_AUTO_CONNECT=y

hermabe, 2023-08-11:
The "pending" statuses are only relevant for BR/EDR and not available on LE, see errata ES-18351. Which TCRL are you on?

MSEEHenrik, 2023-08-11:
I'm using the latest TCRL - TCRL-2023-1.

MariuszSkamra, 2023-08-18:
ES-18351 is still not resolved (incorporated).
I pinged people in https://bluetooth.atlassian.net/browse/ES-18351.
There's a waiver that can be used: https://bluetooth.atlassian.net/browse/ES-22070 until the issue is resolved

carlescufi, 2023-09-14:
> ES-18351 is still not resolved (incorporated). I pinged people in https://bluetooth.atlassian.net/browse/ES-18351. There's a waiver that can be used: https://bluetooth.atlassian.net/browse/ES-22070 until the issue is resolved

@hermabe @MSEEHenrik if this is the case, can we close this issue?

hermabe, 2023-09-14:
> > ES-18351 is still not resolved (incorporated). I pinged people in https://bluetooth.atlassian.net/browse/ES-18351. There's a waiver that can be used: https://bluetooth.atlassian.net/browse/ES-22070 until the issue is resolved
> 
> @hermabe @MSEEHenrik if this is the case, can we close this issue?

Yes. There is a waiver available until next TCRL is out.

closed 2023-09-14 by carlescufi

MSEEHenrik, 2023-09-15:
Yes - It can be closed.