zephyrproject-rtos/zephyr #61410: Thread name in log messages

jori-nordic, 2023-08-11:
When debugging applications which have APIs that are callable from multiple threads, it's nice to know which thread the debug log messages came from.

I propose to have another (opt-in) prefix to the log line: `[my-thread-name]`. Here's some sample messages:
```
[00:00:04.071,533] <dbg> [BT RX] bt_att: tx_meta_data_alloc: in RX context: disabling timeout
[00:16:39.962,066] <inf> [main] central: notify_peers: notify: 60:7E:E3:46:D2:B1 (random)
[00:16:39.962,066] <inf> [sysworkq] bt_hci_core: tx_processor: TX process start
```

The hacky way I made this:
```diff
diff --git a/subsys/logging/log_output.c b/subsys/logging/log_output.c
index 3b77f6f556..7826eca5e0 100644
--- a/subsys/logging/log_output.c
+++ b/subsys/logging/log_output.c
@@ -339,6 +339,11 @@ static int ids_print(const struct log_output *output,
                total += print_formatted(output, "%s/", domain);
        }
 
+       /* hack: add thread name */
+       #if defined(CONFIG_THREAD_NAME)
+       total += print_formatted(output, "[%s] ", k_thread_name_get(k_current_get()));
+       #endif
+
        if (source) {
                total += print_formatted(output,
                                (func_on &&
```

I can prepare a PR, but I'd like to get some guidance on the proper (configurable, maintainable) way of achieving this.

jori-nordic, 2023-08-21:
@pdgendt where in the logger should I place this?

Is the place the patch is good enough or is it too late in the processing? I ask because the other stuff (the domain, etc) seems to come all the way down from the macros.

pdgendt, 2023-08-22:
> @pdgendt where in the logger should I place this?

I'm not sure how far up the chain this should go, looking at the macros, I assume it should be handled similar to domain/level/...

CC: @nordic-krch @dcpleung

jori-nordic, 2023-08-29:
Just ran this on target and it looks like it doesn't work for deferred logging. So will probably have to mess with the macros then :/

PR #62109 (2023-08-31): logging: Add option to prefix message with thread ID
Added `CONFIG_LOG_THREAD_ID_PREFIX`.
If set then `tid` is stored in the log message and log message formatting can include thread information (name if `CONFIG_THREAD_NAME=y`)

Example output enabling thread prefix in the logging sample:

```
demo_tag [00:00:01.165,204] <err> [log_demo_thread_id] main: Error message example.
demo_tag [00:00:01.165,230] <wrn> [log_demo_thread_id] main: Warning message example.
demo_tag [00:00:01.165,258] <inf> [log_demo_thread_id] main: Info message example.
```

Fixes #61410.
closed 2023-08-31 by carlescufi