zephyrproject-rtos/zephyr #61413: Callback on sleep enter/leave

maxd-nordic, 2023-08-11:
**Is your feature request related to a problem? Please describe.**
To support ETB on nRF91 in a power-optimized way, we used a power-management interface that was removed from Zephyr: https://github.com/nrfconnect/sdk-nrf/commit/bd581bca7fed90d457b25c892f53741815e22ee6

**Describe the solution you'd like**
To reintroduce the feature, we need to execute code right before and right after sleeping. Please provide a callback interface for that.

**Describe alternatives you've considered**
Instead of callbacks, other ways of notifying may be possible. Also, this is highly SoC-specific and might be placed somewhere in the SoC files.

gmarull, 2023-08-11:
Clarification: nothing has been removed from Zephyr. The main problem in ETB implementation was that it relied on `_kernel.idle`, a Kernel internal field only available if PM is enabled. In a first place, users should never rely on internals to build stuff. So this issue is about the possibility to add a new public, documented interface that allows users to query the scheduled idle time by the Kernel. There's nothing to access this information right now.

maxd-nordic, 2023-08-14:
@gmarull This new title does not fit what we need.

maxd-nordic, 2023-08-14:
@jtguggedal

gmarull, 2023-08-14:
> @gmarull This new title does not fit what we need.

It’s the main thing required, hooks are already available even if not ideal or well documented. The main issue with ETB code is that it’s accessing an internal Kernel struct member.