zephyrproject-rtos/zephyr #61434: Fuel Gauge Battery Cutoff

aaronemassey, 2023-08-11:
## Introduction

Part of https://github.com/zephyrproject-rtos/zephyr/issues/44847 work.

### Problem description

Many fuel gauges embedded within battery packs expose a register address that
when written to with a specific payload will do a battery cutoff.  This battery
cutoff is often referred to as ship, shelf, or sleep mode due to its utility in
reducing battery drain while devices are stored or shipped.

Fuel gauge driven battery cutoff is especially common in SBS compliant fuel
gauge ICs embedded within "smart" battery packs.

This functionality is not currently exposed by the fuel gauge API.

### Proposed change

Expose battery cutoff through the devicetree bindings and the fuel gauge API.

## Detailed RFC

### Proposed change (Detailed)

See PR:

1. Allow configurable cutoff support, address, and payloads through devicetree.
   1. Add a standard battery_cutoff.yaml binding to be "included"
   1. Add device that makes use of the aforementioned binding
1. Add support in the generic SBS Driver implementing the fuel gauge API. Such
   that there is minimal memory footprint for fuel gauges that do not support
   this feature.
   
### Graphic

[image]

### Dependencies

* Adding a battery_cutoff.yaml that may be included by various devices
* SBS driver is leveraged more as a generic driver
  * This makes sense given many fuel gauges are SBS compliant but also have
    battery cutoff (not an SBS feature).

## Alternatives

### Don't involve devicetree

This could arguably be done without using the devicetree at all.  We'd just be
adding custom drivers for each fuel gauge.  This would require us to maintain
multiple drivers instead of multiple yaml bindings.  A yaml binding is much
easier to change as time goes forward.  If we end up deciding that creating many
drivers is the better approach, then this is an easily reversable decision.
Even if we extracted the SBS logic as a driver library (something I plan on
doing in the coming weeks), we'd still be creating numerous small drivers that
we'd have to maintain.  That would likely get onerous in the long-term.

The battery cutoff register address and payload are quite specific to each chip
and can easily be represented in devicetree.

We have successfully been doing this for some time in the ChromiumOS EC Firmware
code.

aaronemassey, 2023-08-22:
Adding support via block write and then closing this issue.

rruuaanng, 2025-03-04:
Looks like it's done.

closed 2025-03-04 by rruuaanng