zephyrproject-rtos/zephyr #61441: Convert Zephyr/samples/net/cloud/tagoio_http_post for aws_http_post

SinghKarminder, 2023-08-12:
I have done chanes in the server address for aws. But getting dns server error as follows.
 **<err> aws_http_post: Could not resolve dns, error: -11**

I want to post an image to s3 bucket of aws. 
What other changes are required.

SinghKarminder, 2023-08-15:
Dear Developers

I want to send the images captured from camera which has integreted with the stm32f412g_disco microcontroller using the Quectel EC200U-CN 4G module for GSM connection. I am using the Zephyr library for that. The Zephyr has provided support for publishing data to AWS IoT Core using the MQTT protocol as per given in https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/net/cloud/aws_iot_mqtt link. I am able to send the data to AWS IoT Core using the MQTT protocol as per there suggestions.

But Zephyr does not provide any support for sending data to an S3 bucket for AWS using HTTP.
However, Zephyr offered support for sending arbitrary temperature values via HTTP to the TAGOIO server, which I tested, and it works fine as per given in https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/net/cloud/tagoio_http_post link.

The issue that I am facing is "Could not resolve dns, error: -11". This is because the function that Zephyr has defined to get the IP address is not able to resolve the dns for the server address of the AWS S3 bucket. The Function which Zephyr has used for getting server IP address is as follows.

while (dns_attempts--) {
           ret = getaddrinfo(AWS_SERVER, port, &hints, &addr);
            if (ret == 0) {
                       break;
            }
            k_sleep(K_SECONDS(1));
}
where the AWS_SERVERs with port, which I have explored are as follows but getting same error in all the options.
S. NO.      AWS_SERVER                                                                    PORT
1.)             [bucket_name.s3.amazonaws.com](http://bucket_name.s3.amazonaws.com/)                                    80
2.)             [bucket_name.s3.region.amazonaws.com](http://bucket_name.s3.region.amazonaws.com/)                        80
3.)             http://[bucket_name.s3.amazonaws.com](http://bucket_name.s3.amazonaws.com/)                          80
4.)             http://[bucket_name.s3.region.amazonaws.com](http://bucket_name.s3.region.amazonaws.com/)              80
5.)             [bucket_name.s3.region.amazonaws.com](http://bucket_name.s3.region.amazonaws.com/)                        443
6.)             https://[bucket_name.s3.region.amazonaws.com](http://bucket_name.s3.region.amazonaws.com/)            443

The parameter &hints it is using the TCP connection and will take this parameter automatically. The parameter &addr is used to store the resolved address and also no need to configure this parameter also. the only thing that we have to configure is the AWS_SERVER and port.

The above mentioned error is also faced with the TagoIo server if I used the TagoIO server name as "[api.tago.io](http://api.tago.io/)". But when I configure the DNS of TagoIO manually, i.e., "75.2.65.153" then it worked properly. But in case of AWS S3 the DNS address is not fixed it is changing very time. Hence I could not used DNS address manullay in this case.

You can help in folowing ways.
1.) Can you suggest correct parameters of AWS_SERVER and port for s3.
2.) Can you provide the fixed DNS address from the URL of the S3 bucket so that I can connect to s3 and send data to it? 
3.) You can provide other way how can I connect to s3 bucket of AWS using Zephyr library. 

I shall be very thankful to you.

nandojve, 2023-08-16:
duplicated #61526

closed 2023-08-16 by nandojve