zephyrproject-rtos/zephyr #61452: MPU9250 does not initialize

fatimatayyab71, 2023-08-13:
I am using the mpu9250 driver in zephyr-rtos, with sensor API generated with the help of MPU6050, but it does not initialize.
Although there is no error detected at the time of compiling.
Sensor API for MPU9250 [https://github.com/fatimatayyab71/mpu9250](url)

nordicjm, 2023-08-14:
Did you enable debugging? What does the debugging error message show?

fatimatayyab71, 2023-08-14:
I am new to the nRF connect sdk, i had run the debug on the connected device. 
[image]

nordicjm, 2023-08-15:
Debugging as in the logging subsystem and monitored the UART for any errors

closed 2023-10-30 by github-actions[bot]