zephyrproject-rtos/zephyr #61464: New USB device stack assertion on STM32

GeorgeCGV, 2023-08-14:
**Describe the bug**
When using the new USB device stack ECM class on STM32H7 the assertion is raised due to mutex usage within ISR.

**Expected behavior**
The existing stack doesn't have this issue. Assuming the new stack shall not fail as well.

**Impact**
New stack stops working.

**Logs and console output**
The "mutexes cannot be used inside ISRs" assertion happens within `udc_stm32_lock` that is called from `HAL_PCD_SetupStageCallback`.

```
#0  arch_system_halt (reason=4) at <workspace>/zephyr/kernel/fatal.c:32
#1  0x90077762 in k_sys_fatal_error_handler (reason=4, esf=0x240203a8 <z_interrupt_stacks+3496>) at <workspace>/zephyr/kernel/fatal.c:46
#2  0x90077988 in z_fatal_error (reason=4, esf=0x240203a8 <z_interrupt_stacks+3496>) at <workspace>/zephyr/kernel/fatal.c:131
#3  0x9008c3cc in z_arm_fatal_error (reason=4, esf=0x240203a8 <z_interrupt_stacks+3496>) at <workspace>/zephyr/arch/arm/core/aarch32/fatal.c:73
#4  0x9008c3ec in z_do_kernel_oops (esf=0x240203a8 <z_interrupt_stacks+3496>, callee_regs=0x2) at <workspace>/zephyr/arch/arm/core/aarch32/fatal.c:118
#5  0x90013e76 in _oops () at <workspace>/zephyr/arch/arm/core/aarch32\swap_helper.S:573
#6  <signal handler called>
#7  assert_post_action (file=0x900e0aa0 "WEST_TOPDIR/zephyr/kernel/mutex.c", line=101) at <workspace>/zephyr/lib/os/assert.c:44
#8  0x9007988a in z_impl_k_mutex_lock (mutex=0x240009ec <udc0_data+132>, timeout=...) at <workspace>/zephyr/kernel/mutex.c:101
#9  0x90098624 in k_mutex_lock (mutex=0x240009ec <udc0_data+132>, timeout=...) at <workspace>/<app>/zephyr/include/generated/syscalls/kernel.h:984
#10 0x900986a6 in udc_lock_internal (dev=0x900cb0d8 <__device_dts_ord_88>, timeout=...) at <workspace>/zephyr/drivers/usb/udc/udc_common.h:257
#11 0x90098704 in udc_stm32_lock (dev=0x900cb0d8 <__device_dts_ord_88>) at <workspace>/zephyr/drivers/usb/udc/udc_stm32.c:60
#12 0x90031564 in udc_ep_buf_alloc (dev=0x900cb0d8 <__device_dts_ord_88>, ep=0 '\000', size=8) at <workspace>/zephyr/drivers/usb/udc/udc_common.c:622
#13 0x90097ed8 in udc_ctrl_alloc (dev=0x900cb0d8 <__device_dts_ord_88>, ep=0 '\000', size=8) at <workspace>/zephyr/drivers/usb/udc/udc_common.c:646
#14 0x900322da in HAL_PCD_SetupStageCallback (hpcd=0x24003fa4 <udc0_priv>) at <workspace>/zephyr/drivers/usb/udc/udc_stm32.c:149
#15 0x900719f6 in PCD_EP_OutSetupPacket_int (hpcd=0x24003fa4 <udc0_priv>, epnum=0) at <workspace>/modules/hal/stm32/stm32cube/stm32h7xx/drivers/src/stm32h7xx_hal_pcd.c:2327
#16 0x900c3076 in HAL_PCD_IRQHandler (hpcd=0x24003fa4 <udc0_priv>) at <workspace>/modules/hal/stm32/stm32cube/stm32h7xx/drivers/src/stm32h7xx_hal_pcd.c:1140
#17 0x900988fa in udc_stm32_irq (dev=0x900cb0d8 <__device_dts_ord_88>) at <workspace>/zephyr/drivers/usb/udc/udc_stm32.c:361
#18 0x90013fc6 in _isr_wrapper () at <workspace>/zephyr/arch/arm/core/aarch32\isr_wrapper.S:264
```

**Environment:**
 - OS: irrelevant
 - Toolchain: gcc-arm-none-eabi-10.3-2021.07
 - Zephyr : [d9fe26](https://github.com/zephyrproject-rtos/zephyr/commit/d9fe261f8eae2ca8b999a35852024ecf2ec3dc2a)

GeorgeCGV, 2023-08-14:
Tested the old stack (_ECM device class_) against the same host (_ST usb host library_) and experienced similar issue:

```
#0  arch_system_halt (reason=4) at <workspace>/zephyr/kernel/fatal.c:32
#1  0x900749ae in k_sys_fatal_error_handler (reason=4, esf=0x24022018 <z_interrupt_stacks+2968>) at <workspace>/zephyr/kernel/fatal.c:46
#2  0x90074bd4 in z_fatal_error (reason=4, esf=0x24022018 <z_interrupt_stacks+2968>) at <workspace>/zephyr/kernel/fatal.c:131
#3  0x9008a34e in z_arm_fatal_error (reason=4, esf=0x24022018 <z_interrupt_stacks+2968>) at <workspace>/zephyr/arch/arm/core/aarch32/fatal.c:73
#4  0x9008a36e in z_do_kernel_oops (esf=0x24022018 <z_interrupt_stacks+2968>, callee_regs=0x2) at <workspace>/zephyr/arch/arm/core/aarch32/fatal.c:118
#5  0x90017c5a in _oops () at <workspace>/zephyr/arch/arm/core/aarch32\swap_helper.S:573
#6  <signal handler called>
#7  assert_post_action (file=0x900d9464 "WEST_TOPDIR/zephyr/kernel/mutex.c", line=101) at <workspace>/zephyr/lib/os/assert.c:44
#8  0x90076ad6 in z_impl_k_mutex_lock (mutex=0x24001750 <__net_if_eth_netusb_0+8>, timeout=...) at <workspace>/zephyr/kernel/mutex.c:101
#9  0x9008ce3e in k_mutex_lock (mutex=0x24001750 <__net_if_eth_netusb_0+8>, timeout=...) at <workspace>/<app>/zephyr/include/generated/syscalls/kernel.h:984
#10 0x9001fa80 in net_if_lock (iface=0x24001748 <__net_if_eth_netusb_0>) at <workspace>/zephyr/include/zephyr/net/net_if.h:544
#11 0x900219fe in net_if_carrier_on (iface=0x24001748 <__net_if_eth_netusb_0>) at <workspace>/zephyr/subsys/net/ip/net_if.c:4470
#12 0x9001a1e6 in netusb_enable (func=0x24000570 <ecm_function>) at <workspace>/zephyr/subsys/usb/device/class/netusb/netusb.c:101
#13 0x9001aa9a in ecm_status_interface (desc=0x240017e7 <cdc_ecm_cfg+56> "\t\004\001\001\002\n\006") at <workspace>/zephyr/subsys/usb/device/class/netusb/function_ecm.c:351
#14 0x9001abbc in ecm_status_cb (cfg=0x24001888 <cdc_ecm_config>, status=USB_DC_INTERFACE, param=0x240017e7 <cdc_ecm_cfg+56> "\t\004\001\001\002\n\006") at <workspace>/zephyr/subsys/usb/device/class/netusb/function_ecm.c:369
#15 0x9000cb68 in forward_status_cb (status=USB_DC_INTERFACE, param=0x240017e7 <cdc_ecm_cfg+56> "\t\004\001\001\002\n\006") at <workspace>/zephyr/subsys/usb/device/usb_device.c:1263
#16 0x9000c0ca in usb_set_interface (setup=0x2400319c <usb_dev>) at <workspace>/zephyr/subsys/usb/device/usb_device.c:799
#17 0x9000c5c8 in usb_handle_std_interface_req (setup=0x2400319c <usb_dev>, len=0x240031ac <usb_dev+16>, data_buf=0x240031a4 <usb_dev+8>) at <workspace>/zephyr/subsys/usb/device/usb_device.c:981
#18 0x9000c9a2 in usb_handle_standard_request (setup=0x2400319c <usb_dev>, len=0x240031ac <usb_dev+16>, data_buf=0x240031a4 <usb_dev+8>) at <workspace>/zephyr/subsys/usb/device/usb_device.c:1167
#19 0x9000ae02 in usb_handle_request (setup=0x2400319c <usb_dev>, len=0x240031ac <usb_dev+16>, data=0x240031a4 <usb_dev+8>) at <workspace>/zephyr/subsys/usb/device/usb_device.c:229
#20 0x9000b180 in usb_handle_control_transfer (ep=0 '\000', ep_status=USB_DC_EP_SETUP) at <workspace>/zephyr/subsys/usb/device/usb_device.c:336
#21 0x90031006 in HAL_PCD_SetupStageCallback (hpcd=0x24004c20 <usb_dc_stm32_state>) at <workspace>/zephyr/drivers/usb/device/usb_dc_stm32.c:1069
#22 0x9006ec42 in PCD_EP_OutSetupPacket_int (hpcd=0x24004c20 <usb_dc_stm32_state>, epnum=0) at <workspace>/modules/hal/stm32/stm32cube/stm32h7xx/drivers/src/stm32h7xx_hal_pcd.c:2327
#23 0x900bcdf6 in HAL_PCD_IRQHandler (hpcd=0x24004c20 <usb_dc_stm32_state>) at <workspace>/modules/hal/stm32/stm32cube/stm32h7xx/drivers/src/stm32h7xx_hal_pcd.c:1140
#24 0x9002f2da in usb_dc_stm32_isr (arg=0x0) at <workspace>/zephyr/drivers/usb/device/usb_dc_stm32.c:197
#25 0x90017daa in _isr_wrapper () at <workspace>/zephyr/arch/arm/core/aarch32\isr_wrapper.S:264
#26 <signal handler called>
#27 arch_cpu_idle () at <workspace>/zephyr/arch/arm/core/aarch32\cpu_idle.S:136
#28 0x9007651a in idle (unused1=0x0, unused2=0xaaaaaaaa, unused3=0x9007651b <idle+70>) at <workspace>/zephyr/kernel/idle.c:88
#29 0x00000000 in ?? ()
``` 

Not sure what the root cause is. The issues are not observed with *nix host.

jfischer-no, 2023-08-30:
These are different bugs. For the first, I am probably partly to blame for not having clear documentation or examples. The second is due to the current device stack design, and if the driver does not offload the callback call to a workqueue or thread, it hits various areas where the USB device is involved, I thought all drivers were fixed.

erwango, 2023-10-02:
@jfischer-no is that part of the fixes you pushed recently ?

jfischer-no, 2023-10-25:
> @jfischer-no is that part of the fixes you pushed recently ?

No, there are stm32 shim driver implementation issues.

FRASTM, 2024-02-06:
@loicpoulain do you have any comment on this ?

loicpoulain, 2024-02-06:
> @loicpoulain do you have any comment on this ?

Yes, so we need to move some operations out of isr context, I can look at this next week. I'll update that issue accordingly.

erwango, 2024-02-06:
> except if someone else can/want to fix that, I can look at this next week.

It would be great if you have some time. We're pretty busy with HW model migration, it would be helpful if you can tackle this.

erwango, 2024-06-17:
@GeorgeCGV, @loicpoulain Is that still valid ?

GeorgeCGV, 2024-06-17:
@erwango there was no change in the `HAL_PCD_SetupStageCallback` implementation in the drivers.

henrikbrixandersen, 2024-07-02:
I can confirm this issue. Any plans for reworking the driver to offload these tasks from ISR context?

carlescufi, 2024-07-10:
@erwango who can take a look at this issue?

MSeesko, 2024-07-18:
Hi, i just ran into this problem. Is a fix for this issue on the roadmap?

For the meantime - is there a workaround except disabling CONFIG_ASSERT?

ddavidebor, 2024-08-19:
Ran into the same issues on a [ST STM32H747 Discovery](https://docs.zephyrproject.org/latest/boards/st/stm32h747i_disco/doc/index.html#st-stm32h747i-discovery)

Currently blocking a project. Several pieces of code that ran with no issues on NRF52 trigger this bug on STM32 H7.

erwango, 2024-08-20:
Since we don't have a huge bandwidth atm, my proposal is to work on fixing this in new STM32 USB driver and keep current behavior in old version (which will avoid risk of adding regressions in an otherwise stable driver which will be deprecated soon anyway).
@loicpoulain For my information, did you already tried fixing this in new STM32 USB driver and if yes, did you face specific issues to do so ? Or more likely, you just didn't had time...

@jfischer-no Does it sound like a reasonable plan to you (taking into account the resources I can allocate to this) ?
Also quick questions as I'm not a USB expert:

Do you suggest using thread or the work queue to offload treatment ? I've seen that NRF implementation switched from workq (usb/device) to thread (udc) while NXP moved from thread to workq on the new driver. Is that just a matter of footprint ? There are more things to consider ?

jfischer-no, 2024-08-20:
> @jfischer-no Does it sound like a reasonable plan to you (taking into account the resources I can allocate to this) ? Also quick questions as I'm not a USB expert:

I think you will not be able to keep up with the progress of the DWC2 driver. It may be better to use all SoCs with the DWC2 controller with native driver. DWC2 driver implementation is vendor neutral, vendors have to implement quirks if required. To determine if this is already possible (STM32F4 devices should work fine), we need to know what IP configuration the STM32 SoCs have. Extracting it from the register files in the HAL is a bit tedious. Other STM32 controller type may have a simpler shim or native driver.

> Do you suggest using thread or the work queue to offload treatment ? I've seen that NRF implementation switched from workq (usb/device) to thread (udc) while NXP moved from thread to workq on the new driver. Is that just a matter of footprint ? There are more things to consider ?

That is a question of taste. Both are okay.

erwango, 2024-08-21:
> I think you will not be able to keep up with the progress of the DWC2 driver. It may be better to use all SoCs with the DWC2 controller with native driver. DWC2 driver implementation is vendor [...]

I took some time to ask around internally and here is the status:

There are 4 types of USB IPs on overall STM32 offer (with a 5th one coming).
- 2 blocks are ST own IPs
- 2 IPs are DWC2 based (OTGHS, OTGFS), each one available in ~10 revisions.

For DWC2, this would represent more than 20 configurations versions to analyse and develop to make support available.
And even if I do that, DWC2 native driver would only work for part of the STM32, and I'd still need to maintain the HAL driver for the ST IPs.

OTOH, all of those IPs are supported through STM32 HAL based USB driver.

So, I do see indeed the pros in contributing to DWC2, but that is anyway significantly more work than continuing to maintain the HAL based driver for all STM32.

loicpoulain, 2024-08-21:
> @loicpoulain For my information, did you already tried fixing this in new STM32 USB driver and if yes, did you face specific issues to do so ? Or more likely, you just didn't had time...

Unfortunately I've not much time for now. The idea would indeed to mimic what other driver are doing, moving some stuff in valid thread/schedulable context.

erwango, 2024-08-21:
> Unfortunately I've no much time for now. The idea would indeed to mimic what other udc drivers are doing, moving some stuff in valid thread/schedulable context.

Ok, sure thanks. Next task is to define "stuff" :D

ddavidebor, 2024-10-21:
Not stale ! Still very problematic

PR #80820 (2025-01-28): drivers: usb: udc_stm32: Handle HAL callbacks on separate thread 
All HAL callback handling is offloaded to a separate thread.
Tested on STM32H747 (arduino portenta) with the high speed controller.

This is an attempt at fixing https://github.com/zephyrproject-rtos/zephyr/issues/61464.
Marked as a draft, as i would at least like to do some minor stylistic cleanup before submitting, but if anyone could help test, I would appreciate it. As far as i can tell, no additional locking should be required, and no timing invariants are broken, but I could use some extra eyes on that as well.

Fixes: #61 [... cut]
commit 7b19e4f1 (2025-01-28): drivers: udc: stm32: Offload data callbacks to thread

All HAL callback handling is offloaded to a separate thread, as they
involve non isr-compatible operations such as mutexes.

Fixes #61464
closed 2025-01-28 by kartben