zephyrproject-rtos/zephyr #61465: A disconnect shortly after a pairing completes results in pairing failed

jilu-ot, 2023-08-14:
**Describe the bug**

A device that disconnects the BLE connection shortly after having completed a pairing will for some controllers make the Zephyr host report the pairing as failed.

This happens because the smp_pairing_complete handler in smp.c detects that the connection is in the BT_CONN_DISCONNECT_COMPLETE state and then aborts the pairing.

This is because the disconnect event is processed before the smp_pairing_complete handler is called even though the pairing completes before the disconnect in the air.

**To Reproduce**

#### Devices

1. Device 1 - A BLE device that wants to pair with Device 2.
1. Device 2 - A BLE device running Zephyr version 3.4.0.

#### Steps to Reproduce

1. Device 2 is in pairing mode and advertises.
1. Device 1 connects to Device 2.
1. Device 1 writes a value to a GATT characteristic. (Not sure this step is necessary to reproduce the bug)
1. Device 1 pairs with Device 2.
1. Pairing completes successfully in the air.
1. Device 1 disconnects from Device 2 very shortly after pairing completed in the air.
1. Pairing is reported failed to the application in Device 2.

#### Analysis

The order of events received by the application in device 2 depends on the choice of bluetooth controller.

The following order of events is seen in the debug output when using PacketCraft controller version 3349 for the nRF5340 Audio board as Device 2.

| Prefix      | Description               |
|-------------|---------------------------|
| APP         | The application           |
| ZEPHYR CONN | The conn module in Zephyr |
| ZEPHYR SMP  | The smp module in Zephyr  |

1. APP: connected handler
1. ZEPHYR SMP: smp_pairing_req
1. APP: pairing accept handler
1. ZEPHYR SMP: legacy_pairing_req and more SMP stuff from the pairing process.
1. APP: settings save handler (remove key)
1. ZEPHYR SMP: bt_smp_encrypt_change and more SMP stuff from the pairing process.
1. APP: security changed handler
1. APP: identity resolved
1. APP: settings save handler (remove ccc)
1. ZEPHYR CONN: disconnect complete (state set)
1. ZEPHYR SMP: smp_signing_info
1. ZEPHYR SMP: smp_pairing_complete - Not connected!
1. APP: settings save handler (remove key)
1. APP: pairing failed
1. ZEPHYR SMP: bt_smp_disconnected
1. APP: disconnected handler

A different order of events can occur with other controllers. The Zephyr controller gives a different result on the nRF5340 Audio board. But the same pairing failed event also happens with our own Zephyr based controller for our own platform.

The above order of events observed shows that the disconnect state is set before the smp_pairing_complete handler is called which makes it abort pairing even though the pairing DID complete successfully.

**Expected behavior**

The diconnect event should be handled after the pairing complete handler has been called since it comes after the pairing events (SMP) in the air time-wise.

**Impact**

This bug makes it impossible for a device to only maintain a connection while it is using it. For instance as described in the steps to reproduce.

**Logs and console output**

None.

**Environment (please complete the following information):**

- Zephyr 3.4.0 (Based on 13a6c454529e64cdb9f69cab0f0a9fb8c539b192)

**Additional context**

None.

jilu-ot, 2023-08-14:
I just discovered that bt_hci_evt_get_flags in hci_driver.h gives disconnects priority.

Removing that priority makes the above pairing flow work again.

Like this:

```c
static inline uint8_t bt_hci_evt_get_flags(uint8_t evt)
{
	switch (evt) {
	case BT_HCI_EVT_DISCONN_COMPLETE:
		// return BT_HCI_EVT_FLAG_RECV | BT_HCI_EVT_FLAG_RECV_PRIO;
		return BT_HCI_EVT_FLAG_RECV;
		/* fallthrough */
#if defined(CONFIG_BT_CONN) || defined(CONFIG_BT_ISO)
	case BT_HCI_EVT_NUM_COMPLETED_PACKETS:
#if defined(CONFIG_BT_CONN)
	case BT_HCI_EVT_DATA_BUF_OVERFLOW:
		__fallthrough;
#endif /* defined(CONFIG_BT_CONN) */
#endif /* CONFIG_BT_CONN ||  CONFIG_BT_ISO */
	case BT_HCI_EVT_CMD_COMPLETE:
	case BT_HCI_EVT_CMD_STATUS:
		return BT_HCI_EVT_FLAG_RECV_PRIO;
	default:
		return BT_HCI_EVT_FLAG_RECV;
	}
}

```

But the PR that introduced this says it was done to resolve a deadlock. See PR https://github.com/zephyrproject-rtos/zephyr/pull/25954

So I guess the priority cannot just be removed. But it could if this deadlock issue has since been resolved. Does anyone know something about this?

alwa-nordic, 2023-08-15:
Looking at the issues resolved by the above referenced PR https://github.com/zephyrproject-rtos/zephyr/pull/25954:
 - https://github.com/zephyrproject-rtos/zephyr/issues/23364
   I think this will not be a problem since https://github.com/zephyrproject-rtos/zephyr/pull/60309 was merged.
 - https://github.com/zephyrproject-rtos/zephyr/issues/26467
   This needs a new fix. We should ensure that a call to `bt_conn_unref` is never blocking.
   
   Ideally, we want to be able to put the call to `bt_le_adv_resume` on the system work queue. This requires restructuring `bt_le_adv_resume` from a blocking function into an event-driven co-routine. The blocking stems from using `bt_hci_cmd_send_sync`, which will have to be eliminated. The code for `bt_le_adv_resume` looks complex, so it may be a lot of effort.
   
   Alternatively, there may be some other thread we can pile `bt_le_adv_resume` onto that will tolerate blocking. Nothing comes to mind immediately, other than a new thread or maybe the "bt long work queue". @jori-nordic: I remember we have been experimenting with this before, but it proved to not work. Do you remember the reason?

I agree in principle with this issue and the fixing PR. The disconnection should happen after the data is processed, if the data comes first. (By, 'first', I mean over HCI from the Host's point of view. The controller has to ensure it does not send the disconnect before host has sent all expected host-num-complete commands.)

jori-nordic, 2023-08-16:
Cf joakim's deadlock fix: I think this is still relevant, at least skimming the issue description I think it could still happen.

@alwa-nordic and I have been preparing a rewrite of the data and command flow throughout the host. 
We will very soon open a PR for it, and that should resolve once and for all <sup>(famous last words)</sup> the RX thread lockup issues.

Then we can remove this disconnect from the prio event list. This list should only ever have _complete events, that signal to the host that the controller has a cmd/data free buffer.

Regarding this:
> The blocking stems from using bt_hci_cmd_send_sync

When that's merged, we probably won't need an async/coroutine approach for resuming the advertiser. 

We have also been toying with the idea of adding a callback/event listener for when a connection object is free/available. 
Then the app (or `adv.c`) could restart advertising from there instead of the hack that we currently have in `bt_conn_unref()`. This would also make that troublesome (but useful) feature work with `CONFIG_BT_EXT_ADV`.

carlescufi, 2023-08-17:
The PR that introduced the issue, https://github.com/zephyrproject-rtos/zephyr/pull/59807, was solving a different problem. However, we need to make sure when modifying that code to avoid re-introducing it: https://github.com/zephyrproject-rtos/zephyr/issues/59788.

@jhedberg `smp_pairing_complete()` is being called in the `BT_CONN_DISCONNECT_COMPLETE` state.

closed 2023-09-06 by carlescufi