zephyrproject-rtos/zephyr #61545: net: getsockname: fails for bound socket

cfriedt, 2023-08-16:
**Describe the bug**
`getsockname()` fails, setting `errno` to `EINVAL` when there is no connection handler, even when a socket has been bound with `bind()`.

The exact snippet is
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/sockets/sockets.c#L2390-L2393 

So either the connection handler is not getting set within the call to `bind()` or this check is invalid.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? `qemu_x86`
 - What have you tried to diagnose or workaround this issue? step-debugged, found broken check
 - Is this a regression? Unsure.

**To Reproduce**
Steps to reproduce the behavior:
1. see sample code below

**Expected behavior**
`getsockname()` should return successfully, returning the assigned sockaddr

**Impact**
This is very clearly broken, and will affect any service using an ephemeral port number that wishes to obtain the assigned port number from the network subsystem.

**Logs and console output**
```
*** Booting Zephyr OS build v3.4.0-rc2-150-g83587a3cbca9 ***
I: Type 'quit' to quit
E: getsockname: 22
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK v0.16.1
 - Commit SHA or Version used: e0e7f5888f63555fbfa028331dfaf450d004d965

**Additional context**
Discovered in the context of working on #59669

Sample code:
```cpp
		/* Create a socket */
		ret = socket(af, SOCK_STREAM, 0);
		if (ret < 0) {
			LOG_ERR("socket: %d", errno);
			ret = -errno;
			break;
		}
		fd = ret;

		if (setsockopt(fd, SOL_SOCKET, SO_REUSEADDR, &(int){1}, sizeof(int)) < 0) {
			LOG_ERR("setsockopt: %d", errno);
			ret = -errno;
			break;
		}

		/* Bind to the specified address */
		if (bind(fd, &addr4, len) < 0) {
			LOG_ERR("bind: %d", errno);
			ret = -errno;
			break;
		}

		if (*port == 0) {
			/* ephemeral port - read back the port number */
			len = sizeof(addr4);
			ret = getsockname(fd, &addr4, &len);
			if (ret < 0) {
				LOG_ERR("getsockname: %d", errno);
				ret = -errno;
				break;
			}
			*port = addr4.sin_port;
		}
```

cfriedt, 2023-08-16:
The fix for me was removing the check, but I have not yet submitted a PR

rlubos, 2023-08-16:
That check indeed looks wrong as it covers only part of the cases, I think that `ctx->local` should be checked instead to verify whether the socket has been bound (either by explicit call to `bind()`, or as a result of `connect()` and others). In case `net_sin_ptr(&ctx->local)->sin_addr`/`net_sin6_ptr(&ctx->local)->sin6_addr` pointer is NULL, the function should report an error.

cfriedt, 2023-08-16:
It isn't 100% clear from the link below, but I always thought that e.g. the fetched `sin_port` value should be in network byte order from a call to `getsockname()`. It's in host byte order according to my experiments.

https://pubs.opengroup.org/onlinepubs/9699919799/functions/getsockname.html

cfriedt, 2023-08-16:
```shell
gcc -o /tmp/blah /tmp/blah.c
/tmp/blah &
# connect to the one that says "network byte order"
curl 'localhost:33967'
```

```cpp
#include <netinet/in.h>
#include <stdio.h>
#include <stdint.h>
#include <unistd.h>
#include <sys/types.h>
#include <sys/socket.h>

int main()
{
  int fd;
  int fd2;
  socklen_t len;
  struct sockaddr_in addr = {0};

  fd = socket(AF_INET, SOCK_STREAM, 0);
  if (fd < 0) {
    perror("socket");
  }
  if (setsockopt(fd, SOL_SOCKET, SO_REUSEADDR, &(int){1}, sizeof(int)) < 0) {
    perror("setsockopt");
  }
  len = sizeof(addr);
  if (bind(fd, (struct sockaddr *)&addr, len) < 0) {
    perror("bind");
  }
  len = sizeof(addr);
  if (getsockname(fd, (struct sockaddr *)&addr, &len) < 0) {
    perror("getsockname");
  }

  if (listen(fd, 10) < 0) {
    perror("listen");
  }

  printf("accepting connections at\n");
  printf("curl 'localhost:%u' # (host byte order), or\n", addr.sin_port);
  printf("curl 'localhost:%u' # (network byte order)\n", ntohs(addr.sin_port));

  len = sizeof(addr);
  fd2 = accept(fd, (struct sockaddr *)&addr, &len);
  if (fd2 < 0) {
    perror("accept");
  }

  close(fd);
  close(fd2);

  return 0;
}
```

rlubos, 2023-08-16:
> @rlubos - it also appears that `getsockname()` returns (at least) the port number in host byte order, but it should be returned in network byte order.

I can't really reproduce what you say, I'm using your example code and can successfully connect with the port printed under `network byte order`. I can also see, that when I bind the socket to a specific port explicitly (`addr.sin_port = htons(10000);`), it is printed correctly after `ntohs()` conversion (so it was stored in the network byte order, as `struct sockaddr_in` requires).

cfriedt, 2023-08-17:
> I can't really reproduce what you say, I'm using your example code and can successfully connect with the port printed under `network byte order`.

On Linux, that is what I expect, but I found it wasn't the case in Zephyr earlier today.

I'll give it a shot again.

PR #61636 (2023-08-27): net: sockets: Fix getsockname() fucntion and add tests to cover it
This PR relies on FIN handling being fixed in https://github.com/zephyrproject-rtos/zephyr/pull/61353, as this interferes with the tests (the issue (context leak) was already present in the test suite, but it did not manifest itself until new tests involving TCP were added). So DNM for now.

This PR fixes `getsockname()` function operation, and adds tests to cover its functionality. 

Additionally, the PR fixes a corner case encountered in the TCP implementation which prevented the tests from running properly (i. e. tests were  [... cut]
closed 2023-08-27 by cfriedt