zephyrproject-rtos/zephyr #61587: Configurable IPV6 MTU

Hiihtaja77, 2023-08-17:
https://github.com/zephyrproject-rtos/zephyr/blob/45dcc6c5dbfb98944133a7c046959d763021386a/include/zephyr/net/net_ip.h#L413

Can value of NET_IPV6_MTU be configurable ?

I use 1500 bytes quite often and need to modify source code.

jukkar, 2023-08-18:
The 1280 is just the minimum MTU that needs to be supported for IPv6. You can use `net_if_set_mtu()` to change the interface MTU.

Hiihtaja77, 2023-08-18:
But this constant is used to reserve some static buffers.
As usually net_if_set_mtu() is used for set network interface MTU what is less than 255 bytes and after that lowpan is able to fragment it. But overall IPV6 stack is not support more than NET_IPV6_MTU size.

PR #68791 (2024-02-28): net: ipv6: Allow user to configure the IPv6 MTU size
This makes it possible to set the minimum IPv6 packet size that can be sent without fragmentation. The default value is 1280 bytes. This commit allows user to set the IPv6 MTU value within reasonable limits [1280, 1500].

Fixes #61587
commit 1df8aaee (2024-02-28): net: ipv6: Allow user to configure the IPv6 MTU size

This makes it possible to set the minimum IPv6 packet size that
can be sent without fragmentation. The default value is 1280 bytes.
This commit allows user to set the IPv6 MTU value within reasonable
limits [1280, 1500].

Fixes #61587
closed 2024-02-28 by aescolar