zephyrproject-rtos/zephyr #61588: Create a policy for introducing changes in public and internal headers

wbober, 2023-08-17:
This is a sub-issue for https://github.com/zephyrproject-rtos/zephyr/issues/48887.

The issue is about establishing guidelines to indicate which functions and data structures declared in header files are public, internal, and private.

nashif, 2023-08-23:
my rough thoughts:

- **private api**: an api that is used within a logical sub-set (area, component, subsystem, etc.) of the software. such APIs have no life cycle and shall not be used outside the sub-set  they are defined in. private APIs can be changed or removed at any time and they do not need any deprecation. Private APIs historically were prefixed with underscore (`_`) but now use the `z_` prefix
- **internal api**: A class of APIs that serve as the interface between two or more low level  areas.  `arch_` is a good example of such an API whcih is the interface between the kernel and the architecture. However, those APIs have been also used out of tree and in SoC code making them pseudo public. We need to look at how they are used and if those should just become public APIs or change how they are used to extend architecture functionality out of tree (!internal).  This is a big area of discussion
- **public api**: Anything that a user or a vendor can implement or hook into out of tree. This include extending hardware support using device driver APIs, platforms hooks and middleware and subsystem APIs. It also include the APIs used to extend functionality of a subsystem, for example, a custom backend of a subsystem.

The idea here is to have all public APIs use a modified lifecycle based on the current API lifecycle mostly targeting device drivers and tweak the life cycle to accomodate different classes of API, i.e. application facing APIs, hardware support APIs and other types. For example, a user facing API might require more time to stabilize than a hardware interface or device driver API etc, same for deprecation periods and so on.  (just an example)...

keith-zephyr, 2023-08-30:
@nashif - Need to define next steps.  Defer discussion until @wbober is available.  @nashif to work with @wbober offline.

keith-zephyr, 2023-09-06:
Summary from Process WG.

1. Need to cleanup public headers (under include/zephyr).  Remove all uses of private/internal APIs.  Fix misnaming of public APIs that are using z_ prefix.
2. Also cleanup doxygen.  Make sure all files create a group and define an anchor.
3. Need to identify "core low-level areas".

@wbober - to open main bug to start cleanup of includes

----

Discussion of the draft API types/classes. on https://github.com/zephyrproject-rtos/zephyr/pull/62198

Start discussion with C header file APIs.

@nashif - z_ was meant to be private to the unit where they are defined.  Shouldn't be between units/subsystem.  Internal APIs are for use between units, but there some internal APIs using z_ (and public) that need to be cleaned up.

@nashif - public headers use doxygen hidden tags, but this should be avoided.  Users tend to look at headers before documentation.

@nashif - need a guard from including headers outside of include/zephyr.

@wbober - next simple step, grep through code and identify z_ symbols and decide on an action (misnamed, move to somewhere else).  @nashif - there will be a lot (1000+ references).

@nashif - need to classify usage of APIs.  Need to identify items that don't belong in the public headers at all.  Some internal APIs use z_ and need to move to a location for internal headers.

@keith-zephyr - are private/internal APIs ever allowed in public APIs.  @nashif - there should not.

@henrikbrixandersen - one exception is helper macros defined between functions.  This can be fixed by moving to an internal header, and using appropriate guards.

@nashif - contributors do document internal APIs, but this causes confusion to the users.   Current doxygen rule bypass z_ APIs and.

@kartben - we should adopt "@internal" tags.

@kartben - will look into using "@internal" to see if there are any problems with Zephry's usage.

@nashif - too many files at top level of include/zephyr.  Some should be moved to sub-directories.

@kartben - would like pair up to cleanup kernel.h.  @nashif suggests Peter.  Some items missing documentation for a very prominent public API.

@jfischer-no - Did @nashif try to remove z_ previously?  This was done as a forcing function.  It's not possible to just use sed to drop all usage.

@jfischer-no - double-underscore reserved for compiler provided functions.

@nashif - for coding guidelines, single underscore was replaced by z_ to avoid toolchain conflicts.  Other code bases use "p_" (private).  Users have been confused by z_ - private APIs are easily discoverable and are used.   Users assume "z_" means zephyr-API.

@jfischer-no - Linux kernel provides an API directory, with separate public and private files.  

@nashif - should be able to fix include/zephyr incrementally.  

@wbober - we need to agree on the guidance, which will then allow users to start fixing violators.

@nashif - can defer dropping "z_" prefix down the road.

@wbober - Actions for cleanup: identify public routines misusing z_, identify private/internal APIs that are present under include/zephyr, but should be moved.
@wbober - suggests divide and conquer asking maintainers to help.  @nashif - probably need to file bugs to actually get action from the maintainers.  Need an owner to proactively drive this.

@nashif - Example of public APIs using z_ https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/sys/fdtable.h.  @kartben - these functions aren't included in the documentation.  The file is missing the group specifier to designate a landing spot in the published documentation.

@nashif - need a check to catch when functions documented with doxygen but missing the group and the anchor.

@gregshue - what's the distinction between private and internal APIs?

@gregshue - internal APIs are essentially public.  

@nashif - internal APIs are for use by the subsystem, and provides the glue between basic components.

@gregshue - what about downstream users that implement a new architecture? 

@galak - Downstream users can potentially access all APIs.  The internal/public distinction is more about documenting the contract/life cycle of the API class.

@gregshue - description of "internal" needs more definition to be clearer to users.  @nashif - agrees that the definition needs expansion.

@dleach02 - need to identify the core low-level areas that fall into the "internal" APIs.

@nashif - doxygen can hopefully catch these problems and improve the tooling.

@kartben - use EXTRACT_ALL to detect problems. https://github.com/zephyrproject-rtos/zephyr/issues/56855

gregshue, 2023-09-06:
> @galak - Downstream users can potentially access all APIs. The internal/public distinction is more about documenting the contract/life cycle of the API class.

I think this case justifies and gives a good characterization of an "internal" category of APIs. 

I'm not convinced "internal API" really captures the issue @nashif seems most concerned about: that of clients not following the supported layering of interfaces (e.g., calling directly into an arch API instead of going through a kernel API). Anas' concern is a very good one, and the issue can appear across any pairing of subsystems/drivers/hal/etc. Whatever solution we come up with needs to naturally scale to the scope of the problem. Any suggestions for a better phrase to describe his concern?

wbober, 2023-09-07:
Created https://github.com/zephyrproject-rtos/zephyr/issues/62378 to track the clean-up.

gregshue, 2023-09-07:
BTW, I just found documentation that `module.yml` supports an `arch_root:` key. It looks like this was introduced into `$workspace/zephyr/scripts/zephyr_module.py` by Aug 2020. Doesn't this require that the Arch/Kernel interfaces be public?

keith-zephyr, 2023-09-13:
Process WG 2023-09-13
@wbober to incorporate updates the API classses.
@wbober - wants to close on internal/private headers next meeting
@wbober - to create definition of "software component"

____
@wbober - Expanded definitions of private and internal API classes.

@gregshue - Is the only difference between internal/public, is the API lifecycle.  Are we concerned with breaking out of tree users with changes to internal APIs?

@nashif - internal APIs and out of tree usage is in conflict.

@wbober - what about the arch_ APIs that allow out of tree users to add a new architecture?

@nashif - officially out of tree architectures are not supported.  Internal APIs will not be subject to a lifecycle.  There are many examples using internal APIs that need to be fixed.

@nashif - user of internal APIs is only Zephyr in tree.  Hooks is an example of this.  Users are internal, implementers provide a public API.

@gregshue - APIs that cross modules/subsystem boundaries, need to be controlled.

@nashif - Example for public/not callable by application APIs: https://github.com/zephyrproject-rtos/zephyr/pull/62595#discussion_r1324426207

@wbober - Public APIs include top facing to the application, bottom facing to the drivers is an example.

@gregshue - which APIs that are currently used by a module.  @nashif - suggests skipping module discussion.   @gregshue other repos can implement backends that hook into Zephyr subsystems.  

@gregshue - struggling to see the distinction between public and internal APIs.  @wbober - private has a defined prefix, but internal needs some work.  @gregshue believes that as defined, the set of internal APIs will be very small following cleanup.

@nashif - With these new rules, out of tree users have clear documentation as to when they are using non=public APIs and have to accept changes at any time.

@wbober - Need an indication for intended usage for public APIs.  

@keith-zephyr - Do we need a set prefix for internal APIs.  @wbober - we can define a list of known internal API prefixes.  A single prefix maybe isn't needed.

@nashif - need a single place of truth that documents internal API prefixes.  There are inconsistencies.  Suggests adding a internal flag to the doxygen

@fabiobaltieri - Suggests an different namespace for the internal header files.  

@wbober - how about include/Zephyr for public, include/private and include/internal.  @fabiobaltieri - this will be a big effort to refactor headers.

@wbober - any objections to excluding internal APIs from API lifecycle.  None from the group.

@kartben - is there a middle ground for internal API?  Maybe skip documenting in release notes?  @dleach02 - changes to internal APIs still need documentation.

@keith-zephyr - suggest changing language to say API lifycle only applies to Zephyr.  External modules do their own thing.

@wbober - has anyone looked for violations?  @nashif has found some organically, no plans to document these fixes.

@nashif - arch APIs needs to cleanup to refactor into the include file namespace changes.

@wbober - There are multiple "arch_interface.h" files defined in the tree, which is confusing.  @nashif - one of these is private to kernel

@wbober - need a proposal as to what falls into the public API space.    @nashif, @keith-zephyr - make this anything that is added to include/Zephyr.  

@gregshue - what's the distinction between component and subsystem?  On other projects, components are the smallest replaceable unit.  Wants to see how components rare elated to the public APIs.  

@nashif - more commonly projects use "module" to define a software unit.  But zephyr has it's own definition of module.  

@kestewart - Zephyr needs to provide a consistent definition. 
@nashif - area, component, subsystem are all terms used somewhat interchangably.

@gregshue components can be single C file or multiple C files.  It's a unit of composition.  
@wbober - drivers and backends are examples of components.

@wbober to update a document to define "software component" and add to the glossary.  @kartben  agrees with this approach.

keith-zephyr, 2023-09-20:
@gregshue - would like clarification on internal APIs.
@nashif - there are APIs beyond arch_ that are internal APIs.  These are APIs not usable by the application.  This includes many kernel APIs (but not for the application).
@gregshue what is the application in this context? The application does have visibility of kernel symbols.  The current definition of internal API doesn't have a clear distinction from public.
@nashif - public APIs are available to the application, and is maintained with a defined life cycle.  Applications can access internal APIs, but it's not publicly supported.
@gregshue why are internal APIs not just private?
@nashif  - a 3rd party creating a synchronization object would need to use the internal (low-level APIs).  
@keith-zephyr - internal API is just a place holder.  There is no fixed rules as to what APIs are internal vs. public.  @wbober - it really becomes that any API that added to zephyr/internal.
@gregshue - is there a better name than internal?  How about "non-managed"?  @nashif - internal is a well understood name.
@wbober - we can extend definition to indicate in-tree usage only.
@nashif - kernel.h has many undocumented APIs.  It also has many misclassified APIs/structures as "internal/hidden".  
@nashif - the API cleanup will likely need to promote some APIs to public because there is enough usage across the tree.
@wbober - best effort approach is to leave the internal API definition
@gregshue - How does user distinguish inline implementation that are internal/private? @wbober - use the location - internal symbols should be placed under include/zephyr/internal.  There is also "well-known" prefixes that are designated internal (arch_ for example). 
@gregshue arch_ may be too specific.  Suggests using a more general pattern.
@nashif - need a contract and source of truth.   Doxygen could be the source of truth.
@keith-zephyr - the contract is the header file location.  Prefix for internal APIs can be added later.
@kartben - Should internal be allowed in sub-directories?  e.g. include/zephyr/logging/internal
@wbober - will convert wiki naming conventions to rst file.

keith-zephyr, 2023-10-18:
Process WG: 2023-10-18

@wober - need to experiment with Kconfig tooling to explore what's possible in detecting Kconfig changes.  Need a list of requirements/feature requests that should be detected.
- Kconfig symbol renames
- changes in default assignment, could possible use a golden config to detect changes
- Changes to select/depends/imply hierarchy
- Changes to conditionals (this is already needed to detect changes to select/depends/imply)
- Conditional exposure of definitions (menuconfig).  Hiding a previously exposed symbol behind a new menuconfig could potentially break users
@fabiobaltieri - what is scope of symbols to check?  @wbober conclusion previously that Kconfig symbols need a relation to the subsystem so they can be tagged as experimental.
@keith-zephyr - generally Kconfig symbols are organized under the corresponding directory.  So this provides a relationship.
@wbober - versioning can be used to detect unstable/experimental APIs.  v1.0 is considered stable, less than 1.0 is not stable.
@nashif - public/internal/private still not well defined
@keith-zephyr - for Kconfig there are only internal and public.  Helper/promptless symbols are internal.  Everything else is public.
@gregshue - there are conditionally prompted symbols.  How are these classified?  Kconfig does support turning on/off prompts of Kconfig symbols - this is part of the spec.  @gregshue to provide an example.
@jhedberg found 7 instances in the tree
@dleach02 - if symbols are defined as internal - external projects can access the symbol, but it's outside the API contract
@gregshue - users can create a wrapper and hide configuration via conditional prompts.  Is promptless the correct signal for internal.
@fabiobaltieri  - promptless symbols can still be overridden by the user.
@gregshue - does standard Kconfig give us the control to validate internal/public. 
@fabiobaltieri - when could come up with a naming convention, but this will involve lots of changes
@keith-zephyr - let's defer the internal/public distinction and work on Kconfig symbol change detection. @wbober  and @nashif  agree with this approach  
@gregshue - what about moving symbols to another file? @nashif this shouldn't impact users. @gregshue - moving symbols between modules would be an example.  @nashif - probably not worth considering at this stage.  But we can re-evaluate after having some tooling in place.
@fabiobaltieri - Can we use the samples to detect breakages?  Golden application to detect breakages.  Do this instead of detecting Kconfig breakages specifically.  @wbober - this approach might require building the sample with every Kconfig combination.
@wbober - a variation is compiling all samples on every PR - but this is too expensive to do the in CI

mkschreder, 2023-10-20:
My general approach so far has been to follow the following rules:

* Public api: must be placed under "include". This folder can be either in project root or in component root, but it must be a root folder that replicates standard public include structure. For example, if a zephyr component is placed outside of tree, its public api would be under "component_dir/include/zephyr/..."
* Private api: always placed in headers that are outside of the public include folders. These paths are never added to global include paths. These headers are always included using quotation marks in C sources to indicate that they are private/internal.

gregshue, 2023-10-21:
>  @gregshue to provide an example [of conditional prompts used in Zephyr]:

From github.com/zephyrprojects-rtos/zephyr, main brainch at a4e9aed68 (2023-10-16):
```
./drivers/wifi/esp_at/Kconfig.esp_at: prompt "ROM start offset" if !BOOTLOADER_MCUBOOT
./modules/segger/Kconfig: prompt "TF-M SPM Log Level" if !TFM_LOG_LEVEL_SILENCE
./modules/trusted-firmware-m/Kconfig.tfm: prompt "TF-M Partition Log Level" if !TFM_LOG_LEVEL_SILENCE
./subsys/bluetooth/controller/Kconfig.df: prompt "Optimize for Speed" if !(SOC_SERIES_NRF51X && BT_CTLR_LE_ENC)
./subsys/bluetooth/controller/Kconfig.ll_sw_split: prompt "HCI Tx thread stack size" if BT_HCI_TX_STACK_SIZE_WITH_PROMPT
./subsys/bluetooth/controller/Kconfig.ll_sw_split: prompt "Minimum encryption key size accepted in octets" if !BT_SMP_SC_ONLY
./subsys/bluetooth/Kconfig: prompt "The interval (in seconds) to send the unprovisioned beacon" if BT_MESH_PB_ADV
```

keith-zephyr, 2023-10-25:
Process WG 10-25 - no updates. @wbober to find a resource at Nordic to look at Kconfig requirements.

keith-zephyr, 2023-11-08:
@wbober - Found someone to help with tooling for Kconfig checks.  Will use the #process channel on discord to handle questions.
@wbober Reached out to @mbolivar-ampere for the devicetree scripts, but still waiting on a response.
@wbober What are the DT requirements?
* detect renames
* public/private/internal - 
@galak these categories may not make sense. @fabiobaltieri possibly SoC have properties that are internal.
@nashif - does existance of a binding make it public?
@nashif - are yaml files referenced directly from a devicetree? No.  So moving a yaml file won't generally cause problems.
@fabiobaltieri - changes to base yaml classes could break out of tree users.
@wbober - Drivers have a binding.  How to detect/avoid impact of changes to the binding.  Should we restrict this to the device driver - make the binding match the stability of the driver source.
@galak - yes, the API stability is a good thing to follow.
@galak - all bindings can be considered public
@gregshue - Users use devicetree downstream.  All bindings are public by design.
@wbober - devicetree also has namespace considerations. @galak the compatible field scopes the namespace.
@galak - we could use a version number on the compatible string to help manage this.  @keith-zephyr - are there any examples of using a version with compatibles.  @galak - nothing that he can recall.
@keith-zephyr - Zephyr C APIs don't support multiple versions.
@wbober - consensus is that DT bindings are public.  The stability of the component the binding belongs to set the stability of the DT.   We need to apply the same rules for DT bindings.  @galak - agrees this is a good approach.
@wbober - will add these requirements to the API classification doc.

keith-zephyr, 2023-11-15:
Process WG 2023-11-15

@wbober - What are the next steps to merge https://github.com/zephyrproject-rtos/zephyr/pull/62198?
@wbober - DT and Kconfig will follow lifecycle the component they belong to.  How is the relation defined?
@keith-zephyr - Suggests using a tag to identify the component.  @wbober using the doxygen group is one possibility.
@kartben - this might require versioning of the symbols. @wbober APIs have a version, so Kconfig DT bindings will always match.
@nashif - versioning over complicates this.  Kconfig is not documented using Doxygen today.  
@wbober - using flash API as an example.  
@jfischer-no - DT bindings are more complicated.  Tying DT bindings to API lifecycle.  Currently DT bindings are changed frequently.  
@wbober - Intent of this discussion is to add more control to DT binding changes.  How else can we solve the problem?  Changing DT bindings break out of tree users.
@jfischer-no thinks including DT bindings in the API lifecycle is too broad.  Would like to see a hybrid model where some bindings are stable and some are experimental within a given component.
@wbober - is it too broad to include Kconfig and DT in the API space?  
@nashif - objection is trying to use the same infrastructure for C APIs for Kconfig and DT.
@dkalowsk - Kconfigs can select an API version, correct?  @nashif - we don't have this.  @dkalowsk - we should have this.  This allows downstream users to enable new features. @nashif - this has been done for major changes, but it should be the exception.  It was very hard to manage.
@nashif - managing the combinations of multiple APIs and versions is very hard.
@dkalowsk - disagrees that we can't manage multiple API versions but agrees to table the discussion.
@wbober - What does it mean in practice to change Kconfig and DT? 
@nashif - we do need tests to detect API changes.  We need to define the goal.  Maintainers have the responsibility to document changes to stable APIs.
@wbober - if a contributor changes a C API and Kconfig, how does a new contributor discover that these changes may be subject to the API lifecycle.
@nashif - has seen issues where someone breaks an API, but also changes the test that hides the API change.
@keith-zephyr - is there an objection to tracking Kconfig symbols and DT bindings  under the C API lifecycle.
@nashif - yes this is too broad.  Not every Kconfig is associated with a header file.
@wbober - Can we define rules that specific which Kconfig and DT bindings are directly part of the module contract and designate others that are not subject to this.
@nashif - We need to start actually investigating at the code level and do a case study of the Kconfig/DT namespace.    This can help identify areas that can have tests to catch problems.   In short, start at the bottom instead of from the top.
@jfischer-no Top Kconfig symbols are likely to cause breaking changes.
@nashif - suggests creating draft PRs to add tests to detect Kconfig and API changes.
@wbober - needs to regroup with colleague to see what next steps might be.
@wbober - the process WG wants to seem some practical examples for tracking/detecting API, Kconfig, and DT changes.
@nashif - suggest creating an example in one specific area.   Suggest a small area, maybe a driver class (avoid subsystems at the moment).    This will help identify inconsistencies in approaches (naming conventions, etc).  Kconfigs in particular don't have consistent naming conventions.

keith-zephyr, 2023-12-13:
Process WG notes 2023-12-13

Discussion about landing https://github.com/zephyrproject-rtos/zephyr/pull/62198while tooling experiments continue.

@nashif - This documentation is needed so users have something to reference.
@nashif - will take another look and make suggestions on the documentation.
@nashif - linked to https://github.com/zephyrproject-rtos/zephyr/pull/61994#issuecomment-1845211576 as an example of something that may fall outside the API definitions
@wbober - Will move PR outside of draft status and submit for final review.
@wbober will also solicit feed back from @nordicjm 
@wbober - wants review by end of week.

lmapii, 2024-08-24:
I stumbled upon this issue when looking for any documentation on `DT_HAS` which is also very useful for out of tree drivers and/or libraries. It'd be great to also just place a one-liner in the `Device Driver Model` page?